esphome/feature-requests #1887: Please Add Zengge bluetooth light support for ESP32

Korte68, 2022-09-22:
Currently ESPHome did not support Zengge bluetooth controlled lights.

Currently only Home Assistant support this, but it would be great if ESP32 can support it as well.

 HA server is too far from the lights, and the range of bluetooth not cover it. If we can deploy an ESP32 to pair with the Zengge BT light, and controller it.

**Additional context**

nagyrobi, 2022-09-22:
You will be able to extend your Bluetooth coverage with [bluetooth_proxy](https://esphome.io/components/bluetooth_proxy.html). Currently it's only able to receive BLE messages from anywhere an ESP32 node is located, but work is being done to be able to use it like a local dongle, through the network.

Korte68, 2022-09-22:
Thank you. Promising feature.., but I prefer the local ESP support, because in this case I can control the lights without HA if the HA is not reachable.

nagyrobi, 2022-09-22:
Yes, that means you have to wait until work is finished.

Korte68, 2022-09-22:
I will. Thank you