esphome/feature-requests #1889: Add a function to set the volume of the rtttl buzzer

nilskt, 2022-09-24:
**Describe the problem you have/What new integration you would like**
Add possibility to set the volume of the buzzer. ( https://esphome.io/components/rtttl.html ) 

**Please describe your use case for this integration and alternatives you've tried:**
Use the M5STICK CPlus as a temperature alarm. The default sound volume of the buzzer is too low.

**Additional context**

It is possible in this project: https://github.com/StarsoftAnalysis/ESP8266RTTTLPlus

dmg244, 2023-03-16:
Yes, really need it.

dmg244, 2023-03-16:
[Here you can find how do it.](https://community.home-assistant.io/t/esphome-rtttl-passive-buzzer-volume-control/442536/19)

ssieb, 2023-03-17:
You can possibly make it quieter, but you can't make it louder.  It's all up to the speaker you use and the maximum current from the ESP gpio pin.

dmg244, 2023-03-17:
I'm using exactly buzzer like [here](https://esphome.io/components/rtttl.html) with transistor. And current flow via transistor. Maybe the buzzer is not loud enough, I don't know. But there are situations in which it is really important to be able to change the sound volume.

ssieb, 2023-03-17:
Ok, but it's not really possible with this method of producing the sound.