esphome/feature-requests #1893: Sprinkler - Run a valve for a specific amount of time.

Zebble, 2022-09-24:
The start_single_valve action works for starting a specific valve.  The problem is that it uses the current run_duration for that valve and also uses the current multiplier so you can't tell the valve to run for a very specific amount of time.

The workaround is to use the following sequence:

clear_queue_valves
queue_valve
start_from_queue

This is an acceptable alternative but may clear a queue that we don't necessarily want to clear.

nagyrobi, 2022-09-25:
@kbx81

kbx81, 2022-09-25:
Would it make sense to simply extend the `start_single_valve` action to accept a run duration when it's called? ...and then if no duration is provided, assume the already configured run_duration * multiplier?

Zebble, 2022-09-25:
> Would it make sense to simply extend the `start_single_valve` action to accept a run duration when it's called? ...and then if no duration is provided, assume the already configured run_duration * multiplier?

That would be a perfect solution!

closed 2023-02-22 by kbx81