esphome/feature-requests #1896: Support for Zemismart BCM500DS-TYW curtain motor

leoshusar, 2022-09-27:
**Describe the problem you have/What new integration you would like**
I would like to get support for my Zemismart curtain motor BCM500DS-TYW

**Please describe your use case for this integration and alternatives you've tried:**
I've tried the current `tuya/cover` component but it doesn't work 100% for my motor

**Additional context**

I've got Zemismart curtain motor BCM500DS-TYW, it comes with TYWE1S and I was able to flash ESPHome to it.
Then I tried to configure it using `tuya/cover` component, however it seems to communicate somehow differently than the already verified motor.

First thing is, it gets stuck on `init_state` 3:
```
[23:28:05][C][uart.arduino_esp8266:103]:   TX Pin: GPIO15
[23:28:05][C][uart.arduino_esp8266:104]:   RX Pin: GPIO13
[23:28:05][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[23:28:05][C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[23:28:05][C][uart.arduino_esp8266:109]:   Data Bits: 8
[23:28:05][C][uart.arduino_esp8266:110]:   Parity: NONE
[23:28:05][C][uart.arduino_esp8266:111]:   Stop bits: 1
[23:28:05][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[23:28:05][C][tuya.cover:107]: Tuya Cover:
[23:28:05][C][tuya.cover:116]:    Control has datapoint ID 102
[23:28:05][C][tuya.cover:120]:    Position has datapoint ID 101
[23:28:05][C][captive_portal:088]: Captive Portal:
[23:28:05][C][web_server:125]: Web Server:
[23:28:05][C][web_server:126]:   Address: zemismart-curtain-tywe1s.local:80
[23:28:05][C][mdns:094]: mDNS:
[23:28:05][C][mdns:095]:   Hostname: zemismart-curtain-tywe1s
[23:28:05][C][ota:089]: Over-The-Air Updates:
[23:28:05][C][ota:090]:   Address: zemismart-curtain-tywe1s.local:8266
[23:28:05][C][ota:093]:   Using Password.
[23:28:05][C][api:138]: API Server:
[23:28:05][C][api:139]:   Address: zemismart-curtain-tywe1s.local:6053
[23:28:05][C][api:141]:   Using noise encryption: YES
[23:28:05][C][tuya:033]: Tuya:
[23:28:05][C][tuya:038]:   Configuration will be reported when setup is complete. Current init_state: 3
[23:28:05][C][tuya:041]:   If no further output is received, confirm that this is a supported Tuya device.
[23:28:16][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[23:28:16][D][uart_debug:114]: <<< 55:AA:00:00:00:01:01:01
[23:28:31][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[23:28:31][D][uart_debug:114]: <<< 55:AA:00:00:00:01:01:01
```
and the heartbeat response is `55:AA:00:00:00:01:01:01` - found in the docs [here](https://developer.tuya.com/en/docs/iot/ble-basic-features?id=Kavijuzgxppvs#:~:text=The%20MCU%20returns%2055%20AA%2000%2000%2000%2001%2001%2001%20on%20recurring%20heartbeat%20communication).
It looks like they are for some reason using Bluetooth protocol for WiFi module?

But ESPHome seems to work even with incomplete init state. What could happen if I keep it running like this? Won't ESPHome complain after some time or something? Would it be possible to implement this too?

However the main problem is, my motor has swapped close and stop command.
It should be:
```
open: 0 (correct)
close: 1
stop: 2
```

I was able to find [this](https://developer.tuya.com/en/docs/iot/f?id=K9gf46o5mtfyc#:~:text=Legacy%20versions%20only-,mach_operate,-Switch) - it says legacy only, so maybe my TuyaMCU uses some kind of legacy protocol?

And here I was able to query my datapoints using Tuya API, which is great, after finding many posts about how people do it by trial / error:
```
{
  "result": {
    "category": "cl",
    "functions": [
      {
        "code": "mach_operate",
        "dp_id": 102,
        "type": "Enum",
        "values": "{\"range\":[\"ZZ\",\"FZ\",\"STOP\"]}"
      },
      {
        "code": "opposite",
        "dp_id": 103,
        "type": "Boolean",
        "values": "{}"
      },
      {
        "code": "position",
        "dp_id": 101,
        "type": "Integer",
        "values": "{\"unit\":\"%\",\"min\":0,\"max\":100,\"scale\":0,\"step\":1}"
      }
    ],
    "status": [
      {
        "code": "opposite",
        "dp_id": 103,
        "type": "Boolean",
        "values": "{}"
      },
      {
        "code": "mach_operate",
        "dp_id": 102,
        "type": "Enum",
        "values": "{\"range\":[\"ZZ\",\"FZ\",\"STOP\"]}"
      },
      {
        "code": "position",
        "dp_id": 101,
        "type": "Integer",
        "values": "{\"unit\":\"%\",\"min\":0,\"max\":100,\"scale\":0,\"step\":1}"
      }
    ]
  },
  "success": true,
  "t": 1664316569647
}
```

Let me know if you need anything else.
Thank you!

dancwilliams, 2023-08-29:
@leoshusar Did you ever get the BCM500DS-TYW flashed?

leoshusar, 2023-08-29:
@dancwilliams Yes, I made a messy workaround, but it works!

You can download my custom component and extract it in `esphome/custom_components` (the path will look like `esphome/custom_components/tuya_cover_fix`). It's just the official Tuya component and I hardcoded correct codes for this model.
[tuya_cover_fix.zip](https://github.com/esphome/feature-requests/files/12467602/tuya_cover_fix.zip)

Then your YAML will look like this:

```yaml
cover:
  - platform: tuya_cover_fix
    device_class: curtain
    name: "curtain"
    control_datapoint: 102
    position_datapoint: 101
    direction_datapoint: 103
```

The log still complains about the `init_state`:
[image]

but it doesn't block anything.

dancwilliams, 2023-08-29:
@leoshusar Awesome! Thanks!  What method did you use to flash ESPHome onto the device?

leoshusar, 2023-08-29:
@dancwilliams no problem!

Do you actually have an ESP chip in there? Tuya started swapping ESPs for chinese Beken chips, several months ago, incompatible with ESPHome and Tasmota.

If you have TYWE1S with ESP, you can follow [this](https://community.home-assistant.io/t/tuya-tywe1s-flash-help/273747/5) guide, solder few wires to it and it should just work.

If you have non-ESP module, you have two options:
1) Solder ESP instead of it
2) Try [OpenBK7231T_App](https://github.com/openshwprojects/OpenBK7231T_App) - it's a Tasmota "clone" (built from scratch I believe) for those chinese chips, but I won't tell you if and how it would work with this motor.

dancwilliams, 2023-08-29:
I have the TYWE1S so I am good to go!  Thanks for the links.

[image]