esphome/feature-requests #1899: SX1509 Allow setting PWM frequency and debounce

intermittech, 2022-09-30:
**Describe the problem you have/What new integration you would like**

 - The SX1509 allows setting the global PWM rate for all output pins, I'd like this to be a configurable value in ESPhome instead of a static value.
 - The SX1509 allows setting the global debounce value. ESPhome has implemented this but only under the "keypad" functionality, not as a general binary sensor.

**Please describe your use case for this integration and alternatives you've tried:**

 - Allowing the PWM frequency to be set will make the expander more useful in LED applications. Sometimes a 2MHz PWM rate is not desirable. The SX1509 allows you to set this frequency to various levels but the configuration value is missing from within ESPhome.
 - Allowing the debounce rate to be set as a global value means it can be changed without having a keypad configured since it also applies to binary inputs, etc.. It should be a global value instead of keypad only.

**Additional context**

Part of this request is [linked to a prior request](https://github.com/esphome/feature-requests/issues/1361) in making the SX1509 integration within ESPhome more useful to be used in hardware projects. The SX1509 is quite a capable chip but currently it's functionality can't be fully utilized within ESPhome.

SeByDocKy, 2024-07-07:
Ho yes :) would be great !!!!!!!!!!!  I am exactly on this problem try to understand why I can't control perfectly my SSR (a LSA-H3P90YB).... Via a logic analyser here some screent shot of the SX1509 generated signal:

[image]

In practice, the fout frequency is not the fOsc value.. We are around 130Hz vs (fOsc=2Mhz)

When you are looking the sx1509.cpp code, you have

``` this->read_byte(REG_CLOCK, &temp_byte);
  temp_byte |= (1 << 6);   // Internal 2MHz oscillator part 1 (set bit 6)
  temp_byte &= ~(1 << 5);  // Internal 2MHz oscillator part 2 (clear bit 5)
  this->write_byte(REG_CLOCK, temp_byte);

  this->read_byte(REG_MISC, &temp_byte);
  temp_byte &= ~(1 << 7);  // set linear mode bank B
  temp_byte &= ~(1 << 3);  // set linear mode bank A
  temp_byte |= 0x70;       // Frequency of the LED Driver clock ClkX of all IOs:
  this->write_byte(REG_MISC, temp_byte);|
````
fOSC is effectively set to 2Mhz but the foscout = fosc/2^(REG_MISC[6:4]-1). Here REG_MISC[6:4] is set to 0b111=8 via temp_byte |= 0x70;. It means that foscout = 2.10^6/(2(8-1)).   I agree with you it should be great to add the option to change this default 0x70 value....

poelzi, 2025-04-07:
I'm trying to control some PC fans which require 25khz PWM. This is currently a blocker for this solution