esphome/feature-requests #1900: Improve PWM/LEDC fading up/down routine (use a interrupt timer?)

intermittech, 2022-09-30:
**Describe the problem you have/What new integration you would like**

When changing an LED light using PWM (Either internal LEDC or external chip such as SX1509) a routine is run that uses the "default_transition_length:" value for that LED light to make this transition in LED intensity fade smoothly up and down.

When another process requires time (reading temperature sensor, updating I2C/SPI display, etc.) hiccups can be introduced into this fading. When this LED is a 5m LED strip the introduced choppiness/hiccupping is very visible and distracting.

Once fading is done LEDC itself works fine, it's purely the transitions that suffer from this issue.

**Please describe your use case for this integration and alternatives you've tried:**

Any PWM LED dimmer which wants to include additional features such as a temperature sensor, status display, etc..

**Additional context**

On a prior design ([My QuinLED-An-Penta](https://quinled.info/quinled-an-penta/)) I had to switch from the AHT21 temperature sensor to the SHT31. It seems the AHT21 integration caused these hiccups while the SHT31 integration does not! But this makes it very hard to add features to a board design when an additional feature can influence the primary function of dimming LED lights.

Talked on Discord to @kbx81 about this issue and he verified my findings. His comment was "to use a timer interrupt to refresh LED intensities during fades/transitions".

nagyrobi, 2022-09-30:
I think this qualifies more as an [problem issue](https://github.com/esphome/issues/issues), rather than a new feature request.

intermittech, 2022-10-04:
@nagyrobi Sure, I'll create a problem issue and then transport this request there and close this feature request. :)

closed 2022-10-04 by intermittech