esphome/feature-requests #1901: DSMR hardware example

Sceptorrh, 2022-09-30:
**Describe the problem you have/What new integration you would like**

A complete example of hardware that can be used with the esphome dsmr implementation

**Please describe your use case for this integration and alternatives you've tried:**

I am having difficulties finding the correct way to wire an esp8266/esp32 to the p1 port so that it can be used with esphome.
Are there any more schematics/tutorials available that are known to work with esphome?

**Additional context**

nagyrobi, 2022-09-30:
See, with ESP8266: 
[image]

Sceptorrh, 2022-09-30:
@nagyrobi Awesome, Sadly dont have to components to test it out. But looks promising. Love to hear how it goes!