esphome/feature-requests #1902: Add support for the new T-Display-S3

hapklaar, 2022-09-30:
**Describe the problem you have/What new integration you would like**
Please add support for the T-Display-S3 module, so it is recognized in ESPHome

**Please describe your use case for this integration and alternatives you've tried:**
Using these devices for HA connected energy monitors

**Additional context**
Module info: https://github.com/Xinyuan-LilyGO/T-Display-S3

hapklaar, 2022-10-02:
Was able to get this board working by connecting it to the Home Assistant host directly and flash using the following board config:

```
esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 5.0.0
```

Did not get the display working other than the backlight lit up. Probably [because it needs a special version of the TFT_eSPI library](https://github.com/Xinyuan-LilyGO/T-Display-S3#quick-question-and-answer). Is it possible to use this custom library in ESPHome somehow?

```
display:
  - platform: st7789v
    model: custom
    height: 172
    width: 320
    offset_height: 0
    offset_width: 0
    backlight_pin: GPIO38
    cs_pin: GPIO6
    dc_pin: GPIO7
    reset_pin: GPIO5
    rotation: 90°
```

randybb, 2022-10-02:
The patch is needed only for 8bit parallel mode, ESPHome supports only SPI mode, which should work without that patch. Anyway, have you powered that display? GPIO15 needs to be high.

landonr, 2022-10-02:
here's a branch with the tft_espi patched for tdisplay s3 for esphome https://github.com/landonr/lilygo-tdisplays3-esphome
and here's a branch with lvgl with esphome running on tdisplay s3 https://github.com/landonr/tdisplays3-esphome-lvgl

hapklaar, 2022-10-02:
> here's a branch with the tft_espi patched for tdisplay s3 for esphome https://github.com/landonr/lilygo-tdisplays3-esphome and here's a branch with lvgl with esphome running on tdisplay s3 https://github.com/landonr/tdisplays3-esphome-lvgl

That sounds good, will have a look later on. What would I need for the display section if I plan to run [this code](https://github.com/hapklaar/ttgo-tdisplay-watt/blob/main/ttgo-tdisplay-watt.yaml) on it?

landonr, 2022-10-02:
@hapklaar i can't access that repository. but in the meantime i'll update the lvgl sample to include some labels from home assistant

hapklaar, 2022-10-02:
> @hapklaar i can't access that repository. but in the meantime i'll update the lvgl sample to include some labels from home assistant

Oh sorry, for some reason I made it private. It's public now if you still want to have a look.

eliliam, 2022-10-17:
Amazing work on that repo! Really looking forward to support for this board and display being added to esphome as it's such an amazing device!

brooksbennett, 2022-11-06:
For those experimenting, OTA updating is working for me with the following:

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: arduino
    version: 2.0.5
    platform_version: 5.2.0

eliliam, 2022-11-15:
Hoping to see more updates here, who would be the right person to tag to get that tft_espi patch out that @landonr mentioned?

landonr, 2022-11-15:
I think this is as far as it gets with the tft_espi stuff until we get full esphome support for this display

EynErgy, 2023-02-08:
Same here, except using a patched TFT_eSPI library there is no native way to enable that display on esphome

EynErgy, 2023-02-08:
Never tried it.
I will give it a go to see

hapklaar, 2023-02-08:
> Does this display works with Arduino_GFX? when so, then i have good news i'm working on an integration of this library into esphome. For this to work there need to be some changes to the natives DisplayBuffer component. The PR is already made and is now waiting for merging into the dev branch.

It looks like it does! https://www.youtube.com/watch?v=kpRC64QNQAo

clydebarrow, 2025-03-31:
https://github.com/esphome/esphome/pull/8383

closed 2025-05-13 by jesserockz