esphome/feature-requests #1903: Provide arbitrary attributes for entities 'announced' to HA via MQTT

kquinsland, 2022-09-30:
**Describe the problem you have/What new integration you would like**

Essentially, I am asking for a simple `additional_attributes` field that can be added to light/sensor...etc entities so that the values in the `additional_attributes` field get passed into Home Assistant via the `json_attributes_topic` auto discovery mechanism.

**Please describe your use case for this integration and alternatives you've tried:**

The motivation is to build more scalable and automatic groups for use with home assistant services so it would then become possible to do something like this:

**In the esphome yaml**

```yaml
light:
  - platform: binary
    name: "Desk Lamp"
    <...>
    # Could also support a !lambda that returns a JSON document but for this example, i'm keeping it dead simple with k:v pairs as I think that is sufficient for most uses of this feature
    additional_attributes:
       - light_type: "ambient"
```

** In Home Assistant **

```yaml
  - service: light.turn_off
    data_template:
      entity_id: >-
        # This is a fragment / pseudo-code meant to illustrate the intention, not be copy/paste/run ready :)
        {%
            set non_ambient_lights=states['light] | selectattr('light_type', not_in, ['ambient']) 
        %}
        {{ non_ambient_lights }}
```

This would effectively turn off every light except those tagged with "ambient".

Right now, ESPHome does not have a way to inject arbitrary attributes for an entity so the user must "add" them with `customize.yaml` or add the entity to the group(s) e.g.: `light.ambient_lights` on the Home Assistant side of things so they can then achieve a similar effect by calling the `light.turn_off` service on `light.all_non_ambient_lights`.

This approach does not scale because the number of groups required to cover the different cases for when X should be included in the `all off` call versus when X should not be covered can grow QUICKLY.

**Additional context**

This has been on my mind for quite some time but with the beginning of "What The Heck 2022", I figured i'd take the time to document it and see if there's more interested. This post is meant to accompany this post: https://community.home-assistant.io/t/wth-cant-i-exclude-entities-from-a-service-call-when-the-target-is-set-to-entity-id-all-a-k-a-entity-tags/467416

nagyrobi, 2022-09-30:
This should be universal for all entities, and not specific to MQTT, should be supported via API also.

kquinsland, 2022-10-01:
> This should be universal for all entities, and not specific to MQTT, should be supported via API also.

Of course! I am only interested in the MQTT aspect however. On the MQTT front, HA [already has support for it](https://www.home-assistant.io/integrations/light.mqtt/#json_attributes_topic) but I can't imagine that it would be that difficult to also implement via the protobuf mechanism

RoboMagus, 2022-10-24:
This feels like a subset of FR #1635.
I'm waiting for this feature to be included a long time already and it's the only reason some of my ESP devices are not yet converted to EspHome.

kquinsland, 2022-10-25:
> This feels like a subset of FR https://github.com/esphome/feature-requests/issues/1635.

Oh good catch! Yes this is more or less an ask for the same thing but for radically different use cases. 

When I opened this request, I was not concerned with being able to modify the auto-discovery payload and re-transmit it at runtime. Having thought about it some, that's _very_ desirable because of the information that's contained in the discovery messages.

- The auto discovery payload sent to HA contains icon information. It is possible to transmit a new icon value and HA will respect that.  Being able to do something like this would be AMAZING:

```yaml
sensor:
  - id: ...
    icon: !lambda: if id(thing){ return "mdi:power"} else {return "mdi:power-alert"}
```

- Similar to the use-case in #1635 , you could conceivably have a lambda expression that would update sensor/device properties arbitrarily:

```yaml
sensor:
  - id: ...
    # Assumes that the lambda function takes an argument which is a simple dict
    disovery_attributes: !lambda: if (...) { x["isUrgent"] = true; } else { x["isUrgent"] = false;  }
``` 

Which would update the device in HA with a new attribute/property called `isUrgent`.  From there you could have a lovelace dashboard that displays entities with an attribute of `isUrgent` set to a True value.

kquinsland, 2022-10-25:
This is a request that - depending on implementation - would be superseded by #1635. Closing as dupe.

closed 2022-10-25 by kquinsland