esphome/feature-requests #1905: LD2410 24GHz Radar Presence Detector

wasyl00, 2022-10-02:
**Describe the problem you have/What new integration you would like**
I noticed very promising new presence detector module HLK-LD2410. 

**Please describe your use case for this integration and alternatives you've tried:**
I always want reliable room occupancy detection in my house. 

**Additional context**
Module is very cheap which is great:
https://www.aliexpress.com/item/1005004351593073.html?ug_edm_item_id=1005004351593073

So more info here:
https://www.youtube.com/watch?v=dAzHXpP3FcI

nagyrobi, 2022-10-02:
There's also a Bluetooth-enabled version, HLK-LD2410B, which has the advantage of being wireless => better possible placement for optimal detection.

nagyrobi, 2022-10-02:
https://community.home-assistant.io/t/mmwave-wars-one-sensor-module-to-rule-them-all/453260/2

nagyrobi, 2022-10-02:
Also: https://github.com/rain931215/ESPHome-LD2410

Doc, chinese: https://github.com/circircir-circle/human_exist_device

keniji, 2022-11-05:
> 

Just have bought LD2410B and waiting for it. The manufacture has released a tool for Android and iOS. With this tool, users can adjust/config the module directly on a mobile phone:
[https://www.youtube.com/watch?v=wgP3yod4ALI](url)
I can find the APP in iPadOS's APP Store in China region, but no luck with Google Play.

randybb, 2022-11-05:
you have to download it from their web and it is only in chinese https://community.home-assistant.io/t/mmwave-wars-one-sensor-module-to-rule-them-all/453260/40

keniji, 2022-11-05:
> you have to download it from their web and it is only in chinese https://community.home-assistant.io/t/mmwave-wars-one-sensor-module-to-rule-them-all/453260/40

Many thanks, in fact downloading the APP is not my concern 😆
Just wondering if we can communicate with LD2410B via an ESP32 through Bluetooth.

Niorix, 2022-11-22:
English docs https://drive.google.com/drive/folders/1p4dhbEJA3YubyIjIIC7wwVsSo8x29Fq-?spm=a2g2w.detail.0.1.59b64881KlgyJ4

nagyrobi, 2022-11-22:
Wonderful, including all serial protocol description. @ssieb maybe you wanna have some fun

nagyrobi, 2022-11-23:
https://github.com/esphome/esphome/pull/3919

Niorix, 2022-11-29:
Get data over Bluetooth algoritm (use gatttol):
1. gatttool -b <MAC> -I
2. connect
3. char-write-req 0x9 0100 //enable notification on 0x8 handle, but no any output
4. mtu 512
5. char-write-cmd 0x6 fdfcfbfa0800a80048694c696e6b04030201 //undocumented command a800, in HLKRadarTool.apk described as CMD_TYPE_CHECK_KEY.  48694c696e6b - HiLink in HEX. After this command LD2410 begins to send notifications.
May be release bluetooth case in new <ld2410b> platform?

regevbr, 2023-03-13:
see https://github.com/esphome/esphome/pull/4434 which you can start using right away

ssieb, 2023-10-03:
Since the sensor is supported now, I modified the title to add bluetooth since that's not supported.