esphome/feature-requests #1907: Allow me to set ESP32 esp-idf framework version to a branch

ankushg, 2022-10-04:
**Describe the problem you have/What new integration you would like**
I want to be able to set [esp32.framework.version](https://esphome.io/components/esp32.html#esp-idf-framework) to something that lets me point to a **branch** of esp-idf, instead of an official release. 

Something like the running 4.4 branch: https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/index.html

**Please describe your use case for this integration and alternatives you've tried:**
The root cause of this FR is that I want to be able to build my firmware on my raspberry pi. 

[Support for aarch64 was recently added in ESP-IDF's 4.4 branch](https://github.com/espressif/esp-idf/issues/6432), but there's no 4.4.x release that includes the fix yet.

**Additional context**

mkruiver, 2022-10-09:
I second this request, also looking for ESP32-ULP toolkit on Aarch64.

austlane, 2023-01-10:
You can also use `source` to specify a custom esp-idf framework to use.
Ex:
```yaml
esp32:
  board: lolin_s2_mini
  framework:
    type: esp-idf
    platform_version: "https://github.com/Jason2866/platform-espressif32.git#new_ULP"
    version: 4.4.3
    source: https://github.com/tasmota/esp-idf/releases/download/v4.4.3/esp-idf-v4.4.3.zip
```

nagyrobi, 2023-01-10:
Thank you @austlane .

closed 2023-01-10 by nagyrobi