esphome/feature-requests #1914: ESPHome SML library workaround for Holley DTZ541 energy meter

howeydium, 2022-10-12:
Hi all together,

please implement a workaround in the ESPHome SML library for the Holley DTZ541 energy meter, which has a faulty SML implementation.
The faulty Holley SML implementation and some workarounds are already discussed and described here:
https://github.com/volkszaehler/libsml/issues/75

Without the workaround it is not possible to read any values from the Holley energy meter with an ESP8266 according to
https://esphome.io/components/sml.html

Thank you all for all your efforts 

Best regards 

Howey

NOYB4Europe, 2023-01-15:
https://meinzaehler.com/help.html says :  - Holley DTZ541-ZDBA Achtung: Ab Version 2.0 (ab Baujahr ca. 2020) werden brauchbare Daten erst nach Freischaltung auf "Info on" bereitgestellt. Werte für Spannung, Strom und Frequenz werden ab Version 2.0 nicht mehr an der Schnittstelle ausgegeben! 

It is not working at all, because you need to press "info on" first

bjoernmayer, 2023-08-21:
Looking at the code, I'd assume that the checksum calculation with KERMIN shouldn't be an issue:
https://github.com/esphome/esphome/blob/dev/esphome/components/sml/sml.cpp#L112

It just never succeeds because of the missing bytes