esphome/feature-requests #1916: Allow ".esphome/platformio" storage location to be configured

markormesher, 2022-10-14:
**Describe the problem you have/What new integration you would like**

PlatformIO packages, platforms and cache are currently stored in `.esphome/platformio` under the config folder. When running in a container this folder is usually mounted as a volume, meaning that volume grows considerably. On my very modest setup with about a dozen devices I have a few KB of YAML configs and C++ library files, and over 500MB of data under `.esphome/platformio`.

This is especially problematic when volumes are backed by block storage devices because backups operate at the block level, meaning we have to store over 500MB to backup a few KB of YAML.

**Please describe your use case for this integration and alternatives you've tried:**

I've already changed `esphome.build_path` to move build files to ephemeral storage which redirected a few 10s of MB out of the volume, but that's <10% of the storage usage picture.

**Additional context**

sprat, 2023-01-27:
I also have the same problem, it would be nice to allow the configuration of the cache directory. With the docker install, we can create a mountpoint in /config/.esphome, but the mount directory still appear in the config directory which is a bit annoying (owned by root...)

sprat, 2023-01-27:
It seems to be possible to configure the path with environment variables: see https://github.com/esphome/esphome/blob/dev/docker/docker_entrypoint.sh