esphome/feature-requests #1918: keyboard emulation

tomaszduda23, 2022-10-16:
**Describe the problem you have/What new integration you would like**

Use esphome as USB HID keyboard.

**Please describe your use case for this integration and alternatives you've tried:**

Esphome is used to control multimedia center PC (power on/off, reset, leds status). Everything is done by network. From time to time there is an issue and keyboard needs to be connected e.g. to change bios settings. It would be nice if esphome could work as the keyboard. esp32-s2 with https://github.com/adafruit/Adafruit_TinyUSB_Arduino works quite good (I made proof of concept). I wonder if you would consider to add this to esphome.

**Additional context**

There is a few projects which emulates keyboard already. They seems to duplicate part of functionality.
https://github.com/jensweimann/esphome_ducky
https://github.com/dmamontov/esphome-blekeyboard
so the goal would be:
1. Create keyboard component which can handle most of logic
2. Create keyboard driver for each protocol: USB, BLE, etc.
3. Add keyboard to web interface

nagyrobi, 2022-10-16:
What a great idea!

tomaszduda23, 2022-10-28:
https://github.com/esphome/esphome/pull/3917#issuecomment-1295301150

First working version on firefox.
[image]

RoboMagus, 2022-10-28:
Looks neat!

But wouldn't devices that you use to access the ESPhome webserver have a keyboard that could be used, instead of an on screen keyboard? 

That way the addition to the webserver would probably be much smaller and keep it more suitable for including it into the binary without it consuming too much flash on the esp devices.

tomaszduda23, 2022-10-28:
> But wouldn't devices that you use to access the ESPhome webserver have a keyboard that could be used, instead of an on screen keyboard?

You can control it either way:
a) on screen keyboard
b) press key on physical keyboard

> That way the addition to the webserver would probably be much smaller and keep it more suitable for including it into the binary without it consuming too much flash on the esp devices.

By default esphome gets web UI online so size does not matter much. For 'local' I was thinking to make it configurable. For now it takes `Flash: [=====     ]  49.3%` with on screen keyboard.

tomaszduda23, 2022-11-20:
https://github.com/tomaszduda23/esphome/pull/1 BLE keyboard. It needs base keyboard to be integrated first.