esphome/feature-requests #1919: Templatable `icon` setting

nagyrobi, 2022-10-17:
**Describe the problem you have/What new integration you would like**
Allow (at least for template sensors) to use the icon templatable: For example:
```yaml
  icon: !lambda: |-
      if (x == 1) {
        return {"mdi:lightbulb-on"};
      } else {
        return {"mdi:lightbulb"};
      }
```
or:
```yaml
  icon: !lambda: |-
      if (id(pump_status).state == 1 and id(valve_status).state == 0) {
        return {"mdi:hand-water"};
      } else if (id(pump_status).state == 1 and id(valve_status).state == 1) {
        return {"mdi:hydro-power"};
      } else if (id(pump_status).state == 0 and id(valve_status).state == 0) {
        return {"mdi:radiator-off"};
      } else {
        return {"mdi:alert"};
      }

```

**Please describe your use case for this integration and alternatives you've tried:**
Home Assistant supports `icon_template` or `icon` configuration option for their template sensors, which can set different icons based on jinja templates - referring to states of self, or other entities from the system.

This could be beneficial in ESPHome too.

**Additional context**
https://www.home-assistant.io/integrations/switch.template/
https://www.home-assistant.io/integrations/light.template/
https://www.home-assistant.io/integrations/template/
etc.

Shadewing666, 2022-10-26:
I'd like to add my support to this suggestion.

My own use case that ended up with me finding this is a template binary sensor for the state of an automated pump:
```
  icon: !lambda: |-
      if (id(relay).state) {
        return {"mdi:pump"};
      } else {
        return {"mdi:pump-off"};
      }
note: relay is an internal switch, not presented to Home Assistant
```

With this, all device configuration is contained within ESPHome.

The current solution is to create another template sensor within Home Assistant, containing the icon template, to mirror the devices sensor. So we end up with 2 sensors in HA with the extra steps required of adding a template sensor and hiding the devices sensor. One thing this does not address is that the new template sensor is not associated with/part of the device within HA.

ssieb, 2022-10-26:
This will require protocol changes and changes to the HA integration.

nagyrobi, 2022-10-26:
Yes