esphome/feature-requests #1920: BedJet - shutoff time support

javawizard, 2022-10-18:
**Describe the problem you have/What new integration you would like**
The BedJet remote control and app allow setting an amount of time before the unit shuts itself off. This defaults to half an hour in heat mode and 10 hours in cool mode. I'd like to be able to set that via the ESPHome BedJet integration as well.

**Please describe your use case for this integration and alternatives you've tried:**
Right now when I turn my BedJet on to heat mode it automatically shuts off after half an hour. I'd like to be able to have it stay on for longer periods than that.

I can sort of work around this by setting up an automation that runs once every half hour plus a few seconds that turns the BedJet back on, but that's a pain.

**Additional context**

javawizard, 2022-10-18:
@jhansche - I edited your name into the issue description so not sure if it pinged you.

Thanks again for what has proven to be one of my all-time favorite ESPHome/HA integrations 😄

jhansche, 2022-10-18:
Have you tried using the extended heat mode? By default it will be using the regular heat mode (30 minutes), but if you configure it to use extended heat mode, then setting the climate device to heat will cause it to select the bedjet's EXT HT mode instead.

See https://esphome.io/components/climate/bedjet.html#id1
Just add `heat_mode: extended` to your yaml config. You can also use the EXT HT custom preset.

It's not exactly what you asked for, but it addresses what appears to be your primary concern. Then you can use an automation to make sure it's turned off if you need it turned off sooner than 10hrs (mine shuts off when I dismiss my alarm). Another option is to configure the Biorhythm or memory presets, and change the way you start the BedJet to set the custom preset to one of the M1-M3 preset buttons.

---

As for adjusting the runtime, there are two options that can be used today:

1. Something like a template number, using lambdas to get/set the remaining runtime. This was added as part of the "hub" refactoring in #3522: https://github.com/esphome/esphome/pull/3522/files#diff-a5b4d13b378ba30021c2ddf35493c7602db3aaf0e9184e988367fd0d8fa27384R83
2. Use the `send_button` lambda with a template button, to recreate button entities in HA that will send the "TIME +/-" buttons: https://github.com/esphome/esphome/pull/3522/files#diff-a5b4d13b378ba30021c2ddf35493c7602db3aaf0e9184e988367fd0d8fa27384R62

The intent of the "hub" change was so that we can add things like this as built-in optional platform components, such as what I added for the `fan` domain. In this case it would be a `number:platform=bedjet` or `button:platform=bedjet`. I just haven't had much time to work on those myself.

There are other examples in the BedJet component docs for how you can use lambdas in your yaml config in order to work with the component without having full official support built-in: https://esphome.io/components/climate/bedjet.html#lambda-calls
You could use the firmware or clock button examples and adapt those for the send_button lambda, to increment the runtime by one minute, for example. But it's probably more convenient to use a template number for this

jhansche, 2022-10-18:
More info on how you can setup a template number:
https://esphome.io/components/number/template.html

You would use `on_value` automation to call `set_time_remaining(hr,min)`. Assuming the number should represent seconds, then hr=x/3600, and min=(x%3600)/60. And use the `lambda` config to update the value using `get_time_remaining()`.

Using a number might be tricky though, because the BedJet reports the runtime remaining as seconds, but it only allows setting the value by the minute. So even though it will countdown the second, as soon as you start dragging the number slider (which represents seconds), we won't be able to reconcile those, so the slider will snap to the nearest minute instead. To account for that it might be better to consider the number entity to represent minutes instead of seconds. So the lambda should divide the number by 60. Then it will look more natural and accurate when you're selecting a number on the slider, at the expense of losing the per-second countdown.