esphome/feature-requests #1924: RP2040 Support

jesserockz, 2022-10-20:
**Describe the problem you have/What new integration you would like**

Mainstream support for RP2040

Here is a list of things that have not be written or tested:

- [ ] preferences
- [ ] mDNS
- [ ] I²C
- [ ] SPI
- [ ] UART
- [ ] Improv serial
- [ ] one-wire/dallas

Please leave a comment below if there are more things to add to this list, or you have a PR that solves something for the RP2040

bertvdijk, 2022-10-26:
Thanks for working on support for RP2040 in esphome!

I understand this is probably a bit early to ask, but what's holding I2C back? I'd love to be able to attach a BME280 sensor, which requires I2C to be defined.
```
# file: rp2040.yaml
...
i2c:
  - id: i2c1
    sda: GPIO4
    scl: GPIO5
    scan: true
```
currently results in:
```
Compiling .pioenvs/picobme280/src/esphome/components/i2c/i2c_bus_arduino.cpp.o
...
src/esphome/components/i2c/i2c_bus_arduino.cpp: In member function 'virtual void esphome::i2c::ArduinoI2CBus::setup()':
src/esphome/components/i2c/i2c_bus_arduino.cpp:29:70: error: no matching function for call to 'TwoWire::begin(int, int)'
   29 |   wire_->begin(static_cast<int>(sda_pin_), static_cast<int>(scl_pin_));
      |                                                                      ^
In file included from src/esphome/components/i2c/i2c_bus_arduino.h:7,
                 from src/esphome/components/i2c/i2c_bus_arduino.cpp:3:
/Users/b/.platformio/packages/framework-arduinopico/libraries/Wire/src/Wire.h:42:10: note: candidate: 'virtual void TwoWire::begin()'
   42 |     void begin() override;
      |          ^~~~~
/Users/b/.platformio/packages/framework-arduinopico/libraries/Wire/src/Wire.h:42:10: note:   candidate expects 0 arguments, 2 provided
/Users/b/.platformio/packages/framework-arduinopico/libraries/Wire/src/Wire.h:44:10: note: candidate: 'virtual void TwoWire::begin(uint8_t)'
   44 |     void begin(uint8_t address) override;
      |          ^~~~~
/Users/b/.platformio/packages/framework-arduinopico/libraries/Wire/src/Wire.h:44:10: note:   candidate expects 1 argument, 2 provided
```

geerlingguy, 2022-10-27:
When trying to compile for a Pico W (`rpipicow`) without any `wifi` configured, I run into this compile error:

```
$ esphome run led-blink.yml --device /Volumes/RPI-RP2             
INFO Reading configuration led-blink.yml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing rpi-pico (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing git+https://github.com/maxgerhardt/platform-raspberrypi.git
INFO Installing git+https://github.com/maxgerhardt/platform-raspberrypi.git
git version 2.37.3
Cloning into '/Users/jgeerling/.platformio/.cache/tmp/pkg-installing-cw_7rp8y'...
remote: Enumerating objects: 96, done.
remote: Counting objects: 100% (96/96), done.
remote: Compressing objects: 100% (58/58), done.
remote: Total 96 (delta 46), reused 56 (delta 25), pack-reused 0
Receiving objects: 100% (96/96), 130.96 KiB | 2.01 MiB/s, done.
Resolving deltas: 100% (46/46), done.
Submodule 'builder/frameworks/arduino/mbed-core' (https://github.com/platformio/builder-framework-arduino-core-mbed.git) registered for path 'builder/frameworks/arduino/mbed-core'
Cloning into '/Users/jgeerling/.platformio/.cache/tmp/pkg-installing-cw_7rp8y/builder/frameworks/arduino/mbed-core'...
remote: Enumerating objects: 31, done.        
remote: Counting objects: 100% (31/31), done.        
remote: Compressing objects: 100% (23/23), done.        
remote: Total 31 (delta 8), reused 24 (delta 7), pack-reused 0        
Receiving objects: 100% (31/31), 10.79 KiB | 10.79 MiB/s, done.
Resolving deltas: 100% (8/8), done.
Submodule path 'builder/frameworks/arduino/mbed-core': checked out 'ed9a3458e336660fdf2b1d4a4c828a2ddfd214ad'
Platform Manager: raspberrypi@1.7.0+sha.702b36f has been installed!
INFO raspberrypi@1.7.0+sha.702b36f has been installed!
Tool Manager: Installing earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.220714
INFO Installing earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.220714
Unpacking  [####################################]  100%          
Tool Manager: toolchain-rp2040-earlephilhower@5.100300.220714 has been installed!
INFO toolchain-rp2040-earlephilhower@5.100300.220714 has been installed!
Tool Manager: Installing https://github.com/earlephilhower/arduino-pico/releases/download/2.6.2/rp2040-2.6.2.zip
INFO Installing https://github.com/earlephilhower/arduino-pico/releases/download/2.6.2/rp2040-2.6.2.zip
Unpacking  [####################################]  100%
Tool Manager: framework-arduinopico@1.20602.0 has been installed!
INFO framework-arduinopico@1.20602.0 has been installed!
Tool Manager: Installing platformio/tool-rp2040tools @ ~1.0.2
INFO Installing platformio/tool-rp2040tools @ ~1.0.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-rp2040tools@1.0.2 has been installed!
INFO tool-rp2040tools@1.0.2 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40400.0
INFO Installing platformio/tool-scons @ ~4.40400.0
Unpacking  [####################################]  100%
Tool Manager: tool-scons@4.40400.0 has been installed!
INFO tool-scons@4.40400.0 has been installed!
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.20602.0 (2.6.2) 
 - tool-rp2040tools @ 1.0.2 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 1.0.0
|   |-- Updater @ 1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|   |   |-- PicoOTA @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |-- MD5Builder @ 1.0.0
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
[... 26 lines cut ...]
src/esphome/core/helpers.cpp: In function 'uint32_t esphome::random_uint32()':
src/esphome/core/helpers.cpp:101:15: error: 'rosc_hw' was not declared in this scope
  101 |     result |= rosc_hw->randombit;
      |               ^~~~~~~
src/esphome/core/helpers.cpp: In function 'bool esphome::random_bytes(uint8_t*, size_t)':
src/esphome/core/helpers.cpp:120:17: error: 'rosc_hw' was not declared in this scope
  120 |       result |= rosc_hw->randombit;
      |                 ^~~~~~~
Compiling .pioenvs/rpi-pico/liba89/MD5Builder/MD5Builder.cpp.o
src/esphome/core/helpers.cpp: In constructor 'esphome::InterruptLock::InterruptLock()':
src/esphome/core/helpers.cpp:403:53: error: 'save_and_disable_interrupts' was not declared in this scope
  403 | IRAM_ATTR InterruptLock::InterruptLock() { state_ = save_and_disable_interrupts(); }
      |                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/helpers.cpp: In destructor 'esphome::InterruptLock::~InterruptLock()':
src/esphome/core/helpers.cpp:404:45: error: 'restore_interrupts' was not declared in this scope
  404 | IRAM_ATTR InterruptLock::~InterruptLock() { restore_interrupts(state_); }
      |                                             ^~~~~~~~~~~~~~~~~~
Compiling .pioenvs/rpi-pico/lib8c4/LittleFS/LittleFS.cpp.o
Compiling .pioenvs/rpi-pico/lib8c4/LittleFS/lfs.c.o
Compiling .pioenvs/rpi-pico/lib8c4/LittleFS/lfs_util.c.o
*** [.pioenvs/rpi-pico/src/esphome/core/helpers.cpp.o] Error 1
============================================= [FAILED] Took 14.06 seconds =============================================
```

If I add a valid `wifi` configuration, it compiles. There are use cases where I'd like to compile for the Pico W without WiFi enabled, so it would be nice if that works too.

geerlingguy, 2022-10-27:
I also wanted to control the onboard LED on the Pico W.

On the regular Pico, I can do so with GPIO 25:

```yaml
output:
  # For Raspberry Pi Pico (non-W)
  - platform: gpio
    pin:
      number: 25
      mode: output
    id: LED
```

Is there any way to control `WL_GPIO0` on the Pico W? It is routed through the WiFi chip, unfortunately.

geerlingguy, 2022-10-27:
The LED now works with:

```yaml
output:
  # For Raspberry Pi Pico W
  - platform: gpio
    pin:
      number: 32  # 25 for Pico (non-W)
      mode: output
    id: LED
```

But for some reason whenever I try adding WiFi config, e.g.

```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
```

(With or without fast_connect, and with valid ssid/password stored in `secrets.yaml`)

...it results in a crash — with `VERY_VERBOSE` logging via serial port (`esphome run led-blink.yml --device /dev/tty.usbmodem2101`) all I see is the following before it seems like it crashes:

```
INFO Successfully uploaded program.
INFO Starting log output from /dev/tty.usbmodem2101 with baud rate 115200
[16:41:49][C][wifi:038]:   Local MAC: 28:CD:C1:01:42:16
[16:41:49][I][wifi:260]: WiFi Connecting to 'geerling'...
[16:41:49][V][wifi:262]: Connection Params:
[16:41:49][V][wifi:263]:   SSID: 'geerling'
[16:41:49][V][wifi:268]:   BSSID: Not Set
[16:41:49][V][wifi:286]:   Password: '[redacted]'
[16:41:49][V][wifi:293]:   Channel: Not Set
[16:41:49][V][wifi:300]:   Using DHCP IP
[16:41:49][V][wifi:302]:   Hidden: NO
ERROR Serial port closed!
```

And if I monitor the serial port `/dev/tty.usbmodem2101` in CoolTerm (or minicom), I see it just gets in some sort of boot loop, where it tries to connect for a bit, then serial disconnects, then it tries again, then disconnects...

jesserockz, 2022-10-27:
@bertvdijk Thanks for testing this.

I have made a [PR](https://github.com/esphome/esphome/pull/3959) today to fix I2C due to API differences (So much for Arduino frameworks/libraries having a common API)

```yaml
external_components:
  - source: github://pr#3959
    components: [i2c, rp2040]
```

I have tested this code with my i2c pn532 and it is reading nfc tags correctly.

Jesse

geerlingguy, 2022-10-28:
@jesserockz @balloob — Thanks for the help over the past day; I am still having the WiFi issue, and I've tried on two different Pico W's, and two different ESP32s (all brand new); the ESP32s connected without issue, with the same configuration, whereas the Pico W's did not (they did the connect/boot loop).

I also tested on two different WiFi networks—one an ASUS router the other a Netgear, both 2.4 GHz networks that I have other ESP and Pi/Pico devices on. And both networks also worked fine when I flashed the MicroPython examples within this repo—I could access the webserver running on the Pico W from my Mac on the same network.

I also tested with two different USB cables and on two different USB ports on my Mac, and also separately on a fresh install of Raspberry Pi OS directly on my Raspberry Pi 4.

In all cases, it keeps doing that WiFi connect/boot loop :(

So I've created a set of reproducible steps using the official Docker image. Can you please try the same thing and triple-check that it works on your end. If it does, I think I'm just cursed or something:

  1. `git clone https://github.com/geerlingguy/pico-w-garage-door-sensor.git`
  2. `cd pico-w-garage-door-sensor`
  3. `docker-compose up -d`
  4. `docker exec -it esphome bash`
  5. `esphome compile test-picow-wifi.yml`
  6. `cp .esphome/build/rpi-pico/.pioenvs/rpi-pico/firmware.uf2 ./rpi-pico.uf2`
  7. `exit` (to exit the Docker build environment)
  8. Plug in the Pico W to your computer while holding down BOOTSEL to enter BOOTSEL mode
  9. Copy the `rpi-pico.uf2` file over to the Pico W's filesystem (on my Mac, the disk labeled `RPI-RP2`)

Then monitor serial output and see if the Pico W connects or just gets stuck in a boot loop (on my Mac, I used CoolTerm and watched `usbmodem2101`, though the device will be different depending on what Mac you have and what USB port you're plugged into).

peter-mount, 2022-10-28:
@geerlingguy I've just tried this with a virgin Pico-W captured with Minicom under Linux.

[image]

Once it gets to this point it then just sits there but it is connecting fine, no loops & I can ping the pico.

geerlingguy, 2022-10-28:
@peter-mount - Thanks so much! So the exact same setup works for you — can you share what type of WiFi router you have?

I also heard from @mietz8 on Twitter (https://twitter.com/mietz8/status/1586053083594985473) that he's getting a boot loop.

peter-mount, 2022-10-28:
@geerlingguy Linksys Velop AC3900 Mesh WiFi (dual band) with 3 nodes in the mesh.
Core Router (handling dhcp) is a MikroTik rb3011uias-rm 1U

mietzen, 2022-10-28:
```
[C][wifi:038]:   Local MAC: 1234567890abcde
[I][wifi:260]: WiFi Connecting to 'my-wifi'....
[V][wifi:262]: Connection Params:.
[V][wifi:263]:   SSID: 'my-wifi'.
[V][wifi:268]:   BSSID: Not Set.
[V][wifi:286]:   Password: .xxxxxxxxxxxxxxxxxxxxxx
[V][wifi:293]:   Channel: Not Set.
[V][wifi:300]:   Using DHCP IP.
[V][wifi:302]:   Hidden: NO.
[C][wifi:038]:   Local MAC: 1234567890abcde
[I][wifi:260]: WiFi Connecting to 'my-wifi'....
[V][wifi:262]: Connection Params:.
[V][wifi:263]:   SSID: 'my-wifi'.
[V][wifi:268]:   BSSID: Not Set.
[V][wifi:286]:   Password: .xxxxxxxxxxxxxxxxxxxxxx
[V][wifi:293]:   Channel: Not Set.
[V][wifi:300]:   Using DHCP IP.
[V][wifi:302]:   Hidden: NO.
[C][wifi:038]:   Local MAC: 1234567890abcde
[I][wifi:260]: WiFi Connecting to 'my-wifi'....
[V][wifi:262]: Connection Params:.
[V][wifi:263]:   SSID: 'my-wifi'.
[V][wifi:268]:   BSSID: Not Set.
[V][wifi:286]:   Password: .xxxxxxxxxxxxxxxxxxxxxx
[V][wifi:293]:   Channel: Not Set.
[V][wifi:300]:   Using DHCP IP.
[V][wifi:302]:   Hidden: NO.
```

@geerlingguy 

Tested on fresh new PicoW, fresh copy of µPython is running fine.
Hardware Build: MacBookPro 14" M1Pro 
Router: Fritz!Box 7490

axipher, 2022-10-28:
@geerlingguy 

I was able to compile the UF2 on a remote Ubuntu box that has docker on it, and copy the UF2 on to my Pico W using my Windows 10 box.

Power cycling the device and quickly connecting to it via Putty on COM8 and I managed to get at least this last bit of the output.  Putty won't keep the window open to capture everything form initial boot so if anyone knows of a good Windows utility I can use to monitor Serial COM ports through disconnects, I can try something other than Putty.

```
[I][wifi:521]: WiFi Connected!
[C][wifi:365]:   Local MAC: XX:XX:XX:XX:XX:XX
[C][wifi:366]:   SSID: 'XXXXXXXXXXXX'
[C][wifi:367]:   IP Address: 10.2.1.140
[C][wifi:368]:   BSSID: EE:EE:EE:EE:EE:EE
[C][wifi:370]:   Hostname: 'rpi-pico'
[C][wifi:372]:   Signal strength: 0 dB ▂▄▆█
[C][wifi:376]:   Channel: 0
[C][wifi:377]:   Subnet: 255.255.255.0
[C][wifi:378]:   Gateway: 10.2.1.1
[C][wifi:379]:   DNS1: 10.2.1.236
[C][wifi:380]:   DNS2: 8.8.8.8
[I][app:062]: setup() finished successfully!
[I][app:102]: ESPHome version 2022.11.0-dev compiled on Oct 28 2022, 14:04:45
[C][wifi:507]: WiFi:
[C][wifi:365]:   Local MAC: XX:XX:XX:XX:XX:XX
[C][wifi:366]:   SSID: 'XXXXXXXXXXXX'
[C][wifi:367]:   IP Address: 10.1.1.140
[C][wifi:368]:   BSSID: EE:EE:EE:EE:EE:EE
[C][wifi:370]:   Hostname: 'rpi-pico'
[C][wifi:372]:   Signal strength: 0 dB ▂▄▆█
[C][wifi:376]:   Channel: 0
[C][wifi:377]:   Subnet: 255.255.255.0
[C][wifi:378]:   Gateway: 10.1.1.1
[C][wifi:379]:   DNS1: 10.1.1.236
[C][wifi:380]:   DNS2: 8.8.8.8
[C][logger:277]: Logger:
[C][logger:278]:   Level: VERY_VERBOSE
[C][logger:279]:   Log Baud Rate: 115200
[C][logger:280]:   Hardware UART: UART0
[C][mdns:103]: mDNS:
[C][mdns:104]:   Hostname: rpi-pico
[V][mdns:105]:   Services:
[V][mdns:107]:   - _http, _tcp, 80
[V][mdns:109]:     TXT: version = 2022.11.0-dev

```

As for my router, using a Unifi UDM-Pro + UAP-FlexHD with I believe the standard settings for the 2.4 GHz network I have the Pico W connecting to.

geerlingguy, 2022-10-28:
Okay, adding more info here since I've been up to my neck in debugging:

| WiFi Router / Vendor | Network | Encryption | Works (esphome) | Works (MicroPython) |
| --- | --- | --- | --- | --- |
| ASUS RT-AX86U | 2.4 GHz (ax / WiFi 6 mode) | WPA2-Personal | No | Yes |
| ASUS RT-AX86U | 2.4 GHz (legacy mode) | WPA2-Personal | Yes | Yes |
| Netgear WAX620 | 2.4 GHz (ax mode) | WPA3-Personal | No | Yes |
| RaspAP (Pi 4 model B) | 2.4 GHz (802.11g mode) | WPA2 | Yes | Yes |

Tested with fast_connect both enabled and disabled.

So it looks like any newer WiFi 6 routers that default to setting up their 2.4 GHz networks in ax / WiFi 6 mode might have an issue here (strangely, they don't have an issue under MicroPython on the same Pico W...).

I had to manually set my ASUS WiFi 6 router (RT-AX86U) into Legacy mode to get esphome on the Pico W to connect and not do the reboot/crash loop:

[image]

I did not have to set it into the Legacy mode for either the Pico W with MicroPython or the ESP32 with the same esphome configuration.

axipher, 2022-10-28:
Is the WiFi library being used by ESPHome missing some WiFi connection settings or needs some additional checks for newer 2.4 GHz networks.

Maybe the MicroPython network library has some newer WiFi code that can correctly handle newer WiFi networks.

emersonsc, 2022-10-30:
Is there any documentation on how to get ESPHome on the Pico W? I'd love to use the stack of Pico Ws i have instead of the chinese D1s to make temp sensors and such for the house.

vidosits, 2022-10-30:
> Is there any documentation on how to get ESPHome on the Pico W? I'd love to use the stack of Pico Ws i have instead of the chinese D1s to make temp sensors and such for the house.

I guess the same way as you get ESPHome on the ESP chips. I, personally, as a first time user of esphome, just launched a dashboard and followed the (very) straightforward GUI. After using the GUI 1-2 times (first trying to blink a LED, then trying a vl53l1x w/ i2c) I guess once you have a config down you can just the CLI too, again it's all straightforward.

For the GUI you can do 2 things:
- [Go to the hosted dashboard](https://web.esphome.io/?dashboard_wizard), but you'll need Google Chrome or Microsoft Edge
- Self-host the dashboard

I've self-hosted the dashboard, because I'm using Firefox.

For the self-hosted dashboard you have 3 options:
- Use `docker`
- Use `docker-compose`
- Use `python3/pip3`

I used Jeff Geerling's (Thanks Jeff!) `docker-compose.yml` with a little modification so that it launches the WebGUI dashboard:

```yaml
---
version: '3'
services:
  esphome:
    container_name: esphome
    image: esphome/esphome:dev
    command: dashboard /configs --port 8080
    volumes:
      - .configs/:/configs
      - /etc/localtime:/etc/localtime:ro
    restart: unless-stopped
    privileged: true
    network_mode: host
```

The project dir looked like this for me and my working directory at the time of launching was `esphome-picow`:

```
esphome-picow
├── configs
│   ├── pico-pi-w-led.yaml
│   └── pico-pi-w-tof-sensor.yaml
└── docker-compose.yml
```

Launching a dashboard is done by `docker-compose up -d` in the project directory, after that you can visit `localhost:8080` in your browser and you should see all the configs set up in the config directory.

If you're using `docker`, but not `docker-compose` you can use the following command from your working directory:
`docker run -it --restart unless-stopped -v $(pwd)/configs:/configs -v /etc/localtime:/etc/localtime:ro --privileged --network host --name esphome-dashboard esphome/esphome:dev dashboard /configs --port 8080`

If you're using `python3 and pip(3)` install `esphome` with `pip install git+https://github.com/esphome/esphome` or `pip3 install git+https://github.com/esphome/esphome`, depending on whether python2 or python3 is your default.

The above `docker/docker-compose` commands will create a `configs` directory in your working directory. Change the port to something else if you already have some program binding `localhost:8080`.

If you've used `python3/pip` to install you can launch the dashboard with:

`esphome dashboard $(pwd)/configs --port 8080 --open-ui`. This will open the UI in your browser automatically.

If you've used `docker/docker-compose` to launch the UI you should go to `http://localhost:8080` (or whatever port you've used).

After you've opened the dashboard you'll see a screen like this:

[image]

Click on the `+ New device` button and click on `Continue` if you're running the self-hosted dashboard version.
Don't worry you'll be able to change these later under the `Secrets` option in the top right of the screen.

[image]

You'll see the "Create configuration" screen, which you should fill out with a name that will be the configs name and the credentials of the WiFi network your Pico W will connect to.

Click `Next` and you'll see a list of supported boards, select `Raspberry Pi Pico W` and click on `Next`.
[... 29 lines cut ...]
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: **********************-

ota:
  password: ************************-

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Pico-Pi-W-Led Fallback Hotspot"
    password: **********-
```

Either way you get to a point where you have a `.uf2` file on your computer. Now, take the micro USB cable and connect your Pico Pi W to your computer **(and this is very important) while holding the BOOTSEL button**. This will make the Pico Pi W flashable and it should show up as a regular mass storage device on your computer. Copy your `.uf2` file to the device, once it's done copying it will restart and you should be able to get the logs from it through the esphome dashboard. All further upgrading can be done from `esphome` and the wireless OTA updating system. Hope this helped!

emersonsc, 2022-10-30:
> > Is there any documentation on how to get ESPHome on the Pico W? I'd love to use the stack of Pico Ws i have instead of the chinese D1s to make temp sensors and such for the house.
> 
> I guess the same way as you get ESPHome on the ESP chips. I, personally, as a first time user of esphome, just launched a dashboard and followed the (very) straightforward GUI. After using the GUI 1-2 times (first trying to blink a LED, then trying a vl53l1x w/ i2c) I guess once you have a config down you can just the use the CLI too, again it's all straightforward.
> 
> For the GUI you can do 2 things:
> 
> * [Go to the hosted dashboard](https://web.esphome.io/?dashboard_wizard), but you'll need Google Chrome or Microsoft Edge

The hosted dashboard only gives you the option for ESP32 & ESP8266 devices. There is ZERO option for the RP2040.

Using the python 3 method again gave me no option for the RP2040. When i typed in RP2040 or RPIPICOW in the wizard prompt it didnt recognize either and wouldnt go further until i selected a esp32 or esp8266. Yes it was the most recent version  suing the instructions right from ESPHomes install ESPHome manually page.

I guess im going to have to figure out how to use docker-compose and see if I can get that to work for me. Hoping it works for windows 11

vidosits, 2022-10-30:
> The hosted dashboard only gives you the option for ESP32 & ESP8266 devices. There is ZERO option for the RP2040.
>
> Using the python 3 method again gave me no option for the RP2040. When i typed in [...]

Because the hosted dashboard and `pip install esphome` both give you the latest stable version released onto [PyPI](https://pypi.org/project/esphome/) (version 2022. 10. 1). This is *not* new enough for these features. You need 2022. 11. 0-dev. 

You can get that at least in 3 different ways, some of them I've outlined in my previous comment. If you have your Python set up (sounds like it), you don't have to muck around with `docker` (especially since Docker on Windows is uh, less than great sometimes). Just do `pip install git+https://github.com/esphome/esphome`. After that, verify your version with `esphome version`, it should print out `Version: 2022.11.0-dev`. After that you can do `esphome dashboard $(pwd)/configs --port 8080 --open-ui`. All the `$PATH` environment variable setup and other stuff you can follow from the official install guide.

balloob, 2022-10-31:
This is not a place to discuss how to run Pico. I thought I had made this clear by marking the original issue as off-topic. Stop posting about this.

This issue is to track development and that's what all replies need to be about.

geerlingguy, 2022-11-07:
@brunsy - That should probably be elsewhere. Right now, at least, bluetooth support (though supposedly supported at the hardware level) is not even an official feature of the Pico W.

bertvdijk, 2022-11-12:
Not sure if this is the correct place to report missing functionality but I've come across the following issues while attempting to replace Espressif devices with Raspberry Pico W.

It's currently not possible to use the _Time_ component (platforms: sntp, homeassistant) in ESPhome with RP2040.

```
# file: picodebug.yaml
time:
  - platform: sntp
```
and:
```
# file: picodebug.yaml
time:
  - platform: homeassistant
    id: homeassistant_time
```

- it depends on ESP specific code in libraries

First example (time platform sntp):

```
Compiling /tmp/.esphome/.pioenvs/picodebug/src/esphome/components/time/real_time_clock.cpp.o
src/esphome/components/sntp/sntp_component.cpp: In member function 'virtual void esphome::sntp::SNTPComponent::setup()':
src/esphome/components/sntp/sntp_component.cpp:36:3: error: 'sntp_setservername' was not declared in this scope
   36 |   sntp_setservername(0, strdup(this->server_1_.c_str()));
      |   ^~~~~~~~~~~~~~~~~~
src/esphome/components/sntp/sntp_component.cpp:47:3: error: 'sntp_init' was not declared in this scope
   47 |   sntp_init();
      |   ^~~~~~~~~
src/esphome/components/sntp/sntp_component.cpp: In member function 'virtual void esphome::sntp::SNTPComponent::update()':
src/esphome/components/sntp/sntp_component.cpp:59:7: error: 'sntp_enabled' was not declared in this scope
   59 |   if (sntp_enabled()) {
      |       ^~~~~~~~~~~~
src/esphome/components/sntp/sntp_component.cpp:60:5: error: 'sntp_stop' was not declared in this scope
   60 |     sntp_stop();
      |     ^~~~~~~~~
src/esphome/components/sntp/sntp_component.cpp:62:5: error: 'sntp_init' was not declared in this scope
   62 |     sntp_init();
      |     ^~~~~~~~~
```

- see https://github.com/esphome/esphome/blob/dev/esphome/components/sntp/sntp_component.cpp#L4-L12

Second example (time platform homeassistant):

```
Compiling /tmp/.esphome/.pioenvs/picodebug/src/esphome/components/time/real_time_clock.cpp.o
src/esphome/components/time/real_time_clock.cpp: In member function 'void esphome::time::RealTimeClock::synchronize_epoch_(uint32_t)':
src/esphome/components/time/real_time_clock.cpp:25:3: error: 'timezone' was not declared in this scope; did you mean 'timezone_'?
   25 |   timezone tz = {0, 0};
      |   ^~~~~~~~
      |   timezone_
src/esphome/components/time/real_time_clock.cpp:26:35: error: 'tz' was not declared in this scope; did you mean 'tm'?
   26 |   int ret = settimeofday(&timev, &tz);
      |                                   ^~
      |                                   tm
src/esphome/components/time/real_time_clock.cpp:26:13: error: 'settimeofday' was not declared in this scope
   26 |   int ret = settimeofday(&timev, &tz);
      |             ^~~~~~~~~~~~
```

- see https://github.com/esphome/esphome/blob/dev/esphome/components//time/real_time_clock.cpp#L25-L26

It's not possible to use the MQTT in ESPHome with RP2040.

```
# file: picodebug.yaml
mqtt:
  broker: 192.0.2.1
```

- the JSON component wants a `free_heap` number but it's hardware specific and only available for ESP8266/ESP32
- the MQTT library similarly only support ESP8266/ESP32

```
Compiling /tmp/.esphome/.pioenvs/picodebug/src/esphome/components/json/json_util.cpp.o
src/esphome/components/json/json_util.cpp: In function 'std::string esphome::json::build_json(const json_build_t&)':
src/esphome/components/json/json_util.cpp:29:34: error: 'free_heap' was not declared in this scope
   29 |   size_t request_size = std::min(free_heap, (size_t) 512);
      |                                  ^~~~~~~~~
```
- see https://github.com/esphome/esphome/blob/dev/esphome/components/json/json_util.cpp#L19-L29

and:

```
Compiling /tmp/.esphome/.pioenvs/picodebug/src/esphome/components/mqtt/custom_mqtt_device.cpp.o
In file included from /tmp/.esphome/.piolibdeps/picodebug/AsyncMqttClient-esphome/src/AsyncMqttClient.h:4,
                 from src/esphome/components/mqtt/mqtt_backend_arduino.h:6,
                 from src/esphome/components/mqtt/mqtt_client.h:15,
                 from src/esphome/components/mqtt/custom_mqtt_device.h:7,
                 from src/esphome/components/mqtt/custom_mqtt_device.cpp:1:
/tmp/.esphome/.piolibdeps/picodebug/AsyncMqttClient-esphome/src/AsyncMqttClient.hpp:14:2: error: #error Platform not supported
   14 | #error Platform not supported
```

- see https://github.com/OttoWinter/async-mqtt-client/blob/master/src/AsyncMqttClient.hpp#L8-L15

jesserockz, 2022-11-14:
Hi @bertvdijk, thanks for the report. I have fixed the `time` component issues in https://github.com/esphome/esphome/pull/4024.

MQTT on the other hand will probably not be implemented by me for now. I won't stop someone else sorting it out though. I will mark is an unavailable in https://github.com/esphome/esphome/pull/4025

skandragon, 2022-11-16:
@jesserockz Can you give a quick summary of what you think would be needed for mqtt support?  I'm a heavy mqtt user in my setup.

Sese-Schneider, 2022-11-17:
@jesserockz Is this the right place to track RP2040 support for the `fastled_*` component?
What do you think would be needed for fastled support?
<sub>P.S.: +1 for MQTT</sub>

jesserockz, 2022-11-17:
Hi everyone,
Please search for or create a new feature request from this point forward.

Thanks,
Jesse

closed 2022-11-17 by jesserockz

marcrleonard, 2022-11-17:
> Hi everyone, Please search for or create a new feature request from this point forward.

Does this mean RP2040 is now merged into the main branch?!

balloob, 2022-11-17:
It's already released: https://esphome.io/changelog/2022.11.0.html

zrides, 2022-11-18:
> Okay, adding more info here since I've been up to my neck in debugging:
>
> WiFi Router / Vendor	Network	Encryption	Works (esphome)	Works (MicroPython)
> ASUS RT-AX86U	2.4 GHz (ax / WiFi 6 mode) [...]

Just an update on this seems like the issue lies in DHCP if you switch to using a manual IP (see below) then there is no need to mess with router settings. 

```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
  manual_ip:
    static_ip: 192.168.86.5
    gateway: 192.168.86.1
    subnet: 255.255.255.0
```