esphome/feature-requests #1926: SFA30 Support

iLeopod, 2022-10-20:
**Add Support for Sensirion SFA30 sensor**

https://github.com/Sensirion/arduino-i2c-sfa3x
https://sensirion.com/products/catalog/SFA30/

ghsensdev, 2023-09-15:
Does anyone plan to implement this component?
Does anyone have this sensor?

I plan to buy one and implement the component for it.

ghsensdev, 2023-10-12:
Hello,

I bought this sensor, and implement a component for it.

I have a pull request that needs a code review.

Can someone help me what I have to do next?

nagyrobi, 2023-10-12:
Thank you for your effort. Now it's time to sit back and wait for it to be reviewed.

ghsensdev, 2023-10-12:
Should I choose a reviewer or one of them will notice the pull request some time?

nagyrobi, 2023-10-12:
No need to choose a reviewer. It may take quite a long time as they are overwhelmed.

closed 2023-10-15 by jesserockz

ghsensdev, 2023-10-18:
It has been merged to dev, thanks. @jesserockz, in the next release will it be available?

iLeopod, 2023-11-04:
[12:54:43][D][sensor:094]: 'Formaldehyde': Sending state 21.60000 ppb with 1 decimals of accuracy
[12:54:43][D][sensor:094]: 'Humidity': Sending state 65.71000 % with 2 decimals of accuracy
[12:54:43][D][sensor:094]: 'Temperature': Sending state 19.36500 °C with 2 decimals of accuracy`

Test works great so far:)