esphome/feature-requests #1929: [DSMR] Support for variabel /templating of the request interval

Balou067, 2022-10-23:
**Describe the problem you have/What new integration you would like**
I'm using newly the ESPHome DSMR component and it would be great if it would be possible to change the request interval dynamically via a value entered in the front end. Trying to do this per lambdas lead to error message that for this value it is not possible to use lambdas templets. Would be great if this would be possible. 

**Please describe your use case for this integration and alternatives you've tried:**
With a request interval definable via the front end, it would be possibel to use a moderate reading rate as long as no accurate consumption data are needed. If the update interval shall be faster, it could be changed via the front end without editing the ESPHome configuration and uploading it.

**Additional context**
I'm using the up-to date ESPHome DSMR component (updated via Home Assistant) and reading the data from the smart meter via a D1 Minu Board.

nagyrobi, 2022-10-23:
As far as I know, most DSMR meters send the data by themselves at regular intervals (mine does every 10 seconds). You could use add `throttle` filter to your sensors with the desired refresh interval if you want.
Or you could add a `copy` or a `template` sensor returning your original sensor with a desired update interval.

Balou067, 2022-10-24:
@nagyrobi : Thanks for the proposal. My Smart Meter those report every second if the data request pin stays at +5V. If I change the request interval in then ESPHome configuration, the Smart Meter reports at the defined interval. To use a throttle filter to adjust the update interval of the sensors is a good idea. However, I would still prefer the direct adjustment of the request interval via the frontend since it would reduce the traffic from the Smart Meter to the D1 Mini Board. (Okay, this might be not really a problem but still, why sending data if you do not need/wont them?  :-)  )

nagyrobi, 2022-10-24:
Please head on to our Discord server for further help.

closed 2022-10-24 by nagyrobi