esphome/feature-requests #1934: Third Trigger on_stop for Rotary Encoders

Pippowicz, 2022-10-26:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

The current Implementation gives back the last movement state.

**Additional context**

RoboMagus, 2022-10-28:
Accoding to the [Docs](https://esphome.io/components/sensor/rotary_encoder.html#on-clockwise-and-on-anticlockwise-triggers) these triggers `trigger on every step`. So the triggers are not really used to indicate _is moving_, but just the increment it has stepped. As such I do not think an `on_stop` trigger makes much sense.

I've modified a yaml config to somewhat match your usecase as I've done something similar using a restarting script and a timeout before it's action is performed:

```yaml

sensor:
  - platform: rotary_encoder
    id: my_rotary_encoder
    on_clockwise:
      - logger.log: "Turned Clockwise"
      - script.execute: stopped
    on_anticlockwise:
      - logger.log: "Turned Anticlockwise"
      - script.execute: stopped

script:
  - id: stopped
    mode: restart
    then:
      - delay: 500ms
      - logger.log: "Stopped turning"
```

nagyrobi, 2022-10-28:
That's kind of how I also do this.

nagyrobi, 2022-10-28:
> For my Garage Door

By the way, how do you use a rotary encoder for a garage door?

bkaufx, 2022-10-28:
I got the idea from this dude but haven't implemented it yet:  https://paulwieland.github.io/ratgdo/

Look at his soldered wiring diagram and accompanying photo.  I've been meaning to solder the rotary encoder of my garage door to a d1 mini.

Pippowicz, 2022-10-29:
> > For my Garage Door
> 
> By the way, how do you use a rotary encoder for a garage door?

By using a toothed Belt and getting the rotation from the shaft of the Motor. Combined with 2 Endstops i can see how wide the door ist opened. The counts reset at the Endstops.

At the Moment i got this: 

`output:
  - platform: gpio
    pin: 32
    id: output1

button:
  - platform: output
    name: "Garage Door"
    output: output1
    duration: 500ms

binary_sensor:
  - platform: gpio
    name: "Door open"
    id: door_open
    pin:
      number: 33
      inverted: true
      mode:
        input: true
        pullup: true
    on_press:
      - sensor.rotary_encoder.set_value:
          id: encoder
          value: 500
  - platform: gpio      
    name: "Door closed"
    id: door_closed
    pin: 
      number: 25
      inverted: true
      mode:
        input: true
        pullup: true
    on_press:
      - sensor.rotary_encoder.set_value:
          id: encoder
          value: 0

sensor:
  - platform: template
    name: "Direction Sensor"
    id: direction_sensor

  - platform: rotary_encoder
    name: "Öffnung"
    id: encoder
    min_value: 0
    max_value: 500
    on_clockwise:
      - sensor.template.publish:
          id: direction_sensor
          state: 1
    on_anticlockwise:
      - sensor.template.publish:
          id: direction_sensor
          state: -1
    pin_a:
      number: 12
      mode:
        input: true
        pullup: true
    pin_b: 
      number: 14
      mode:
        input: true 
        pullup: true 
    filters: 
    - lambda: |-
        auto first_mass = 100;
        auto first_value = 500;
        auto second_mass = 0;
        auto second_value = 5;

        auto r = map (x, first_value, second_value, first_mass, second_mass);
        if (r >0) return r;
        return 0;
    unit_of_measurement: "%"`

nagyrobi, 2022-10-29:
I wonder how that could be properly combined with a `cover` component...

Pippowicz, 2022-10-29:
But do you have to use the Cover component instead of using the single Entities? Perhaps im overthinking this and there is a better way, but i dont think here is the right place. 

Thinking about dropping a post in the Forums.

nagyrobi, 2022-10-30:
Post the link to the forums with your discussion on this

Pippowicz, 2022-10-31:
Here is the Thread: https://community.home-assistant.io/t/esphome-custom-garage-door-integration/482125

RoboMagus, 2022-10-31:
I guess this issue can be closed in favor of the forum thread then?

Pippowicz, 2022-10-31:
Yeah, if you dont think about the on_stop trigger this can be closed.

closed 2022-10-31 by nagyrobi