esphome/feature-requests #1936: How to set up development platform for ESPHome

pedjas, 2022-10-28:
**Describe the problem you have/What new integration you would like**

I have ESP2066 device that is identified as esp01_1m. ESPHome does not have PIN aliases defined for this device so I have to use GPIO designation for pins. I would like to add those aliases. I found out file https://github.com/esphome/esphome/blob/dev/esphome/components/esp8266/boards.py which seems to be place where aliases should be defined.

Problem is I am first time meeting with ESPHome so I am not familiar with the development platform. I would like to learn how to set up development platform to be able to test if I did all changes properly. I cannot find such instructions.

It would be good to have some tutorial how to set up development platform and run tests of the altered code.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

I use ESPHome with Home Assistant as Virtual Machine on Windows. Windows is main platform I use, thus I would like to use it for this development too.

ssieb, 2022-10-28:
It is recommended that you use GPIO numbers anyway.
https://esphome.io/guides/contributing.html is the documentation for setting up development.

jackhax1, 2022-11-04:
Easiest way is to first fork the esphome repo. Then add "external_components:" line to your yaml. 
For example:
`external_components:
  - source:
      type: git
      url: https://github.com/<your_github_username>/esphome
      ref: dev
    components: [the_component_you_edited]`

closed 2022-11-04 by nagyrobi