esphome/feature-requests #1939: How to create a UART binary sensor?

phpphph, 2022-10-31:
Use esp8266 to extend io, after setting the input in the external mcu, it will be sent to the esp8266 through the serial port, and the esp8266 creates the "binary_sensor:" entity in the esphome. .
Since I don't know programming very well, the problem I'm facing now is rather confusing.
The command sent by mcu to esp8266 is "0x12", but it is not displayed in the esphome log. After adding the symbol "0x12\r", there is an action in the esphome log, but it is an empty command.
Who can help me with this project? I would really appreciate it! The following is the command format:

`
esphome:
  name: esp-uart
  includes:
    - uart_read_line_sensor.h
    - .....................
captive_portal:
uart:
  tx_pin: 0
  rx_pin: 3
  id: uart_bus
  baud_rate: 115200

text_sensor:
  - platform: custom
    lambda: |-
      auto my_custom_sensor = new UartReadLineSensor(id(uart_bus));
      App.register_component(my_custom_sensor);
      return {my_custom_sensor};
    text_sensors:
      id: "uart_readline"
    
binary_sensor:
  - platform: template
    name: "Garage Door Open"
    lambda: |-
      if (id(uart_readline).state == "0x12\r") {
        return true;
      } else if(id(uart_readline).state == "0x02\r") {
        return false;
      } else {
        return {};
      }
      `

nagyrobi, 2022-10-31:
Please head on to our Discord server for fast support.

closed 2022-10-31 by nagyrobi

phpphph, 2022-10-31:
> 请前往我们的 不和谐服务器 获得快速支持。

I'm in China, I'm inconvenient to connect due to network problems, sorry

nagyrobi, 2022-10-31:
Discord is better because people answer immediately and there's more people to help.
Try https://esphome.io/cookbook/uart_text_sensor.html

phpphph, 2022-10-31:
> 不和谐更好，因为人们会立即回答，并且有更多的人需要帮助。试试https://esphome.io/cookbook/uart_text_sensor.html

I agree with your statement, but I occasionally have problems accessing github from the network here. I will find a way to establish a VPN.
I am using the UART_Text sensor. But the current problem of my project is that the command received by esphome is empty, that is to say, the command I defined does not work.