esphome/feature-requests #1941: Infrared control codes for R11HG/E remote or Electrolux A/C units

csomyy, 2022-11-01:
**Describe the problem you have/What new integration you would like**

I would control my Electrolux A/C with ESPHome. I have a working IR transmitter but the codes i copy from my original remote controller are longer than other codes like from my LG TV button.

**Please describe your use case for this integration and alternatives you've tried:**

There is no IR Climate platform for my device. My IR trasmitter is working with other devices. I do know that A/C controlling is different a bit, i found the climate_ir component but neither existing code sets are working for me. Tried copy and transmit codes in pronto and raw formats, but had no luck. My device is an Electrolux ESA12PCGMB device. 

**Additional context**

I tried copy commands from my existing remote controller wich is an R11HG/E type remote. Google answers this like a Midea controller but the midea_ir platform isn't working.
I can copy my remote commands if this helps.

csomyy, 2023-02-01:
Upvoting this for myself :(

Badis-Bfifteen, 2023-10-29:
I have the same for a Cherokee Maestro brand   R11HG/E type remote