esphome/feature-requests #1942: Inkplate 6Color Support

JohnNeville, 2022-11-02:
**Describe the problem you have/What new integration you would like**

I would like support for the Inkplate 6Color's display. It seems largely similar to the Inkplate 6 and 10 which are already supported but supports 7 colors and only 600x448. 

**Please describe your use case for this integration and alternatives you've tried:**

It would be nice to be able to use my color display in addition to the other Inkplate line products.

**Additional context**

https://www.crowdsupply.com/soldered/inkplate-6color

jakemauer, 2023-12-04:
I recently purchased one of these and would love to help add support for it.

skurfuerst, 2024-08-21:
Hey,

anything we can do to make this happen? Would you accept a pull request?

All the best,
Sebastian