esphome/feature-requests #1943: BTHome local sensors

nagyrobi, 2022-11-03:
**Describe the problem you have/What new integration you would like**
[BTHome](https://bthome.io/) combined with [Bluetooth Proxy](https://esphome.io/components/bluetooth_proxy.html) works great to forward the data to Home Assistant, and also provides fault tolerance for the cases when an ESPHome node fails or RF reception is degraded.

The downside is, that the sensor data is not accessible anymore on ESPHome node directly like [it was before](https://esphome.io/components/sensor/xiaomi_ble.html#lywsd03mmc) the proxy functionality was implemented. This makes it a bit problematic to run automations directly in ESPHome (eg. create a thermostat/climate) which can run completely offline from Home Assistant. 

**Please describe your use case for this integration and alternatives you've tried:**
Please implement the BTHome protocol (with bindkey support) based on BLE tracker, so it can be read by ESPHome nodes directly too.

The use case is when you want to use the proxy method to forward the data in HA with fault tolerance, but also provide a local control solution for the case when HA or the Wi-Fi network goes down for any reason. This doesn't have to be on the same node as the proxy - a separate node could be used without the proxy functionality, but with local BLE sensors received from the same air broadcasts.

Currently this is not possible because the BTHome protocol can only be decoded at HA level.

**Additional context**

thatslolo, 2022-12-28:
I would love to see the feature as well.

afarago, 2023-06-24:
Meet project Beethowen, my latest development that  mplements ian energy efficient component to transfer data over ESP-NOW Communication Protocol over BTHome protocol, and bthome_receiver component that enables anonymous reception on an ESP32 ESPHome node from remote devices.

Beethowen could enable 3-9 years of smart home sensor lifespan and easy development to everyone.
https://medium.com/@attila.farago.hu/implementing-low-energy-smart-devices-with-several-years-of-independence-970077900cf4

Github open source repository: https://github.com/afarago/esphome_component_bthome

nagyrobi, 2023-06-24:
Gratulálok!

asucrews, 2024-06-10:
This would be a great add! Looks like Beethowen not under active development so there does seem to be a esphome component

hwmaier, 2024-12-24:
I also support the idea of local BT home sensors which are independent of HA.

klim123123, 2025-06-11:
anyone found a way to receive data from BTHome sensors locally in ESPHome?

asucrews, 2025-06-27:
closet thing i can find is eps32_ble_server and ble_client, doesn't look like bthome to me.

https://esphome.io/components/esp32_ble_server and https://esphome.io/components/ble_client

klim123123, 2025-06-27:
> closet thing i can find is eps32_ble_server and ble_client, doesn't look like bthome to me.
> 
> https://esphome.io/components/esp32_ble_server and https://esphome.io/components/ble_client

total different thing. ble_client is for direct connect to device.
but BTHome protocol is broadcaster-based, you dont' need to connect, just listen and decode signal.

StefanoPierini, 2025-11-22:
I expected this feature to be present in ESPHome. For me this feature should be added directly to ESPHome natively, starting from the great job of @afarago.