esphome/feature-requests #1948: Integration new Xiaomi sensor door&windows (MCCGQ02HL) (2nd attempt)

ahpohl, 2022-11-05:
**Describe the problem you have/What new integration you would like**

I would like to have an integration with the new Xiaomi sensor door&windows (MCCGQ02HL) which has BLE sensor.

**Please describe your use case for this integration and alternatives you've tried:**

I would like to have a similar integration like other Xiaomi BLE devices, without Xiaomi Gateway.

**Additional context**

A [feature request](https://github.com/esphome/feature-requests/issues/978) for this integration was opened a long time ago, and a PR developed by @morph027 https://github.com/esphome/esphome/pull/1645. The PR was never merged and something has changed since then since the integration doesn't compile anymore due to configuration errors. Although @szilardcsotya [reported](https://github.com/esphome/feature-requests/issues/978#issuecomment-1017917453) that the zigbee version works much better, I still use the bluetooth version myself and would like to have it merged officially.

**Todo:** Need somebody who can fix the python configuration error in `binary_sensor.py` (`AttributeError: module 'esphome.components.binary_sensor' has no attribute 'device_class'`) and submit a new PR . I believe the actual sensor code is still working and doesn't need changing.

ahpohl, 2022-11-05:
I checked out @morph027 Xiaomi-MCCGQ02HL branch https://github.com/esphome/esphome/pull/1645 and used the following configuration:
```
binary_sensor:
  - platform: xiaomi_mccgq02hl
    name: "Garage-Door Sensor"
    mac_address: "E4:AA:EC:4E:5E:E4"
    bindkey: "37a1a4b743ee7bdb6f14631e7acf1f79"
    battery_level:
      name: "Garage-Door Battery-Level"
    has_light:
      name: "Garage-Door Light"
    open:
      name: "Garage-Door Open"
```
The full compile error:
```INFO Reading configuration livingroom.yaml...
ERROR Unable to load component xiaomi_mccgq02hl.binary_sensor:
Traceback (most recent call last):
  File "/home/alex/git/morph027/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/alex/git/morph027/esphome/components/xiaomi_mccgq02hl/binary_sensor.py", line 37, in <module>
    cv.Optional(DEVICE_CLASS_LIGHT): binary_sensor.device_class,
AttributeError: module 'esphome.components.binary_sensor' has no attribute 'device_class'
Failed config

binary_sensor.xiaomi_mccgq02hl: [source livingroom.yaml:114]
  
  Platform not found: 'binary_sensor.xiaomi_mccgq02hl'.

```
The code used to work and I tested it for a couple of month on three different MCCGQ02HL sensors without issues. Unfortunately it got never merged.

[image]

ahpohl, 2022-11-08:
configuration errors fixed in https://github.com/ahpohl/esphome/tree/xiaomi_mccgq02hl. `binary_sensor.py` needs code review (don't really understand what I changed, but it seems to compile correctly)

ahpohl, 2022-11-11:
@ssieb Would it be possible to add me as a collaborator to PR https://github.com/esphome/esphome/pull/1645?

I am not able to comment on the code. The pylint error I see in Travis is exactly the same I have seen when manually compiling the code:

```Error: esphome/components/xiaomi_mccgq02hl/binary_sensor.py:37: [E1101(no-member), ] Module 'esphome.components.binary_sensor' has no 'device_class' member```

Error copied from here: https://github.com/esphome/esphome/actions/runs/3406797462/jobs/5665803590

I have fixed the configuration error in my repository, but might have broken the HomeAssistant look and feel (icons etc.). Easiest would be if @morph027 could fix the code in the original PR.

morph027, 2022-11-11:
Sure, i can add the changes to the original PR :+1:

maorcc, 2023-01-09:
> Sure, i can add the changes to the original PR 👍

Hey, @morph027, any success adding the changes to the original PR?

morph027, 2023-01-10:
Uh, this one went out of my mind ... will try include the changes soon.

Fabian-Schmidt, 2023-05-05:
Hi,

I extracted the code into a separate external component to make it usable while waiting for PR to be merged.

<https://github.com/Fabian-Schmidt/esphome-xiaomi_mccgq02hl>

closed 2026-10-07 by bdraco