esphome/feature-requests #1949: Resol Deltasol support

proasnet, 2022-11-07:
**Describe the problem you have/What new integration you would like**

Support for receive data over Tx from Resol Deltasol solar controller. 

**Please describe your use case for this integration and alternatives you've tried:**

I need data to home assistant
Alterane of  Tasmota smart meter and his script for Resol

**Additional context**

nagyrobi, 2022-11-07:
Support has been added, just not yet released: https://github.com/esphome/esphome/pull/3976
Doc for the PR: https://deploy-preview-2397--esphome.netlify.app/components/vbus.html

ssieb, 2022-11-07:
Which specific device do you want?

nagyrobi, 2022-11-07:
The component currently supports natively Resol Deltasol C, DeltaSol CS2 and DeltaSol BS Plus but any device can be added. 

Try looking [its packet structure](https://danielwippermann.github.io/resol-vbus) maybe CS Plus matches one of the supported ones.

ssieb, 2022-11-08:
> Resol Deltasol CS Plus. Over VBus to Rx pin of esp

Original or V2?

proasnet, 2022-11-08:
Oh. I dont know now. Where I find this info?

ssieb, 2022-11-08:
No idea.  I assume it's written on the device somewhere.  I guess if you don't see a V2 anywhere, it's most likely the original.  You could tell by seeing what code it sends, if you have some way to see that.

proasnet, 2022-11-08:
I think, that is original.
https://www.resol.de/en/produktdetail/98

proasnet, 2022-11-08:
Unfortunately I can't see. Connecting with tasmata doesn't work for me, maybe I can't compile tasmata correctly. The data is coming to my Rx, but I keep seeing zeros in the web interface. That's why I also want to switch to esphome for this device.
I'll just remind you that I'm a beginner with esp, and the only node I'm using now is from @syssi JKBMS.
There are also problems with the simultaneous connection of two JKBMS via BLE, but this is solved by @syssi

nagyrobi, 2022-11-08:
@Gloomyeye yes, read the doc linked above.

ssieb, 2022-11-08:
What is the alternative that was offered?

proasnet, 2022-11-08:
> 

Alernative of? I am not inderstand your ask

nagyrobi, 2022-11-08:
Yeah, a more appropriate label would probably be `sollution-offered` - as there exists a viable solution to the request. 

`alternative` still stands though, as it's not (yet) available by default, but it can be used as an external component config pointing to either @ssieb's repo or to the PR itself. How to do that is in both the docs and the discord thread I linked above.

proasnet, 2022-11-09:
Is it possible to use the given example?
I tried the linked VBUS example yesterday and got an error message when installing

ssieb, 2022-11-09:
What's the error?

nagyrobi, 2022-11-09:
> Is it possible to use the given example?

I actively use it even in this moment

proasnet, 2022-11-09:
If I used the example here
https://deploy-preview-2397--esphome.netlify.app/components/vbus.html
but if I add this to config
```
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/custom_components
      ref: vbus
    components: [ vbus ]
    refresh: 1min
```
bin was compiled, but not tested now, I try as tasmota update now

ssieb, 2022-11-09:
The PR isn't merged yet, so you still need to use it as an external component, either from the PR or my repo.  The CS Plus support is only in the PR though.

proasnet, 2022-11-09:
So, because I do it remotely from my work,
I use RDP to my virtual pc in my home. ESP is not connected by some usb cable to HA.
I load Tasmota minimal, then I upload a resol.bin ( compiled in ESPHome in my HA server ), done.
Then I get a new integration, but state of entities is unavailable.
Here is my resol.yaml config file for ESP, is some bad?
```
esphome:
  name: resol

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "va6X6ktqWihT53U31bB2Zz9fA3VG53sHDl4Dn8VPE+k="

ota:
  password: "e1a0f1d04cce64f1c4053f7d1f70183d"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

external_components:
  - source:
      type: git
      url: https://github.com/ssieb/custom_components
      ref: vbus
    components: [ vbus ]
    refresh: 1min

uart:
  id: resol
  rx_pin: GPIO3
  baud_rate: 9600

vbus:
  uart_id: resol

sensor:
  - platform: vbus
    model: deltasol_bs_plus
    temperature_1:
      name: Temperature 1
    temperature_2:
      name: Temperature 2
    temperature_3:
      name: Temperature 3
    temperature_4:
      name: Temperature 4
    pump_speed_1:
      name: Pump Speed 1
    pump_speed_2:
      name: Pump Speed 2
    operating_hours_1:
      name: Operating Hours 1
    operating_hours_2:
      name: Operating Hours 2
    heat_quantity:
      name: Heat Quantity
    time:
      name: Device Time
    version:
      name: Device firmware version
```

nagyrobi, 2022-11-09:
You configured a BS Plus, is that what you have?

proasnet, 2022-11-09:
No, CS Plus I have.
[image]

nagyrobi, 2022-11-09:
A BS Plus config will likely _not_ be suitable for CS Plus...

proasnet, 2022-11-09:
I changed to cs_plus
no compile, error.
```
INFO Reading configuration /config/esphome/resol.yaml...
Failed config

sensor.vbus: [source /config/esphome/resol.yaml:36]
  
  Unknown value 'deltasol_cs_plus', did you mean 'deltasol_bs_plus', 'deltasol_cs2', 'deltasol_c'?.
  platform: vbus
  model: deltasol_cs_plus
  temperature_1: 
    name: Temperature 1
  temperature_2: 
    name: Temperature 2
  temperature_3: 
    name: Temperature 3
  temperature_4: 
    name: Temperature 4
  pump_speed_1: 
    name: Pump Speed 1
  pump_speed_2: 
    name: Pump Speed 2
  operating_hours_1: 
    name: Operating Hours 1
  operating_hours_2: 
    name: Operating Hours 2
  heat_quantity: 
    name: Heat Quantity
  time: 
    name: Device Time
  version: 
    name: Device firmware version
```

nagyrobi, 2022-11-09:
As @ssieb wrote earlier, your CS Plus is only in the PR so you need instead:
```yaml
external_components:
  - source: github://pr#3976
    components: [ vbus ]
    refresh: 1min
```

proasnet, 2022-11-09:
```
INFO Reading configuration /config/esphome/resol.yaml...
Failed config

external_components: [source /config/esphome/resol.yaml:20]
  - source: github://pr#3976
    components: 
      
      Could not find __init__.py file for component vbus. Please check the component is defined by this source (search path: /config/esphome/.esphome/external_components/a6abb671/esphome/components/vbus/__init__.py).
      - vbus
    refresh: 1min
```

proasnet, 2022-11-09:
try again, another and more errors
```
INFO Reading configuration /config/esphome/resol.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/3976/head
ERROR Unable to import component vbus.sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/a6abb671/esphome/components/vbus/sensor/__init__.py", line 4, in <module>
    from esphome.const import (
ImportError: cannot import name 'UNIT_HOUR' from 'esphome.const' (/esphome/esphome/const.py)
Failed config

sensor.vbus: [source /config/esphome/resol.yaml:37]
  
  Platform not found: 'sensor.vbus'.
  platform: vbus
  model: deltasol_cs_plus
  temperature_1: 
    name: Temperature 1
  temperature_2: 
    name: Temperature 2
  temperature_3: 
    name: Temperature 3
  temperature_4: 
    name: Temperature 4
  heat_quantity: 
    name: Heat Quantity
```

ssieb, 2022-11-09:
Right, that's an annoying problem with using PRs as external components.
Ok, I merged the changes from the PR to my repo, so you can try it out there instead.

proasnet, 2022-11-09:
Okay, I try it now. Wait a moment please.

proasnet, 2022-11-09:
One ask for you. The schematic of vbus level converter with 2 transistors, working fine? I have it as in schematic, but not tested. In another comments in Tasmota discussions, some peaple wrote, that this schematic not working.

ssieb, 2022-11-09:
I don't have this device, so I can't say anything.  However, the person that put that diagram there is using it.

proasnet, 2022-11-09:
so, still unavailable entities status

ssieb, 2022-11-09:
It would be better if you join the discord server to discuss it there.  I'll be back there in about 9 hours.  But check the logs.  I assume you're not getting any data from the device.  Set the logging level to verbose.

proasnet, 2022-11-09:
I try swap wires from Resol, is possible that is swapped if my tasmota not working and I trying test why....
Discord server is what? I am sorry, beginner...

proasnet, 2022-11-09:
with verbose level
```
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from resol.local using esphome API
INFO Successfully connected to resol.local
[10:55:34][I][app:102]: ESPHome version 2022.10.2 compiled on Nov  9 2022, 10:50:38
[10:55:35][C][wifi:502]: WiFi:
[10:55:35][C][wifi:360]:   Local MAC: E0:98:06:0C:35:0D
[10:55:35][C][wifi:361]:   SSID: [redacted]
[10:55:35][C][wifi:362]:   IP Address: 192.168.0.151
[10:55:35][C][wifi:363]:   BSSID: [redacted]
[10:55:35][C][wifi:365]:   Hostname: 'resol'
[10:55:35][C][wifi:367]:   Signal strength: -70 dB ▂▄▆█
[10:55:35][V][wifi:369]:   Priority: 0.0
[10:55:35][C][wifi:371]:   Channel: 6
[10:55:35][C][wifi:372]:   Subnet: 255.255.255.0
[10:55:35][C][wifi:373]:   Gateway: 192.168.0.254
[10:55:35][C][wifi:374]:   DNS1: 192.168.0.254
[10:55:35][C][wifi:375]:   DNS2: 195.146.132.58
[10:55:35][C][logger:275]: Logger:
[10:55:35][C][logger:276]:   Level: VERBOSE
[10:55:35][C][logger:277]:   Log Baud Rate: 115200
[10:55:35][C][logger:278]:   Hardware UART: UART0
[10:55:35][C][uart.arduino_esp8266:102]: UART Bus:
[10:55:36][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[10:55:36][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[10:55:36][C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[10:55:36][C][uart.arduino_esp8266:109]:   Data Bits: 8
[10:55:36][C][uart.arduino_esp8266:110]:   Parity: NONE
[10:55:36][C][uart.arduino_esp8266:111]:   Stop bits: 1
[10:55:36][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[10:55:36][W][uart.arduino_esp8266:127]:   You're using the same serial port for logging and the UART component. Please disable logging over the serial port by setting logger->baud_rate to 0.
[10:55:36][C][vbus:011]: VBus:
[10:55:36][C][vbus.sensor:127]: Deltasol CS Plus:
[10:55:36][C][vbus.sensor:128]:   Temperature 1 'Temperature 1'
[10:55:36][C][vbus.sensor:128]:     Device Class: 'temperature'
[10:55:36][C][vbus.sensor:128]:     State Class: 'measurement'
[10:55:36][C][vbus.sensor:128]:     Unit of Measurement: '°C'
[10:55:37][C][vbus.sensor:128]:     Accuracy Decimals: 1
[10:55:37][C][vbus.sensor:128]:     Icon: 'mdi:thermometer'
[10:55:37][C][vbus.sensor:129]:   Temperature 2 'Temperature 2'
[10:55:37][C][vbus.sensor:129]:     Device Class: 'temperature'
[10:55:37][C][vbus.sensor:129]:     State Class: 'measurement'
[10:55:37][C][vbus.sensor:129]:     Unit of Measurement: '°C'
[10:55:37][C][vbus.sensor:129]:     Accuracy Decimals: 1
[10:55:37][C][vbus.sensor:129]:     Icon: 'mdi:thermometer'
[10:55:37][C][vbus.sensor:130]:   Temperature 3 'Temperature 3'
[10:55:37][C][vbus.sensor:130]:     Device Class: 'temperature'
[10:55:37][C][vbus.sensor:130]:     State Class: 'measurement'
[10:55:37][C][vbus.sensor:130]:     Unit of Measurement: '°C'
[10:55:37][C][vbus.sensor:130]:     Accuracy Decimals: 1
[10:55:37][C][vbus.sensor:130]:     Icon: 'mdi:thermometer'
[10:55:38][C][vbus.sensor:131]:   Temperature 4 'Temperature 4'
[10:55:38][C][vbus.sensor:131]:     Device Class: 'temperature'
[10:55:38][C][vbus.sensor:131]:     State Class: 'measurement'
[10:55:38][C][vbus.sensor:131]:     Unit of Measurement: '°C'
[10:55:38][C][vbus.sensor:131]:     Accuracy Decimals: 1
[10:55:38][C][vbus.sensor:131]:     Icon: 'mdi:thermometer'
[10:55:38][C][vbus.sensor:137]:   Heat Quantity 'Heat Quantity'
[10:55:38][C][vbus.sensor:137]:     Device Class: 'energy'
[10:55:38][C][vbus.sensor:137]:     State Class: 'measurement'
[10:55:38][C][vbus.sensor:137]:     Unit of Measurement: 'Wh'
[10:55:38][C][vbus.sensor:137]:     Accuracy Decimals: 0
[10:55:38][C][vbus.sensor:137]:     Icon: 'mdi:radiator'
[10:55:38][C][captive_portal:088]: Captive Portal:
[10:55:38][C][mdns:100]: mDNS:
[10:55:38][C][mdns:101]:   Hostname: resol
[10:55:38][V][mdns:102]:   Services:
[10:55:38][V][mdns:104]:   - _esphomelib, _tcp, 6053
[10:55:39][V][mdns:106]:     TXT: version = 2022.10.2
[10:55:39][V][mdns:106]:     TXT: mac = e098060c350d
[10:55:39][V][mdns:106]:     TXT: platform = ESP8266
[10:55:39][V][mdns:106]:     TXT: board = esp01_1m
[10:55:39][V][mdns:106]:     TXT: network = wifi
[10:55:39][C][ota:089]: Over-The-Air Updates:
[10:55:39][C][ota:090]:   Address: resol.local:8266
[10:55:39][C][api:138]: API Server:
[10:55:39][C][api:139]:   Address: resol.local:6053
[10:55:39][C][api:143]:   Using noise encryption: NO
[10:55:41][D][api:102]: Accepted 192.168.0.249
[10:55:41][V][api.connection:898]: Hello from client: 'Home Assistant 2022.11.1 (192.168.0.249)' | API Version 1.7
[10:55:41][D][api.connection:917]: Home Assistant 2022.11.1 (192.168.0.249): Connected successfully
[10:55:41][V][component:199]: Component api took a long time for an operation (0.09 s).
[10:55:41][V][component:200]: Components should block for at most 20-30ms.
[11:00:19][I][ota:109]: Boot seems successful, resetting boot loop counter.
```

nagyrobi, 2022-11-09:
Please continue on Discord for further support, see linked thread above.

closed 2022-11-09 by nagyrobi

proasnet, 2022-11-09:
I use Discord, but your link above give me
NO TEXT CHANNELS
You find yourself in a strange place. You dont have access to any
text channels, or there are none in this server...

nagyrobi, 2022-11-09:
Channel is: https://discord.gg/hG2r39hT within it, thread link about resol is: https://discord.com/channels/429907082951524364/1030612607738052628

proasnet, 2022-11-10:
I dont know what is bad...
I tried another circuit and still unavailable entities...
[image]
[image]

proasnet, 2022-11-10:
[image]

nagyrobi, 2022-11-10:
Try verifying the polarity of the signal coming out of the Resol. Use a multimeter and try measuring voltage, should be something varying between 0 and about 8V.

Positive should be towards the resistor, negative to the ground.

[image]

proasnet, 2022-11-10:
Yes, this is the same circuit. I am doing with electronic in my word, I am understand this easy circuit with non inverted switch.
I tried both output polarities, and is 8V voltage.

proasnet, 2022-11-10:
This is my working config for Resol Deltasol CS Plus
```
substitutions:
  name: resol
  device_description: "Monitor and control Resol Deltasol by VBUS"

esphome:
  name: resol
  comment: ${device_description}

esp8266:
  board: esp01_1m

logger:
  baud_rate: 0
  
api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

external_components:
  - source:
      type: git
      url: https://github.com/ssieb/custom_components
      ref: vbus
    components: [ vbus ]
    refresh: 1min

uart:
  id: resol
  rx_pin: GPIO3
  baud_rate: 9600

vbus:
  uart_id: resol

sensor:
  - platform: vbus
    model: deltasol_cs_plus
    temperature_1:
      name: Temperature 1
    temperature_2:
      name: Temperature 2
    temperature_3:
      name: Temperature 3
    heat_quantity:
      name: Heat Quantity

binary_sensor:
  - platform: vbus
    model: deltasol_bs_plus
    relay1:
      name: Relay 1 On
```

proasnet, 2022-11-11:
Hi @nagyrobi , today I see working my defined temperatures. Today has sunny afternoon, solar working,
Temperatures and heat quantity works, but relay not.
I have pump on relay 1 in the resol.

if I use
```
model: deltasol_cs_plus
relay1:
```

This is as error.

Please, can you write here possible sensors for deltasol_cs_plus?
Thank you very much.

proasnet, 2022-11-11:
```
INFO Reading configuration /config/esphome/resol.yaml...
Failed config

binary_sensor.vbus: [source /config/esphome/resol.yaml:54]
  platform: vbus
  model: deltasol_cs_plus
  
  [relay1] is an invalid option for [binary_sensor.vbus]. Please check the indentation.
  relay1: 
    name: Relay 1 On
```

proasnet, 2022-11-11:
use deltasol_bs_plus for this sensor is possible install, but binary sensor not working.

nagyrobi, 2022-11-11:
See https://deploy-preview-2397--esphome.netlify.app/components/vbus.html CS plus doesn't have (yet) implemented the relay binary sensors.

nagyrobi, 2022-11-11:
Without PWM, you'll see 0 and 100%.

ssieb, 2022-11-12:
The technical documentation for the CS Plus doesn't show a relay indicator, only the pump speed.  If you can find something that says otherwise, I can add it.

proasnet, 2022-11-12:
Thank you. I'll leave it like that for now, I'll follow the value you recommended and see. If it works, I'll leave it at that.

proasnet, 2022-11-13:
I found in history graph this abnormal value in some time
[image]

proasnet, 2022-11-13:
This is in night hours, my solar can not heat to this temperature in a night :)

ssieb, 2022-11-13:
Are you recording the sensor error binary sensor data?  What is the exact value there?  I wonder if it's some sort of error code.

proasnet, 2022-11-13:
unfortunately I don't record. I'll set it up in yaml and see if it happens again. It was the same last night.
What I do know is that at night the outside temperature is already negative, this can have an effect if only positive values ​​are allowed

proasnet, 2022-11-13:
According to the outdoor temperature history record, this would be coincidental

proasnet, 2022-11-13:
[image]

ssieb, 2022-11-13:
That's a good point.  I was assuming positive values.  I hadn't seen any indication that negative values would be possible.  (I don't actually know what these are measuring. :smile: ).  What are the values at the top of that?

proasnet, 2022-11-13:
[image]
I don't know what the actual temperature range of the regulator is, but the reading range could be set in esp from -30 to 100 even with a margin

ssieb, 2022-11-13:
Ok, that number makes sense as a negative value.

proasnet, 2022-11-13:
In the manual from the regulator, the measuring range on the solar sensor is not mentioned

proasnet, 2022-11-13:
Please let me know when the problem is solved, I will update esp

ssieb, 2022-11-13:
I've pushed a fix for this.

[1 more reply]