esphome/feature-requests #1953: Add conditions to YAML

carlywarly, 2022-11-09:
**Describe the problem you have/What new integration you would like**

Would it be possible to add conditional statements to the YAML script

**Please describe your use case for this integration and alternatives you've tried:**
The would allow for more generic templates, if ipaddess is defined in the substitutions: section the following block would apply, similar to the way it used in Ansible.

``` yaml
  network:
    ipaddress: ${ipaddress}
    subnet: 255.255.255.0
    gateway: 192.168.0.1
  when: ${ipaddress}
```

in addition, would it also be possible to have, something like the java single line if statement:

``` yaml
    logger:
      level: (${log_level} ? ${log_level} : INFO)
```
Again, if log_level is defined in the substitutions: section, it’s uses that log level otherwise default to INFO, again helping to provided more generic templates.

**Additional context**

ssieb, 2022-11-09:
It's being worked on: https://github.com/esphome/esphome/pull/3612

carlywarly, 2022-11-09:
Hi @ssieb,
Thanks for the pointer, however I can't get it to work, trying:

``` yaml
substitutions:
  device_name: esp32test
  loglevel: debug

esphome:
  name: ${device_name}
...

logger: 
  !if
    condition: ${loglevel}
    then:
      level: ${loglevel}
    else:
      level: warn
```
and
``` yaml
logger: !if
    condition: ${loglevel}
    then:
      level: ${loglevel}
    else:
      level: warn
```

I get the following error:
``` yaml
ERROR Error while reading config: Invalid YAML syntax:

could not determine a constructor for the tag '!if'
  in "/config/esp32test.yaml", line 20, column 9:
    logger: !if
            ^

same with, etc:
``` yaml
  manual_ip: !if
    condition: ${ipaddress}
    then:
      static_ip: ${ipaddress}
      gateway: 192.168.0.1
      subnet: 255.255.255.0
      dns1: 192.168.0.199
      dns2: 192.168.0.1
```