esphome/feature-requests #1954: Add on_press/on_release to button component

psxde, 2022-11-10:
**Describe the problem you have/What new integration you would like**

I try to create a button in the esphome webgui which turns on a relais as long as the button is pressed

**Please describe your use case for this integration and alternatives you've tried:**
This is required to switch on and of a pump which should be only pump a short time as long as the button is pressed. A switch item is not useful because it needs two clicks or a predefined time to turn off again

**Additional context**

ssieb, 2022-11-11:
The `button` component was specifically designed to only have a single press.  Using a web interface is a bad idea if you want accurate and short timing.

psxde, 2022-11-11:
It does not need to be accurate but it should switch on for a variable duration. Are there any workarounds to do this?

psxde, 2022-11-11:
Hmm, with a physical button there is no need for any wifi solution.

I think I will continue using multiple switches with turn-off-timers using various timespans.

Sure, there is always a risk using software but it can be minimized e.g. using a timer which forces the relais to turn off after some time or a physical (emergency stop) switch disconnecting the pump from power source. But in this case a quick and simple solution is needed and the damage on failure is low.

closed 2022-11-11 by nagyrobi

ssieb, 2022-11-11:
> I think I will continue using multiple switches with turn-off-timers using various timespans.

This is what `button:` is intended for.