esphome/feature-requests #1956: Write to Tuya MCU data points like Tuya iOT servers can

jazzmonger, 2022-11-12:
**Describe the problem you have/What new integration you would like**
Add support to Tuya MCU to write to datapoints just like the Tuya iOT servers can

**Please describe your use case for this integration and alternatives you've tried:**
I have a newly reflashed TYWE01 Tuya module on my misbehaving wood pellet stove (the oem vendor app has serious bugs) and the set temp dp  controls the stove. It's the only way to set a desired temp on the stove. But there is no way to do this as dps are read only...

**Additional context**
The same issue exists with ECO Mode (0 & 1) and the P1-P4 (0-3) power levels. I can read those values set via the front panel of the stove thru Tuya MCU, but there is no way to set them like the vendor app can up to the tuya data points in the Tuya servers and then back to the stove. Tuya MCU is unidirectional only, and limited by design.

jazzmonger, 2022-11-12:
Relevant Tasmota TuyaMCU docs:
https://tasmota.github.io/docs/TuyaMCU/

ssieb, 2022-11-13:
It would be more helpful if you pasted the output from esphome where it dumps the datapoints and their types.
You can use either https://esphome.io/components/number/tuya.html for a number or https://esphome.io/components/select/tuya.html for an enum.  You can also directly write to datapoints if you need to.
https://esphome.io/api/classesphome_1_1tuya_1_1_tuya.html

I just looked at the linked issue and you did paste the info there.  So the modes are enums and the temp is a number.  However, you probably want to use https://esphome.io/components/climate/tuya.html anyway, since that covers most of it.

closed 2022-11-13 by ssieb

jazzmonger, 2022-11-14:
If the vendor app is not written correctly as is in your example, tuya climate grabs the wrong dps because the vendor wrote two sets of dps for each attribute, then this is the is the only way to have a useable method to get or push data to the correct data point.