esphome/feature-requests #1957: Trigger on Tag removal PN532

adonno, 2022-11-12:
**Describe the problem you have/What new integration you would like**

It would be awesome if for PN532 we could have a trigger that recognises when a tag is removed 

**Please describe your use case for this integration and alternatives you've tried:**

If I use PN532 for a Jukebox, is should be able to remove the tag (of the album / song) and the music should then stop

**Additional context**

nagyrobi, 2022-11-12:
Isn't this working for you: https://esphome.io/components/binary_sensor/pn532.html?highlight=pn532#on-tag-removed ?

adonno, 2022-11-14:
ok that's on my side. I never saw that this had been added. 
Thats actually amazing 

I am really sorry

closed 2022-11-14 by adonno