esphome/feature-requests #1960: Add support for Lilygo T5-4.7 inch ESP32-S3 (aka "plus"/V2.3) e-paper module

nickolay, 2022-11-15:
There's https://github.com/esphome/feature-requests/issues/1109 open for [the original T5-4.7 module](https://www.lilygo.cc/products/t5-4-7-inch-e-paper). This year the vendor introduced [a new version of the module](https://www.lilygo.cc/products/t5-4-7-inch-e-paper-v2-3), which they call "T5 4.7 Inch E-paper V2.3" or "T5-4.7 Plus".

The vendor advertises the new ESP32-S3 chip and TF-card reader as the major new features, but there are multiple other differences as well. The pinout is incompatible, so any attempts to blindly run examples from #1109 on the new board without first figuring out the differences will fail.

I'll copy the description from the original feature request verbatim, as only the internals of the board are different; the use-case is the same:

> *Describe the problem you have/What new integration you would like*
> The Lyligo T5-4.7inch is a great e-paper module based on an ESP32. It has a resolution of 960*540, 16 gray level and support partial refresh. I would be great to have it integrated in ESPHome the same / similar way Waveshare e-paper screens and modules are.
> 
> *Please describe your use case for this integration and alternatives you've tried:*
> To date there is no support for this module in ESPHome. The only support given by the maker is through Arduino IDE
> 
> *Additional context*
> This is the maker's GitHub for this product: https://github.com/Xinyuan-LilyGO/LilyGo-EPD47

nickolay, 2022-11-15:
So far I managed to get `esphome run` to flash the chip, based on vendor's [demo](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/examples/demo/platformio.ini#L37):
```
esphome:
  name: lilygo
  platformio_options:
    upload_speed: 921600
    monitor_speed: 115200
    board_build.mcu: esp32s3
    board_build.f_cpu: 240000000L
    #uncommenting the following causes: .platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/newlib/platform_include/assert.h:20:10: fatal error: sdkconfig.h: No such file or directory
    # board_build.arduino.memory_type: qspi_opi
    board_build.flash_size: 16MB
    board_build.flash_mode: qio
    board_build.flash_type: qspi
    board_build.psram_type: opi
    board_build.memory_type: qspi_opi
    board_build.boot_freq: 80m

esp32:
  variant: esp32s3
  board: esp32-s3-devkitc-1

  framework:
    type: arduino
    version: 2.0.3
    platform_version: 5.1.1
```

With the above, basic ESP32 functionality like OTA updates work normally.
It's easy to detect physical button presses:
```
binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO21 #was GPIO39 on the previous board
      inverted: true
    name: "Button 1"
    on_press:
      - logger.log: PhysButton Pressed
```

Hopefully someone will have time to adapt the screen library.

nickolay, 2022-12-03:
> I'm toying with approach similar to the one taken by tiaanv...

OK, the screen appears to work for me: https://github.com/nickolay/esphome-lilygo-t547plus

besanur, 2023-02-10:
How did you flash, i was not able to flash it via esphome web. Only worked with [esptool](https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/basic-commands.html#erase-flash-erase-flash-erase-region). I have a basic configuration to display the time, but it shows a wrong time, hour is alwas at 01:XX.
And if a look at esphome web ui in home-assistant the devices is not connected, but after i connect it to my computer and read logs via them after that it is connected the wifi.

My config:
```yaml
esphome:
  name: lilygo
  platformio_options:
    # Unless noted otherwise, based on https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/1eb6119fc31fcff7a6bafecb09f4225313859fc5/examples/demo/platformio.ini#L37
    upload_speed: 921600
    monitor_speed: 115200
    board_build.mcu: esp32s3
    board_build.f_cpu: 240000000L
    board_build.arduino.memory_type: qspi_opi
    board_build.flash_size: 16MB
    board_build.flash_mode: qio
    board_build.flash_type: qspi
    board_build.psram_type: opi
    board_build.memory_type: qspi_opi
    board_build.boot_freq: 80m
    build_flags: # the first three defines are required for the screen library to function.
      - "-DBOARD_HAS_PSRAM"
      - "-DARDUINO_RUNNING_CORE=0" # TODO: this conflicts with the value from platformio's idedata, spewing a lot of warnings during the build.
      - "-DARDUINO_EVENT_RUNNING_CORE=0" # and this too
      # In addition to lilygo's settings:
      # To enable reading logs over USB until `hardware_uart: USB_CDC` support
      # is added to `logger:`, as detailed in <https://github.com/esphome/feature-requests/issues/1906>:
      - "-DARDUINO_USB_MODE=1"
      - "-DARDUINO_USB_CDC_ON_BOOT=1"

esp32:
  variant: esp32s3
  board: esp32-s3-devkitc-1

  framework:
    type: arduino
    # Just like in <https://community.home-assistant.io/t/enable-usb-cdc-to-log-hello-world-to-esp32-s3-dev-board-for-esphome/463164/10>
    # I had problems with newer versions; the following combination happens to work, so using it for now.
    version: 2.0.3
    platform_version: 5.1.1

logger:
  level: VERBOSE
  # hardware_uart: USB_CDC  # see note about <https://github.com/esphome/feature-requests/issues/1906> above

# Enable Home Assistant API
api:
  encryption:
    key: "rZjS7VS1G5PPFc7JLfwWYGMZ8LVg81Mokz9UUoCel00="

ota:
  password: "6903efa08992bfb72e6e2545357799ae"
  
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
    # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Lilygo-T5 Fallback Hotspot"
    password: "xHMGPUOTHBjX"

captive_portal:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO21 #was GPIO39 on the previous board
      inverted: true
    name: "Button 1"
    on_press:
      - logger.log: PhysButton Pressed

external_components:
  source:
    type: local
    path: esphome-lilygo-t547plus/components/
  components: ["t547"]

time:
  - platform: homeassistant
    id: esptime

font: 
  - file: "fonts/Roboto-Regular.ttf"
    id: font_clock
    size: 78
    glyphs: [':', '0', '1', '2', '3', '4', '5', '6', '7', '8', '9']

display:
  - platform: t547
    # rotation: 180
    update_interval: 30s
    lambda: |-
      it.strftime(it.get_width()/2, it.get_height()/2, id(font_clock), TextAlign::TOP_CENTER, "%H:%M", id(esptime).now());

```

lehovec, 2023-02-10:
You can add platformio board configuration from [LilyGo-EPD47 GitHub repository](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47) to esphome. I added [lilygo-t5-74-plus.json](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/platformio/boards/lilygo-t5-47-plus.json) file to `/config/esphome/.esphome/build/[your-esphome-device]/boards` and I am am able to flash without problems.

```
esphome:
  name: liligo-t5-47-epaper-plus
  platformio_options:
    board_dir: boards
    #include https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/platformio/boards/lilygo-t5-47-plus.json as custom board
    board: lilygo-t5-47-plus

esp32:
  variant: esp32s3
  board: lilygo-t5-47-plus

  framework:
    type: arduino
    # You don't need to specify version and platform_version

i2c:
  sda: GPIO17
  scl: GPIO18
  # There is some problems with i2c scan so turn scan off if problem appear on your board
  scan: false
  id: bus_a

external_components:
  - source: github://nickolay/esphome-lilygo-t547plus
    components: ["t547"]

display:
- platform: t547
  id: t5_display
  update_interval: 30s
```

nickolay, 2023-02-10:
> i was not able to flash it via esphome web. Only worked with [esptool](https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/basic-commands.html)

I'm not sure how you can provide the necessary platformio_options via esphome web, so I'm flashing using the CLI (`esphome run basic.yaml`), as noted in the [readme](https://github.com/nickolay/esphome-lilygo-t547plus).

I'm always flashing over USB to be able to see the [logs](https://gist.github.com/nickolay/cf171d77777bf4f4e0b5bca115c57ff4) from the first startup, which are very useful to troubleshoot the occasional problems, like the device booting into safe mode (i.e. ignoring most of the config). I've also added the device to Home assistant manually.

> You can add platformio board configuration from [LilyGo-EPD47 GitHub repository](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47) to esphome

Cool tip, thanks!

besanur, 2023-02-10:
> You can add platformio board configuration from [LilyGo-EPD47 GitHub repository](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47) to esphome. I added [lilygo-t5-74-plus.json](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/platformio/boards/lilygo-t5-47-plus.json) file to `/config/esphome/.esphome/build/[your-esphome-device]/boards` and I am am able to flash without problems.
> 
> ```yaml
> esphome:
>   name: liligo-t5-47-epaper-plus
>   platformio_options:
>     board_dir: boards
>     #include https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/platformio/boards/lilygo-t5-47-plus.json as custom board
>     board: lilygo-t5-47-plus
> 
> esp32:
>   variant: esp32s3
>   board: lilygo-t5-47-plus
> 
>   framework:
>     type: arduino
>     # You don't need to specify version and platform_version
> 
> i2c:
>   sda: GPIO17
>   scl: GPIO18
>   # There is some problems with i2c scan so turn scan off if problem appear on your board
>   scan: false
>   id: bus_a
> 
> external_components:
>   - source: github://nickolay/esphome-lilygo-t547plus
>     components: ["t547"]
> 
> display:
> - platform: t547
>   id: t5_display
>   update_interval: 30s
> ```

with this solution it works much better and without problems.. thank you

kaeltis, 2023-04-06:
Has anyone figured out how to read the battery voltage? According to https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/schematic/T5-4.7-Plus.pdf the pin should be GPIO14, but there I'm getting `ESP32S3 doesn't support ADC on this pin.`

nickolay, 2023-04-06:
@kaeltis see https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/87 for a C example, which seems to work okay. Didn’t try to use it from esphome yet.

kaeltis, 2023-04-06:
Thanks, I've forked an existing module that solved this for the old board revision and it seems to work.
I have pretty much no idea what I'm doing so this definitely needs improvements: https://github.com/kaeltis/esphome-lilygo-t547plus

startQ341, 2023-05-09:
> with this solution it works much better and without problems.. thank you

> You can add platformio board configuration from [LilyGo-EPD47 GitHub repository](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47) to esphome. I added [...]

a try your code but have a error code a try change jason file but same problem thanks for help 

Processing obrazovka (board: lilygo-t5-47-plus; framework: arduino; platform: platformio/espressif32 @ 5.3.0)
--------------------------------------------------------------------------------
Error: Could not load broken JSON: /config/esphome/.esphome/build/obrazovka/boards/lilygo-t5-47-plus.json

lehovec, 2023-05-09:
@startQ341 error indicate that lilygo-t5-47-plus.json file is somehow corrupted. Try to download it from source repository again or check in some online json validator what is problem with that file.

startQ341, 2023-05-09:
@lehovec  
thank you for your help browser corrupted data while downloading

zagnuts, 2023-07-04:
Thanks guys for this - it's working perfectly for me now, both screen and battery estimate. Only minor hassle is the lack of partial refresh support but hopefully that will become available eventually. Oh, and it would be nice to be able to disable that blinding blue light with something civilised like
```
status_led:
  pin:
    number: GPIO05
    inverted: True
```
rather than having to contemplate crushing it with pliers!

petrhod, 2023-07-16:
Please help me compile for Lilygo 4.7 pro.
I am able to program with the arduino ino and Lily works. I am not able to display anything in esphome. I always get an error message:
`Tool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
INFO Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
Error: Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3' requirements for your system 'linux_x86_64'`
My code is based on the above:
`esphome:
  name: lilygo
  platformio_options:
    # Unless noted otherwise, based on https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/1eb6119fc31fcff7a6bafecb09f4225313859fc5/examples/demo/platformio.ini#L37
    upload_speed: 921600
    monitor_speed: 115200
    board_build.mcu: esp32s3
    board_build.f_cpu: 240000000L
    board_build.arduino.memory_type: qspi_opi
    board_build.flash_size: 16MB
    board_build.flash_mode: qio
    board_build.flash_type: qspi
    board_build.psram_type: opi
    board_build.memory_type: qspi_opi
    board_build.boot_freq: 80m
    build_flags:  # the first three defines are required for the screen library to function.
      - "-DBOARD_HAS_PSRAM"
      - "-DARDUINO_RUNNING_CORE=0"  # TODO: this conflicts with the value from platformio's idedata, spewing a lot of warnings during the build.
      - "-DARDUINO_EVENT_RUNNING_CORE=0"  # and this too
      # In addition to lilygo's settings:
      # To enable reading logs over USB until `hardware_uart: USB_CDC` support
      # is added to `logger:`, as detailed in <https://github.com/esphome/feature-requests/issues/1906>:
      - "-DARDUINO_USB_MODE=1"
      - "-DARDUINO_USB_CDC_ON_BOOT=1"

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
    version: 2.1.1
    platform_version: 5.1.1

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "maXDU+zyImnTxgTD60TgWY0WUMoNS/ToE97w8WBGVPk="

ota:
  password: "92d1d4c3224f0fc6e118bffe0d4b8834"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Lilygo Fallback Hotspot"
    password: "turNzxBr7C3d"

captive_portal:

binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO21 #was GPIO39 on the previous board
      inverted: true
    name: "Button 1"
    on_press:
      - logger.log: PhysButton Pressed

external_components:
  source:
    type: local
    path: components/
  components: ["t547"]

font: 
  - file: "Google_Sans_Bold.ttf"
    id: font_clock
    size: 78
    glyphs: [':', '0', '1', '2', '3', '4', '5', '6', '7', '8', '9']

time:
  - platform: homeassistant
    id: esptime

display:
  - platform: t547
    # rotation: 180
    update_interval: 30s
    lambda: |-
      it.strftime(it.get_width()/2, it.get_height()/2, id(font_clock), TextAlign::TOP_CENTER, "%H:%M", id(esptime).now());`
Thank you in advance.

special-k-uk, 2023-07-25:
I managed to get it compiled and flashed with the below. I saved the JSON locally into `/.esphome/build/test`

`esphome:
  name: test
  friendly_name: test
  platformio_options:
    board_dir: boards
    #include https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/platformio/boards/lilygo-t5-47-plus.json as custom board
    board: lilygo-t5-47-plus

esp32:
  variant: esp32s3
  board: lilygo-t5-47-plus

  framework:
    type: arduino
    # version: 2.0.8
    # You don't need to specify version and platform_version

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "secret"

ota:
  password: "secret"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
    
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "GGH2cv2rqbUL"

captive_portal:

i2c:
  sda: GPIO18
  scl: GPIO17
  # There is some problems with i2c scan so turn scan off if problem appear on your board
  scan: true
  id: bus_a

external_components:
  - source: github://nickolay/esphome-lilygo-t547plus
    components: ["t547"]

touchscreen:
  - platform: lilygo_t5_47
    id: lilygo_touchscreen
    interrupt_pin: GPIO09
    address: 0x5A
    setup_priority: -100
    on_touch:
      then:
        - display.page.show_next: 't5_display'
        - component.update: 't5_display'
`

special-k-uk, 2023-07-26:
Has anyone managed to get the [Touchscreen ](https://esphome.io/components/touchscreen/lilygo_t5_47.html)component working with this board?

I've been using [this touch panel](https://www.aliexpress.com/item/1005002006389042.html) on the older v3 boards and I'd like to continue using it on the newer S3 based boards.

Unfortunately the interrupt_pin has changed from GPIO13 on the older v3 to GPIO9 on the newer S3 board, and it looks like the Touchscreen Component is fixed at GPIO13, although it requires you to specify the pin in yaml.

I've logged an [bug report here](https://github.com/esphome/issues/issues/4742), just wondering if anyone has found a way to use touch?

htitze, 2023-08-16:
> 

> Please help me compile for Lilygo 4.7 pro. I am able to program with the arduino ino and Lily works. I am not able to display anything in esphome. I always get an error message: `Tool Manager: [...]

Same problem for me...

Fabian-Schmidt, 2023-08-18:
Hi,

I got the display working with the following config:
Still having issues with logging and display is not perfect.

```yaml
esphome:
  name: esp-t5-s3
  friendly_name: "T5 S3"
  platformio_options:
    board_build.f_flash: 80000000L
    board_upload.flash_size: 16MB
    board_build.flash_mode: qio
    board_build.psram_type: opi
    board_build.partitions: default_16MB.csv
    board_build.arduino.memory_type: qio_opi
  on_boot:
    then:
      # read time from external source (connected via I2C)
      - pcf85063.read_time

external_components:  - source: github://Fabian-Schmidt/esphome-lilygo_t5_47_display
    components: ["lilygo_t5_47_s3_display"]
    refresh: 0s
  - source: github://Fabian-Schmidt/esphome@jesserockz-2023-136-Fabian-Schmidt
    components: [ logger ]

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
    version: latest

logger:
  hardware_uart: USB_CDC # USB_CDC # USB_SERIAL_JTAG

i2c:
  sda: 18
  scl: 17
  frequency: 10kHz
  scan: false
  id: i2c_bus_a

time:
  - platform: pcf85063
    id: esptime
    update_interval: never

binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO21
      inverted: true
    name: "Button 1"
    on_press:
    - lambda:
        // Code here

sensor:
  - platform: internal_temperature
    id: esptemp
    name: "Internal Temperature"
    update_interval: 1s

font:
  - file: Google_Sans_Bold.ttf
    id: font_name
    size: 38

display:
- platform: lilygo_t5_47_s3_display
  update_interval: 10s
  # rotation: 180
  # clear: false
  full_update_every: 6
  # power_off_delay_enabled: false
  lambda: 
    it.printf(100, 100, id(font_name), TextAlign::BASELINE_CENTER, "Hello World");
```

petrhod, 2023-08-19:
Hi,  for my i ok this  ....

`esphome:
  name: esp-t5-s3
  friendly_name: "T5 S3"
  platformio_options:
    board_build.f_flash: 80000000L
    board_upload.flash_size: 16MB
    board_build.flash_mode: qio
    board_build.psram_type: opi
    board_build.partitions: default_16MB.csv
    board_build.arduino.memory_type: qio_opi
  on_boot:
    then:
      # read time from external source (connected via I2C)
      # - pcf85063.read_time

external_components:
  - source: github://nickolay/esphome-lilygo-t547plus
    components: ["t547"]

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
    version: latest

# Enable Home Assistant API
api:
  encryption:
    key: "2DvwSZRhQbXP3y1jVlTR5wg9FBPCrbMVVmKikLNFcXw="

ota:
  password: "0b4f13991d00618a668dbc3d78fa09ac"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

#i2c:
#  sda: 18
#  scl: 17
#  frequency: 10kHz
#  scan: false
#  id: i2c_bus_a

#deep_sleep:
#  run_duration: 30s
#  sleep_duration: 1min

time:
  - platform: homeassistant
    id: homeassistant_time

binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO21
      inverted: true
    name: "Button 1"
    on_press:
    - lambda:
        // Code here

#sensor:
#  - platform: internal_temperature
#    id: esptemp
#    name: "Internal Temperature"
#    update_interval: 1s

font:
  - file: Google_Sans_Bold.ttf
    id: font_1
    size: 45
  - file: Google_Sans_Bold.ttf
    id: font_2
    size: 150

display:
- platform: t547
  update_interval: 60s
  # rotation: 180
  #clear: true
  #full_update_every: 3
  #power_off_delay_enabled: true
  lambda: 
    it.printf(10, 100, id(font_1), " Hello World");
 
  
    `

VladimirV01, 2023-11-11:
Hello Guys, I wanted to say thanks to everybody who provided help in this thread. 
It was very helpful and helped me get my Lilygo T5 Plus going. 

However, no matter how hard I tried I didn't manage to establish the connection with the RTC. 
Has anyone succeeded with this task?

special-k-uk, 2023-11-14:
> Hello Guys, I wanted to say thanks to everybody who provided help in this thread. It was very helpful and helped me get my Lilygo T5 Plus going.
> 
> However, no matter how hard I tried, I could not manage to establish the connection with the RTC. Has anyone succeeded with this task?

Hey @VladimirV01 
The PFC8563 is a core component now , although I think the hardware itself might be a bit flakey as it works 90% of the time for me. The other 10% of the time it will throw an error on boot that it's unable to connect.

Here's what I have 
```
esphome:
  name: barn-blind-controller
  friendly_name: barn-blind-controller
  platformio_options:
    board_build.f_flash: 80000000L
    board_upload.flash_size: 16MB
    board_build.flash_mode: qio
    board_build.psram_type: opi
    board_build.partitions: default_16MB.csv
    board_build.arduino.memory_type: qio_opi 
  on_boot:
    priority: -100
    then:
      - pcf8563.read_time
```
then
```
i2c:
  - id: bus_a
    sda: GPIO18
    scl: GPIO17
    # There is some problems with i2c scan so turn scan off if problem appear on your board
    scan: False
    frequency: 100khz

time:
  - platform: pcf8563
    id: internal_clock
    address: 0x51  
    i2c_id: bus_a
  - platform: homeassistant
    id: external_clock
    on_time_sync:
      then:
        - pcf8563.write_time
```

I hope that helps?

VladimirV01, 2023-11-15:
> > Hello Guys, I wanted to say thanks to everybody who provided help in this thread. It was very helpful and helped me get my Lilygo T5 Plus going.
> > However, no matter how hard I tried, I could not manage to establish the connection with the RTC. Has anyone succeeded with this task?
> 
> Hey @VladimirV01 The PCF8563 is a core component now , although I think the hardware itself might be a bit flakey as it works 90% of the time for me. The other 10% of the time it will throw an error on boot that it's unable to connect.
> 
> Here's what I have
> 
> ```
> esphome:
>   name: barn-blind-controller
>   friendly_name: barn-blind-controller
>   platformio_options:
>     board_build.f_flash: 80000000L
>     board_upload.flash_size: 16MB
>     board_build.flash_mode: qio
>     board_build.psram_type: opi
>     board_build.partitions: default_16MB.csv
>     board_build.arduino.memory_type: qio_opi 
>   on_boot:
>     priority: -100
>     then:
>       - pcf8563.read_time
> ```
> 
> then
> 
> ```
> i2c:
>   - id: bus_a
>     sda: GPIO18
>     scl: GPIO17
>     # There is some problems with i2c scan so turn scan off if problem appear on your board
>     scan: False
>     frequency: 100khz
> 
> time:
>   - platform: pcf8563
>     id: internal_clock
>     address: 0x51  
>     i2c_id: bus_a
>   - platform: homeassistant
>     id: external_clock
>     on_time_sync:
>       then:
>         - pcf8563.write_time
> ```
> 
> I hope that helps?

Hello @special-k-uk !

Thank you very much for taking the time to check this issue and write an answer. 

I have followed your advice and checked your configuration. 
After changing the frequency to **100KHz** and setting the address manually to **0x51** as advised by you, I don't see the initialization error on boot of the integrated RTC anymore! 

Now I see that it syncs the RTC on time sync with homeassistant, therefore I can be sure it works.

ex-nerd, 2023-11-18:
I managed to piece together a working config for my new board that includes a bunch of the changes, including the display code from @nickolay and battery stuff from @kaeltis (you two should consider merging your work for https://github.com/nickolay/esphome-lilygo-t547plus/issues/3). I also activated the internal temp sensor, and set things to read/write with the internal clock. So far, everything is working great for me with an old LiPo battery I had sitting around (battery sensor is reading 4.5V, which lines up with what it did with the LilyGo firmware).

Anyway, here it is for posterity (along with some notes since I just copy/pasted my yaml file here)

```yaml
# See: https://github.com/esphome/feature-requests/issues/1960
# See for older hardware: https://github.com/esphome/feature-requests/issues/1109

esphome:
  name: lilygo-t5-eink
  friendly_name: LilyGo T5 eInk
  platformio_options:
    board_build.f_flash: 80000000L
    board_upload.flash_size: 16MB
    board_build.flash_mode: qio
    board_build.psram_type: opi
    board_build.partitions: default_16MB.csv
    board_build.arduino.memory_type: qio_opi
  on_boot:
    priority: -100
    then:
      - pcf8563.read_time

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
    version: latest

# Enable logging
logger:

# Enable Home Assistant API (API password is deprecated in favor of encryption key)
# https://esphome.io/components/api.html
api:
  encryption:
    key: !secret home_assistant_encryption_key

ota:
  password: !secret ota_password

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
  reboot_timeout: 15min

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Lilygo-T5-Eink Fallback Hotspot"
    password: !secret fallback_ssid_password

captive_portal:

external_components:
  # https://github.com/nickolay/esphome-lilygo-t547plus
  - source: github://nickolay/esphome-lilygo-t547plus
    components: ["t547"]
  # https://github.com/kaeltis/esphome-lilygo-t547plus
  - source: github://kaeltis/esphome-lilygo-t547plus
    components: ["lilygo_t5_47_battery"]

#deep_sleep:
#  run_duration: 30s
#  sleep_duration: 1min

i2c:
  - id: bus_a
    sda: GPIO18
    scl: GPIO17
    frequency: 100khz
    # There is some problems with i2c scan so turn scan off if problem appear on your board
    scan: False

time:
  - platform: pcf8563
    id: internal_clock
    address: 0x51  
    i2c_id: bus_a
  - platform: homeassistant
    id: ha_time
    on_time_sync:
      then:
        - pcf8563.write_time

binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO21 #was GPIO39 on the previous board
      inverted: true
    name: "Button 1"
    on_press:
      - logger.log: PhysButton Pressed

sensor:
  - platform: internal_temperature
    id: esptemp
    name: "Internal Temperature"
    update_interval: 1s

  - platform: lilygo_t5_47_battery
    id: battery_voltage
    voltage:
      name: "Battery Voltage"

  - platform: template
    name: "Battery Percentage"
    id: battery_percentage
    lambda: |-
      // tweak values in mapping for calibration
      // 4.1 = max voltage
      // 3.3 = min voltage
      float y = (id(battery_voltage).voltage->state - 3.3) * 100.0 / (4.1 - 3.3);
      if (y < 100.0) { return y; } else { return 100.0; };

font:
  # https://github.com/hprobotic/Google-Sans-Font/blob/master/
  - file: fonts/GoogleSans-Bold.ttf
    id: font_1
    size: 45
  - file: fonts/GoogleSans-Bold.ttf
    id: font_2
    size: 150
  - file: "fonts/GoogleSans-Bold.ttf"
    id: font_clock
    size: 300
    glyphs: [':', '0', '1', '2', '3', '4', '5', '6', '7', '8', '9']

display:
- platform: t547
  update_interval: 60s
  # rotation: 180
  #clear: true
  #full_update_every: 3
  #power_off_delay_enabled: true
  lambda: |-
    it.printf(10, 50, id(font_1), "Battery: %.1f% (%.1fV)", id(battery_percentage).state, id(battery_voltage).voltage->state);
    // https://esphome.io/cookbook/display_time_temp_oled.html 
    it.printf(10, 100, id(font_1), "Internal Temp: %.1f°C", id(esptemp).state);

    it.strftime(it.get_width()/2, it.get_height()/2-150, id(font_clock), TextAlign::TOP_CENTER, "%H:%M", id(ha_time).now());
```

ThierryDi, 2023-11-21:
Hello,
Thanks a lot to all this threads' participants for their work which helped me piece together my own config. (I won't post my code as it doesn't bring anything new to the table ;-))
I was wondering if any of you managed to print in grayscale ?
Cheers

PabloFFF93, 2023-12-11:
Has anyone had such a problem

`INFO ESPHome 2023.11.6
INFO Reading configuration /config/esphome/test.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
Failed config

esp32: None
  
  Please specify flash_size within esp32 configuration only.
  board: esp32-s3-devkitc-1
  framework: 
    version: 2.0.9
    source: ~3.20009.0
    platform_version: platformio/espressif32@5.4.0
    type: arduino
  flash_size: 4MB
  variant: ESP32S3`

zagnuts, 2023-12-14:
> Has anyone had such a problem
> Failed config
> Please specify flash_size within esp32 configuration only.

Try commenting out "board_upload.flash_size" from within the platformio_options section. It will use the default, which should be fine. :-)

PabloFFF93, 2023-12-20:
There is some progress...
It is installed.

It doesn't start, the USB logs show this:
[image]

Januznl, 2023-12-26:
> There is some progress... It is installed.
> 
> It doesn't start, the USB logs show this: <img alt="image" width="287" src="https://private-user-images.githubusercontent.com/66909169/292027030-95a8704d-cbbe-4367-bd8f-1b1f28730cdb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM1ODY0NTEsIm5iZiI6MTcwMzU4NjE1MSwicGF0aCI6Ii82NjkwOTE2OS8yOTIwMjcwMzAtOTVhODcwNGQtY2JiZS00MzY3LWJkOGYtMWIxZjI4NzMwY2RiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI2VDEwMjIzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThmNzBhOTIwNmUwYzI5ZTE5NGRjODdiZDA3NjJkNWMzZGY1ZWM4YWUxZjgwY2Y4MDI5ZWRjMTU4OWIyZTllOWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.-qEWvnwBzqcuJAUIePwXSfnemv3LoVwpCgJdwePHLQw">
> 
> And the browser is not responding

Have you tried ESPHome version: 2023.12.5?
I had the same issues, but since i've updated to the latest version its working.

Steps i took:
- click on Install (plug into this computer (usb)
- Use esphome web to flash 
- i had to bring the arduino in boot mode by holding the button "STR_100" and press the button "REST"
- after flashing i presses "REST" again and the board started the esphome firmware.

I'm currently creating a repo with working clock example and steps to take to get the S3 Plus running. 
(link will follow)

PabloFFF93, 2023-12-28:
yeah!  Your way works!  now my entities are available on eink!
[image]

Januznl, 2023-12-28:
@PabloFFF93 did you use the guide? or upgrading and the boot steps?

PabloFFF93, 2023-12-28:
I used the guide. I didn't really need to special  boot ESP by key.. Everything became clear after using the new board and uploading your yaml ESPHOME: and ESP32: conf. I think this was the problem.

I have a similar problem in esp32 by wroover. I have to click boot to initialize the boot

Januznl, 2023-12-28:
> I used the guide. I didn't really need to special boot ESP by key.. Everything became clear after using the new board and uploading your yaml ESPHOME: and ESP32: conf. I think this was the problem.
> 
> I have a similar problem in esp32 by wroover. I have to click boot to initialize the boot

nice, good news its working!

nickolay, 2023-12-30:
> You can add platformio board configuration from LilyGo-EPD47 GitHub repository to esphome. I added lilygo-t5-74-plus.json file to /config/esphome/.esphome/build/[your-esphome-device]/boards and I am [...]

I finally looked into how this works, because I couldn't understand why I can't use an absolute path for the `esphome.platformio_options.board_dir`. **Turns out if you're using the custom board json, the remaining `platformio_options` do not have any effect and can be dropped.**

Details for the curious:
> * `esphome.platformio_options` [are copied](https://github.com/esphome/esphome/blob/5ebb68f4ffde847b3ac313d941f784ec4b0a4d44/esphome/writer.py#L149) to the [`[env:NAME]` section of `platformio.ini`](https://docs.platformio.org/en/latest/projectconf/sections/env/index.html), while the [`boards_dir`](https://docs.platformio.org/en/latest/projectconf/sections/platformio/options/directory/boards_dir.html) option (note the plural form!) is expected in the `[platformio]` section.
> * `board_dir` is a typo, which is [not reported only because](https://github.com/platformio/platformio-core/blob/14907579cd35b0bb106209effad87388e867575f/platformio/project/config.py#L156) it starts with `board_`, which is a _custom option prefix_, which is a [mechanism allowing to override any path of the JSON board manifest](https://docs.platformio.org/en/stable/projectconf/sections/env/options/platform/index.html) — `board_dir: boards` effectively specifies a top-level `"dir": "boards"` key in terms of the [JSON manifest](https://docs.platformio.org/en/latest/platforms/creating_board.html).
> * Specifying `esphome.platformio_options.board` is unnecessary, because it [gets set to the value of `esp32.board`](https://github.com/esphome/esphome/blob/5ebb68f4ffde847b3ac313d941f784ec4b0a4d44/esphome/components/esp32/__init__.py#L449) anyway.

In other news:
* I found out the cause of occasional `task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time: IDLE (CPU 0)` crashes, that made me pin an old arduino version last year: https://github.com/esphome/issues/issues/5317 - you can try removing `-DARDUINO_RUNNING_CORE=0` from the board JSON if you experience this.
* If you still have old arduino versions pinned, you might get `Error: Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch2' requirements for your system 'linux_x86_64'`, there's a workaround over at https://github.com/nickolay/esphome-lilygo-t547plus/pull/6
* Looks like mainline epdiy won't support this board anytime soon: https://github.com/vroland/epdiy/issues/203#issuecomment-1871886210
* I opened https://github.com/platformio/platform-espressif32/issues/1269 for adding the board JSON to the list of built-in boards, so that you don't have to download it separately

dxdr, 2024-03-11:
Thank you everyone for your help getting this set up... one thing that I can't seem to figure out is how to set up external buttons.  I was able to get one set up on GPIO 15 using the pins in the molex connector, but can't seem to get another working on any of the other pins.  The FAQ says that 12, 13, 14 and 15 should be available if not using the touch screen, but I can't seem to figure out how to physically connect the button to the board.  I thought I could use the 40 pin connector, but I tested each pin and none seem to produce a response when I use:

`  - platform: gpio
    pin:
      number: GPIO13
      inverted: false
      mode:
        input: true
        pullup: true
    name: "${esp_name} Button 13"
    id: button13
    on_click: 
      then:
        - logger.log: "Button 13 Pressed..."`

I'm sure I'm missing something simple, but if anyone has an idea I would appreciate it.

husnainali09, 2024-04-04:
Have been trying to get 4.7 Plus to work. Keep getting ESPTIME error.

`substitutions:
  name: esphome-web-60e604
  friendly_name: Lilygodash

esphome:
  name: lilygodash
  platformio_options:
    # Include https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/platformio/boards/lilygo-t5-47-plus.json as custom board
    board: lilygo-t5-47-plus
    # Directory where the lilygo-t5-47-plus.json file can be found
    board_dir: boards
#  on_boot:
#    priority: -100
#    then:
#      - pcf8563.read_time
  # custom build path, so we can save the lilygo-t5-47-plus.json file in the sub-folder "boards"
  build_path: /config/esphome/.esphome/basic-clock/
  includes:
    - common.h

esp32:
  variant: esp32s3
  board: lilygo-t5-47-plus
  framework:
    type: arduino

logger:

api:
  encryption:
    key: "B1DK4ps4YiUJ0Vit1S4jFxalSAOwZjNvr6QhFeWjngI="

ota:
  password: "B1DK4ps4YiUJ0Vit1S4jFxalSAOwZjNvr6QhFeWjngI="

## External components for T5 support ##
external_components:
  # https://github.com/nickolay/esphome-lilygo-t547plus
  - source: github://nickolay/esphome-lilygo-t547plus
    components: ["t547"]
  # https://github.com/kaeltis/esphome-lilygo-t547plus
  - source: github://kaeltis/esphome-lilygo-t547plus
    components: ["lilygo_t5_47_battery"]

## -- ##

## Only needed if powered via battery ##
#deep_sleep:
#  run_duration: ${run_time}
#  sleep_duration: ${sleep_time}
#  id: deep_sleep_1
#  esp32_ext1_wakeup:
#    pins: GPIO39
#    mode: ALL_LOW
#    allow_other_uses: true
## -- ##

wifi:
  networks:
    - ssid: NETGEAR08_IoT
      password: abbaabba12
  reboot_timeout: 15min
  use_address: 192.168.1.245

web_server:
  port: 80
  include_internal: true 

[... 667 lines cut ...]
In file included from src/esphome/components/display/display.h:10,
                 from src/esphome.h:18,
                 from src/main.cpp:3:
src/esphome/core/time.h:17:8: note:   'esphome::ESPTime'
 struct ESPTime {
        ^~~~~~~
In file included from src/main.cpp:104:
src/common.h:4:73: error: expected primary-expression before 'nameday'
 std::string generateDateFormat(esphome::time::ESPTime time, std::string nameday) {
                                                                         ^~~~~~~
src/common.h:4:82: error: expected ',' or ';' before '{' token
 std::string generateDateFormat(esphome::time::ESPTime time, std::string nameday) {
                                                                                  ^
Compiling .pioenvs/lilygodash/lib64d/WiFi/WiFiMulti.cpp.o
/config/esphome/esphome-web-60e604.yaml: In lambda function:
/config/esphome/esphome-web-60e604.yaml:506:70: error: no match for call to '(std::__cxx11::string {aka std::__cxx11::basic_string<char>}) (esphome::ESPTime&, std::__cxx11::string&)'
       std::string dateFormat = generateDateFormat(time, id(nevnap).state);
                                                                      ^
/config/esphome/esphome-web-60e604.yaml:521:24: error: 'esphome::time::ha_time' has not been declared
         auto t = time::ha_time::from_c_tm(&tp, c_t);
                        ^~~~~~~
*** [.pioenvs/lilygodash/src/main.cpp.o] Error 1
========================== [FAILED] Took 2.95 seconds ==========================`

Can anyone assist with my code please.

husnainali09, 2024-04-04:
'''INFO ESPHome 2024.3.1
INFO Reading configuration /config/esphome/esphome-web-60e604.yaml...
INFO Detected timezone 'Australia/Brisbane'
INFO Detected timezone 'Australia/Brisbane'
INFO Generating C++ source...
INFO Compiling app...
Processing lilygodash (board: lilygo-t5-47-plus; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing Wire
INFO Installing Wire
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
|-- Wire @ 2.0.0
|-- LilyGoEPD47 @ 0.3.0+sha.0f31685
Compiling .pioenvs/lilygodash/src/main.cpp.o
Compiling .pioenvs/lilygodash/lib64d/WiFi/WiFiGeneric.cpp.o
In file included from src/main.cpp:104:
src/common.h:4:47: error: 'ESPTime' is not a member of 'esphome::time'
 std::string generateDateFormat(esphome::time::ESPTime time, std::string nameday) {
                                               ^~~~~~~
src/common.h:4:47: note: suggested alternative:
In file included from src/esphome/components/display/display.h:10,
                 from src/esphome.h:18,
                 from src/main.cpp:3:
src/esphome/core/time.h:17:8: note:   'esphome::ESPTime'
 struct ESPTime {
        ^~~~~~~
In file included from src/main.cpp:104:
src/common.h:4:73: error: expected primary-expression before 'nameday'
 std::string generateDateFormat(esphome::time::ESPTime time, std::string nameday) {
                                                                         ^~~~~~~
src/common.h:4:82: error: expected ',' or ';' before '{' token
 std::string generateDateFormat(esphome::time::ESPTime time, std::string nameday) {
                                                                                  ^
Compiling .pioenvs/lilygodash/lib64d/WiFi/WiFiMulti.cpp.o
/config/esphome/esphome-web-60e604.yaml: In lambda function:
/config/esphome/esphome-web-60e604.yaml:506:70: error: no match for call to '(std::__cxx11::string {aka std::__cxx11::basic_string<char>}) (esphome::ESPTime&, std::__cxx11::string&)'
       std::string dateFormat = generateDateFormat(time, id(nevnap).state);
                                                                      ^
/config/esphome/esphome-web-60e604.yaml:521:24: error: 'esphome::time::ha_time' has not been declared
         auto t = time::ha_time::from_c_tm(&tp, c_t);
                        ^~~~~~~
*** [.pioenvs/lilygodash/src/main.cpp.o] Error 1
========================== [FAILED] Took 2.95 seconds =========================='''

husnainali09, 2024-04-04:
Trying to get homeassistant dashboard working on T5 ESP32-3. Keep getting esptime error. Anyone know how to fix this ?

```substitutions:
  name: esphome-web-60e604
  friendly_name: Lilygodash

esphome:
  name: lilygodash
  platformio_options:
    # Include https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/platformio/boards/lilygo-t5-47-plus.json as custom board
    board: lilygo-t5-47-plus
    # Directory where the lilygo-t5-47-plus.json file can be found
    board_dir: boards
#  on_boot:
#    priority: -100
#    then:
#      - pcf8563.read_time
  # custom build path, so we can save the lilygo-t5-47-plus.json file in the sub-folder "boards"
  build_path: /config/esphome/.esphome/basic-clock/
  includes:
    - common.h

esp32:
  variant: esp32s3
  board: lilygo-t5-47-plus
  framework:
    type: arduino

logger:

api:
  encryption:
    key: "B1DK4ps4YiUJ0Vit1S4jFxalSAOwZjNvr6QhFeWjngI="

ota:
  password: "B1DK4ps4YiUJ0Vit1S4jFxalSAOwZjNvr6QhFeWjngI="

## External components for T5 support ##
external_components:
  # https://github.com/nickolay/esphome-lilygo-t547plus
  - source: github://nickolay/esphome-lilygo-t547plus
    components: ["t547"]
  # https://github.com/kaeltis/esphome-lilygo-t547plus
  - source: github://kaeltis/esphome-lilygo-t547plus
    components: ["lilygo_t5_47_battery"]

## -- ##

## Only needed if powered via battery ##
#deep_sleep:
#  run_duration: ${run_time}
#  sleep_duration: ${sleep_time}
#  id: deep_sleep_1
#  esp32_ext1_wakeup:
#    pins: GPIO39
#    mode: ALL_LOW
#    allow_other_uses: true
## -- ##

wifi:
  networks:
    - ssid: NETGEAR08_IoT
      password: abbaabba12
  reboot_timeout: 15min
  use_address: 192.168.1.245

web_server:
  port: 80
  include_internal: true 

[... 665 lines cut ...]
                                               ^~~~~~~
src/common.h:4:47: note: suggested alternative:
In file included from src/esphome/components/display/display.h:10,
                 from src/esphome.h:18,
                 from src/main.cpp:3:
src/esphome/core/time.h:17:8: note:   'esphome::ESPTime'
 struct ESPTime {
        ^~~~~~~
In file included from src/main.cpp:104:
src/common.h:4:73: error: expected primary-expression before 'nameday'
 std::string generateDateFormat(esphome::time::ESPTime time, std::string nameday) {
                                                                         ^~~~~~~
src/common.h:4:82: error: expected ',' or ';' before '{' token
 std::string generateDateFormat(esphome::time::ESPTime time, std::string nameday) {
                                                                                  ^
Compiling .pioenvs/lilygodash/lib64d/WiFi/WiFiMulti.cpp.o
/config/esphome/esphome-web-60e604.yaml: In lambda function:
/config/esphome/esphome-web-60e604.yaml:506:70: error: no match for call to '(std::__cxx11::string {aka std::__cxx11::basic_string<char>}) (esphome::ESPTime&, std::__cxx11::string&)'
       std::string dateFormat = generateDateFormat(time, id(nevnap).state);
                                                                      ^
/config/esphome/esphome-web-60e604.yaml:521:24: error: 'esphome::time::ha_time' has not been declared
         auto t = time::ha_time::from_c_tm(&tp, c_t);
                        ^~~~~~~
*** [.pioenvs/lilygodash/src/main.cpp.o] Error 1
========================== [FAILED] Took 2.95 seconds ==========================```

zagnuts, 2024-04-08:
Is it possible that the bit that goes:
```
  build_path: /config/esphome/.esphome/basic-clock/
  includes:
    - common.h
``` 
Is throwing the proverbial wrench in the gears? Not sure if this is of any help, but the following works for me:

```
esphome:
  name: ${esp_hostname}
  friendly_name: ${esp_hostname}
  platformio_options:
    board_build.f_flash: 80000000L
#    board_upload.flash_size: 16MB
    board_build.flash_mode: qio
    board_build.psram_type: opi
    board_build.partitions: default_16MB.csv
    board_build.arduino.memory_type: qio_opi
  on_boot:
    priority: -100
    then:
      - pcf8563.read_time

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
    version: latest
``` 

Cheers

xxLeoxx93, 2024-04-28:
Did anyone get the touchscreen to work?
Used this but no luck so far:
```
touchscreen:
  - platform: lilygo_t5_47
    id: lilygo_touchscreen
    interrupt_pin: GPIO47
    address: 0x5A
```
and 

```
  - platform: touchscreen
    name: Top Left Touch Button
    x_min: 0
    x_max: 1000
    y_min: 0
    y_max: 1000
    on_press:
      - logger.log: Touch1 Pressed
```

alexwide, 2024-04-30:
@xxLeoxx93  i foud a working solution for the touchscreen

```
touchscreen:
  - platform: gt911
    id: lilygo_touchscreen
    interrupt_pin: GPIO47
    address: 0x5D
    i2c_id: bus_a
    setup_priority: -100
    on_touch:
      then:
        - component.update: t5_display
```

xxLeoxx93, 2024-05-05:
@alexwide 
Thank you so much! Works perfectly!
How dod you find the platform: gt911 part?
I also tried the same adress and everything but i was missing the correct platform.
Thanks again!

PabloFFF93, 2024-05-07:
Hi,
For a while everything worked fine.
I wanted to upload the change and now this appears:
`INFO Compiling app...
Processing t5lily (board: lilygo_t_display_s3; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
UnknownBoard: Unknown board ID 'lilygo_t_display_s3'`

What to do now?

ex-nerd, 2024-05-09:
> one thing that I can't seem to figure out is how to set up external buttons

@dxdr You're doing better than me. Can you share an image of how you hooked things up to the molex connector? I was trying for capacitive touch that the S3 is supposed to support (I got as far as filing a bug: https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/113) but at this point I'd love even a single external button.

dxdr, 2024-05-19:
> > one thing that I can't seem to figure out is how to set up external buttons
> 
> @dxdr You're doing better than me. Can you share an image of how you hooked things up to the molex connector? I was trying for capacitive touch that the S3 is supposed to support (I got as far as filing a bug: [Xinyuan-LilyGO/LilyGo-EPD47#113](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/113)) but at this point I'd love even a single external button.

Sure.  
[image]

husnainali09, 2024-06-03:
> INFO Compiling app...
> Processing t5lily (board: lilygo_t_display_s3; framework: arduino; platform: platformio/espressif32@5.4.0)
> 
> UnknownBoard: Unknown board ID 'lilygo_t_display_s3'

Im suddenly getting this error too now when trying to compile. Any solutions ?

arminecek, 2024-06-11:
> > INFO Compiling app...
> > Processing t5lily (board: lilygo_t_display_s3; framework: arduino; platform: platformio/espressif32@5.4.0)
> > UnknownBoard: Unknown board ID 'lilygo_t_display_s3'
> 
> Im suddenly getting this error too now when trying to compile. Any solutions ?

wrong board? must be "lilygo-t5-47-plus"

arminecek, 2024-06-11:
Hi, 
What's your experience with pixel burning? I have sings of burned black/white pixels after one month of use my home dashboard.  update_interval : 60s, two pages....
I tried removing it by uploading the firmware created  by Xiyuan´s (screen_repair) but unsuccessfully.

Is posible to buy only new screen - without S3 board?

lehovec, 2024-06-11:
> Hi, What's your experience with pixel burning? I have sings of burned black/white pixels after one month of use my home dashboard. update_interval : 60s, two pages.... I tried removing it by [...]

No problem with my display after more than a year at all, but I have a refresh rate of 20 minutes. The board is basically just a bunch of different manufacturer components got together, so if you find just a display it is easy to change it.

HansiJung, 2024-08-03:
I can’t get my display to work :-(
This is the error message I get:

```
INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/lilygo-s3-tests.yaml...
WARNING This board is unknown. Make sure the chosen chip component is correct.
WARNING This board is unknown. Make sure the chosen chip component is correct.
INFO Generating C++ source...
INFO Compiling app...
Processing lilygo-s3-tests (board: lilygo-t5-47-plus; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
UnknownBoard: Unknown board ID 'lilygo-t5-47-plus'
```

HansiJung, 2024-08-04:
Ok, now I tried from
https://github.com/nickolay/esphome-lilygo-t547plus
the basic.yaml

And now I got this failure:

```
INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/lilygo-s3-tests.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing lilygo-s3-tests (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32@5.1.1)
--------------------------------------------------------------------------------
Library Manager: Installing git+https://github.com/Xinyuan-LilyGO/LilyGo-EPD47
INFO Installing git+https://github.com/Xinyuan-LilyGO/LilyGo-EPD47
git version 2.39.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-bt9bqfwr'...
error: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.
error: 4744 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
VCSBaseException: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/Xinyuan-LilyGO/LilyGo-EPD47', '/data/cache/platformio/cache/tmp/pkg-installing-bt9bqfwr']
```

zagnuts, 2024-08-04:
If this is of any help, this is the start of my code:

```# Compiled and tested on esphome 2024.7.3 and HA 2024.7.4
substitutions:
  esp_name: lilygo01 #Device Name
  esp_hostname: lilygo01 
  run_time: 60s #can be as long as needed to get data 
  sleep_time: 30min # normal sleep time
  night_sleep_time: 10h # 1st sleep time after midnight

esphome:
  name: ${esp_hostname}
  friendly_name: ${esp_hostname}
  platformio_options:
    board_build.f_flash: 80000000L
#    board_upload.flash_size: 16MB
    board_build.flash_mode: qio
    board_build.psram_type: opi
    board_build.partitions: default_16MB.csv
    board_build.arduino.memory_type: qio_opi
  on_boot:
    priority: -100
    then:
      - pcf8563.read_time

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
    version: latest

# Enable logging
# Change to avoid "Components should block for at most 20-30ms" warning messages in the log - an issue since 2023.7.0
# Not really a breaking change - it's an issue I suspect due to the device being slow and this error previously
# simply not being reported
logger:
  level: DEBUG
  logs:
    component: ERROR

api:
  encryption:
    key: !secret esphome_encryption_key

ota:
  password: !secret ota_password
  platform: esphome
  
wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
  reboot_timeout: 15min

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Lilygo-T5-Eink Fallback Hotspot"
    password: !secret wifi_password

captive_portal:

external_components:
  # https://github.com/nickolay/esphome-lilygo-t547plus
  - source: github://nickolay/esphome-lilygo-t547plus
    components: ["t547"]
  # https://github.com/kaeltis/esphome-lilygo-t547plus
  - source: github://kaeltis/esphome-lilygo-t547plus
    components: ["lilygo_t5_47_battery"]

deep_sleep:
  run_duration: ${run_time} 
  sleep_duration: ${sleep_time}
  id: deep_sleep_1
#  esp32_ext1_wakeup:
#    pins: GPIO39
#    mode: ALL_LOW

i2c:
  - id: bus_a
    sda: GPIO18
    scl: GPIO17
    frequency: 100khz
    # There is some problems with i2c scan so turn scan off if problem appear on your board
    scan: False

time:
  - platform: pcf8563
    id: internal_clock
    address: 0x51  
    i2c_id: bus_a
  - platform: homeassistant
    id: ha_time
    on_time_sync:
      then:
        - pcf8563.write_time

binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO21 #was GPIO39 on the previous board
      inverted: true
    name: "Button 1"
    on_press:
      - logger.log: PhysButton Pressed

```

Then just need to add in your display info eg:
```
font:
  # https://github.com/hprobotic/Google-Sans-Font/blob/master/
  - file: fonts/GoogleSans-Bold.ttf
    id: font_1
    size: 45

display:
- platform: t547
  id: t5_display
  update_interval: 60s
  # rotation: 180
  #clear: true
  #full_update_every: 3
  #power_off_delay_enabled: true
  lambda: |-
      it.strftime(it.get_width()/2, 0, id(font_1), TextAlign::TOP_CENTER, "%a %d/%b/%Y", id(internal_clock).now());

```

HansiJung, 2024-08-07:
Its not working.
I get the follwoing fault:

```
INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/lilygo-s3-tests.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Compiling app...
Processing lilygo-s3-tests (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing git+https://github.com/Xinyuan-LilyGO/LilyGo-EPD47
INFO Installing git+https://github.com/Xinyuan-LilyGO/LilyGo-EPD47
git version 2.39.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-ztytlm4d'...
error: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.
error: 4834 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
VCSBaseException: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/Xinyuan-LilyGO/LilyGo-EPD47', '/data/cache/platformio/cache/tmp/pkg-installing-ztytlm4d']
```

zagnuts, 2024-08-07:
OK - I don't understand where:
`Library Manager: Installing git+https://github.com/Xinyuan-LilyGO/LilyGo-EPD47
INFO Installing git+https://github.com/Xinyuan-LilyGO/LilyGo-EPD47
git version 2.39.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-ztytlm4d'...`

Is coming from. My guess is you are using different external components. Can you post your code? Otherwise just start with the example bare code I provided above and go from there. :-)

HansiJung, 2024-08-08:
Here:

```
esphome:
  name: lilygo-s3-tests
  friendly_name: Lilygo_S3_Tests

  platformio_options:
    # Unless noted otherwise, based on https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/1eb6119fc31fcff7a6bafecb09f4225313859fc5/examples/demo/platformio.ini#L37
    upload_speed: 921600
    monitor_speed: 115200
    board_build.mcu: esp32s3
    board_build.f_cpu: 240000000L
    board_build.arduino.memory_type: qspi_opi
    board_build.flash_size: 16MB
    board_build.flash_mode: qio
    board_build.flash_type: qspi
    board_build.psram_type: opi
    board_build.memory_type: qspi_opi
    board_build.boot_freq: 80m
    platform_packages:
      - "toolchain-riscv32-esp @8.4.0+2021r2-patch5"
    build_flags:  # the first three defines are required for the screen library to function.
      - "-DBOARD_HAS_PSRAM"
      - "-DARDUINO_RUNNING_CORE=0"  # TODO: this conflicts with the value from platformio's idedata, spewing a lot of warnings during the build.
      - "-DARDUINO_EVENT_RUNNING_CORE=0"  # and this too
      # In addition to lilygo's settings:
      # To enable reading logs over USB until `hardware_uart: USB_CDC` support
      # is added to `logger:`, as detailed in <https://github.com/esphome/feature-requests/issues/1906>:
      - "-DARDUINO_USB_MODE=1"
      - "-DARDUINO_USB_CDC_ON_BOOT=1"

esp32:
  variant: esp32s3
  board: esp32-s3-devkitc-1

  framework:
    type: arduino
    # Just like in <https://community.home-assistant.io/t/enable-usb-cdc-to-log-hello-world-to-esp32-s3-dev-board-for-esphome/463164/10>
    # I had problems with newer versions; the following combination happens to work, so using it for now.
    version: 2.0.3
    platform_version: 5.1.1

logger:
  level: VERBOSE
  # hardware_uart: USB_CDC  # see note about <https://github.com/esphome/feature-requests/issues/1906> above

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  - platform: esphome
    password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Lilygo-S3-Tests Fallback Hotspot"
    password: ""

font:
  # https://github.com/hprobotic/Google-Sans-Font/blob/master/
  - file: "fonts/Google-Sans-Bold.ttf"
    id: font_1
    size: 45
  - file: "fonts/Google-Sans-Bold.ttf"
    id: font_2
    size: 150
  - file: "fonts/Google-Sans-Bold.ttf"
    id: font_clock
    size: 300
    glyphs: [':', '0', '1', '2', '3', '4', '5', '6', '7', '8', '9']

binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO21 #was GPIO39 on the previous board
      inverted: true
    name: "Button 1"
    on_press:
      - logger.log: PhysButton Pressed

external_components:
  # https://github.com/nickolay/esphome-lilygo-t547plus
  - source: github://nickolay/esphome-lilygo-t547plus
    components: ["t547"]
  # https://github.com/kaeltis/esphome-lilygo-t547plus
  - source: github://kaeltis/esphome-lilygo-t547plus
    components: ["lilygo_t5_47_battery"]

display:
  - platform: t547
    # rotation: 180
    update_interval: 30s
    lambda: |-
      it.line(0, 0, 960, 540);
```

zagnuts, 2024-08-09:
Ok, so just for the heck of it - given that your code currently is not compiling for you, I'd suggest just remove your current esphome and esp32 sections and try the following:
```
esphome:
  name: ${esp_hostname}
  friendly_name: ${esp_hostname}
  platformio_options:
    board_build.f_flash: 80000000L
    board_build.flash_mode: qio
    board_build.psram_type: opi
    board_build.partitions: default_16MB.csv
    board_build.arduino.memory_type: qio_opi
  on_boot:
    priority: -100
    then:
      - pcf8563.read_time

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
    version: latest

```

HansiJung, 2024-08-14:
hmm,
I got still this failure:

```
INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/lilygo-s3-tests.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Compiling app...
Processing lilygo-s3-tests (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing git+https://github.com/Xinyuan-LilyGO/LilyGo-EPD47
INFO Installing git+https://github.com/Xinyuan-LilyGO/LilyGo-EPD47
git version 2.39.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-d9jg4kwd'...
error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8)
error: 8068 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
VCSBaseException: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/Xinyuan-LilyGO/LilyGo-EPD47', '/data/cache/platformio/cache/tmp/pkg-installing-d9jg4kwd']
```

zagnuts, 2024-08-14:
Do a 'clean build' first and try again....

HansiJung, 2024-08-17:
Hurray, 
it finally worked. 

Thank you very much for the tip. 
I am very grateful to you.

Elu43, 2024-08-20:
Hi, is there a way to refresh partially ? Or to have a less "aggressive" refresh, when it's blinking 3 times ?

Thanks

nickolay, 2024-08-21:
> Hi, is there a way to refresh partially ? Or to have a less "aggressive" refresh, when it's blinking 3 times ?

Xref: https://github.com/nickolay/esphome-lilygo-t547plus/issues/12#issuecomment-2303046613

[73 more replies]