esphome/feature-requests #1962: Add Mopeka Water Tank sensors

flyrmyr, 2022-11-16:
**Describe the problem you have/What new integration you would like**

Hoping to have `Mopeka Pro Check Water Sensor` added to the available Mopeka sensors. It's basically the same as the currently supported `Mopeka Pro Check` but for use on water tanks... Shown for purchase on their site [here](https://www.mopekaiot.com/product/mopeka-pro-check-water-sensor-bottom-mount-w-collar).

**Please describe your use case for this integration and alternatives you've tried:**

I've added my `Mopeka Pro Check` LPG sensors succesfully, and my Water sensor is seen by esphome, but upon configuration attempts,  the Water sensor is deemed an unsupported device. Seems like it could be an easy add.

**Additional context**

None

Fabian-Schmidt, 2023-01-26:
Hi @flyrmyr ,

I think I can maybe assist you adding the Water Tank sensor.

I don't have access to a water tank sensor, therefore I need an example dataset to validate from you.

Can you use `nRF Connect` (or similar tool) to read the sensor raw data plus the data visible in the app? A couple of data set should be enough.

Cheers,
Fabian

Fabian-Schmidt, 2023-01-31:
@flyrmyr Alternative is to use the ESP32 itself.

Active the `esp32_ble_tracker` in logger level `VERY_VERBOSE` should print out ` Manufacturer data` of all sensors. So just find the water tank sensor based on its MAC address plus the data presented within the App.

Example config:

```yaml
esphome:
  name: esp32bletracker

esp32:
  board: mhetesp32devkit

logger:
  level: VERY_VERBOSE

esp32_ble_tracker:

```

flyrmyr, 2023-02-07:
@Fabian-Schmidt So it seems my esp has fried... Stopped reporting any data and stuck offline. They just released a Mopeka integration for HA core, which seems to be suiting my needs and reading the water data. Perhaps that code could be copied over if you're still feeling inspired to finish this :)

stroodle96, 2023-03-27:
> @Fabian-Schmidt So it seems my esp has fried... Stopped reporting any data and stuck offline. They just released a Mopeka integration for HA core, which seems to be suiting my needs and reading the water data. Perhaps that code could be copied over if you're still feeling inspired to finish this :)

Do you have a good template sensor for converting the measurement into a percent in ha??

Also I think I have a spare esp32 I can try and capture the data. Once I flash it with the config above, should I just let it run for a little while and upload the logs here?

smai86, 2024-02-22:
Hey Guys, 

I got my Water Tank Sensor last week. @Fabian-Schmidt could you help me to get it into esphome? 
I could help out with the data you need. That would be awesome!