esphome/feature-requests #1963: EspHome bt read features available to Hassio natively

flyrmyr, 2022-11-16:
**Describe the problem you have/What new integration you would like**

Integrate bluetooth readings into Hassio natively.
** Feel free to close this if its in the wrong place but I'd love to understand **

**Please describe your use case for this integration and alternatives you've tried:**

Home Assistant (on my RPi) supports bt/wifi, and I have bt sensors - specifically Ruuvi and Mopeka - in my van build home.  EspHome is the only way to read some of that sensor data (theres tepid `Passive BLE` support for some) into my Hassio setup without writing my own HACS integration for each. My van is 120sqft, with no need for the bt > wifi range boost of an esp32 etc, and yet, I have an esp32 dongle on the shortest cord I can find, dangling from my bt/wifi enabled RPi because your codebase is the only one that supports my sensors...

**Additional context**

To read data from my bt devices. I may be missing something, but wouldn't it be optimal to have a native Hassio version of most of these device supports, with the option to pop them on an esp32 etc. if people need the range support? It costs pennies, and I guess mW of solar in my case, but seems ludicrous to add and flash effectively a "smart" bt/wifi dongle to my bt/wifi supported Hassio hardware, just to access the integration code...

jesserockz, 2022-11-21:
Did you miss all of the announcements that made ESPHome into a `bluetooth_proxy` for Home Assistant?

flyrmyr, 2022-11-21:
> Did you miss all of the announcements that made ESPHome into a `bluetooth_proxy` for Home Assistant?

Perhaps my question was somehow not specific enough, or I misunderstand the bt proxy feature. In my understanding it adds bt range or functionality to a non bt device. I have no need for either. My camper is 120sqft, and my Hassio rpi4 can reach all of my bt sensors. But there is no package/integration/add-on to read signals from things like Mopeka. I bought an esp32 to read my Mopeka sensors, and it's basically sitting on my bt capable rpi. I'd like to throw out the dongle and run the software directly on the pi somehow.

nagyrobi, 2022-11-21:
You're in a wrong place then, you should ask for directions in the Home Assistant support channels, to connect directly your devices without having ESPHome in between.

`bluetooth_proxy` is virtually a network-connected Bluetooth dongle, offering the advantage that the distance between your Home Assistant server and the dongle can be extended to much more, than the classic limitation of USB which is 4.5 meters. The breakthrough is that since it relies on normal ESPHome API connection, which is a standard TCP, sky is the limit...

closed 2022-11-21 by nagyrobi