esphome/feature-requests #1965: Update all devices in the background

SlomotionNL, 2022-11-17:
**Describe the problem you have/What new integration you would like**
I have 21 devices in esphome that is running in HomeAssistant on a RaspberryPi3, which means that updating all the devices will take a long time. 
The problem is that the "Update all" button requires the the webbrowser to remain in contact with the server and the browser that I use (Firefox) will put a webpage to sleep if its not on the foreground, this will result in the update being aborted. 
Updating now will require me to keep that browser active, it would be better if the update all action will continue even if the connection with the browser has been lost.

**Please describe your use case for this integration and alternatives you've tried:**
Currently I have the keep the browser active

**Additional context**

randybb, 2022-11-17:
Might be a good feature, but [screen ](https://www.gnu.org/software/screen/) and [cli ](https://esphome.io/guides/cli.html) are your friends :)
Don't have any experience with HAOS, with docker it works fine (HAOS is using docker for plugins too, so should not be complicated.

fdcastel, 2023-06-18:
I'm seeing a lot of requests deriving from the hardships to manage a large of devices devices (20+, 30+, even 100+).

For now, as @randybb said, **if you need to manage a significant number of devices the CLI is your best bet**.

ESPHome is a wonderful _tool_. And as I see the focus of the team is mainly in this product. 

Now, ESPHome _dashboard_ is a different beast. It is useful for new users since it eases the adoption of ESPHome even by non-technical users. But to turn it into a heavy management panel for (maybe) hundreds of devices -- all while still keeping a simple UI for new users -- would require large changes not only from UI but also architecturally. The problem cited by the OP is just _one_ of these changes.

The core team may (understandably) has no incentives nor resources to dedicate to this task. And wish to focus their efforts in the core tool, instead.

Please don't misunderstand me: I'm not saying that this should NOT be made. Neither that it is impossible. I'm just calling attention to the fact that the webserver part is a minor part of the entire ecosystem. One which MAY be best served by the efforts of the open source community, if the core team cannot provide.

Disclaimer: I *am* a new ESPHome user (but one with already 40+ devices to manage). This is just my $0.02. Please correct me if I'm wrong.

P.S.: I'm getting my hands in the [ESPHome Dashboard source code](https://github.com/esphome/dashboard) right now😉.