esphome/feature-requests #1966: RP2040: MQTT client support for Raspberry Pi Pico W

bertvdijk, 2022-11-17:
**Describe the problem you have/What new integration you would like**

I would like to see MQTT client support for ESPHome on the Raspberry Pi Pico W platform.

The existing MQTT client only supports the ESP8266 and ESP32 platforms.

- https://github.com/OttoWinter/async-mqtt-client

Previously, attempts to build ESPHome with MQTT for Raspberry Pi Pico W failed with compilation errors.

```
# file: picodebug.yaml
esphome:
  name: picodebug
  build_path: /tmp/.esphome

rp2040:
  board: rpipicow
  framework:
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

logger:
api:
  password: ""
ota:
  password: ""
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

mqtt:
  broker: 192.0.2.1
```

- the JSON component wants a free_heap number but it's hardware specific and only available for ESP8266/ESP32
- the MQTT library similarly only support ESP8266/ESP32

Example:

```
Compiling /tmp/.esphome/.pioenvs/picodebug/src/esphome/components/json/json_util.cpp.o
src/esphome/components/json/json_util.cpp: In function 'std::string esphome::json::build_json(const json_build_t&)':
src/esphome/components/json/json_util.cpp:29:34: error: 'free_heap' was not declared in this scope
   29 |   size_t request_size = std::min(free_heap, (size_t) 512);
      |                                  ^~~~~~
```
- see https://github.com/esphome/esphome/blob/dev/esphome/components/json/json_util.cpp#L19-L29

...and:

```
Compiling /tmp/.esphome/.pioenvs/picodebug/src/esphome/components/mqtt/custom_mqtt_device.cpp.o
In file included from /tmp/.esphome/.piolibdeps/picodebug/AsyncMqttClient-esphome/src/AsyncMqttClient.h:4,
                 from src/esphome/components/mqtt/mqtt_backend_arduino.h:6,
                 from src/esphome/components/mqtt/mqtt_client.h:15,
                 from src/esphome/components/mqtt/custom_mqtt_device.h:7,
                 from src/esphome/components/mqtt/custom_mqtt_device.cpp:1:
/tmp/.esphome/.piolibdeps/picodebug/AsyncMqttClient-esphome/src/AsyncMqttClient.hpp:14:2: error: #error Platform not supported
   14 | #error Platform not supported
```

- see https://github.com/OttoWinter/async-mqtt-client/blob/master/src/AsyncMqttClient.hpp#L8-L15

The MQTT component has since been disabled for the RP2040 platform:

- https://github.com/esphome/esphome/pull/4025

**Please describe your use case for this integration and alternatives you've tried:**

I would like to send sensor data over MQTT for recording outside of Home Assistant (HA) as well as do signalling with other services available only over MQTT.

**Additional context**

It might be worth considering that there are open discussions on the topic of MQTT clients:

- https://github.com/esphome/feature-requests/issues/811

NOTE: Signal your interest in this feature with the "thumbs up" on this post.  No need to spam the issue thread with "me too" comments (they cause unnecessary email notifications).

ssieb, 2022-12-09:
That seems rather excessive considering that most people don't use mqtt with esphome.

ilium007, 2022-12-09:
As I said….  brutal, but a large proportion of HA / Nodered users rely on mqtt. The ESPHome API was too slow and buggy for me last time I looked at it.

tferrin, 2022-12-09:
I'm with @ilium007 on this.  But I also was not going to wait around for esphome contributors to act. Since the Pico W supports micropython and since the umqtt.simple package provides solid mqtt client support, I just used this package and some custom python code to accomplish what I wanted.

borpin, 2022-12-09:
> That seems rather excessive considering that most people don't use mqtt with esphome.

Let's not get to the point where ESPHome is just an HA tool. It is far too good to become that.

I suspect more folk than you realise use ESPHome & MQTT.

lolorc, 2023-01-18:
another use case: I only rely on esphome to control my heater, no need to have HASS and all its dependencies for this.
My "heater" device is not using BLE at all, I want it to be as stable as possible.
I'm using climate with a template sensor: local i2c temperature sensors (bme680, scd30 and sps30) averaged with BLE temp sensors from a dedicated BLE-only esphome node with **mqtt_subscribe**

I only use api when needed (infrared & camera here) otherwise I'm a mqtt-only 
guy :-)
but to be honest, rp2040 doesn't even have enough ram to replace my esp32 in this setup ;-)

borpin, 2023-02-18:
Just wondering if this is on a development path anywhere?

bertvdijk, 2023-02-19:
None that I'm aware of.

FWIW, there was https://github.com/khoih-prog/AsyncWebServer_RP2040W with lots of async libraries for the arduino pico core but as of early Feb, 2023 it's been archived.

mirobiro, 2023-03-12:
I've lost interest in the Pico W since I became aware of this shortcoming in ESPHome. As far as I'm concerned, without mqtt client support the pairing is virtually useless. My Picos will languish in a tin untill the day I hear good news on this front. @ssieb Please note that I'm another who does use mqtt with esphome ( why are you so confident that not many do?).

gurselt, 2023-03-12:
> 

I agree, I have been migrating most of my old- pi zeros to esp8266, instead of picos because of this issue
I still think that pico is better value for money and very much fan of Raspberrys from the day one
If you are using node-red heavily, as your mqtt broker, you can get around using the esphome api perhaps: node-red-contrib-esphome or python library: aioesphomeapi
it is not ideal/reliable, especially after loosing connection.
HTH

mirobiro, 2023-03-13:
Thanks, good to know that other options are possible albeit at a very early stage of development. For me, the robustness of the mqtt route is one of the main attractions.

skilau, 2023-08-28:
Hi all,
I have added initial support for it, and have submitted a pull request to ESPHome here:
https://github.com/esphome/esphome/pull/5305