esphome/feature-requests #1967: Support for LED Matrix Panel (aka. FM6124/D7258 or HUB75)

EpicLPer, 2022-11-17:
### **Describe the problem you have/What new integration you would like**
A few weeks ago I bought a LED Matrix panel (Clone of a Waveshare model from here: https://www.waveshare.com/wiki/RGB-Matrix-P3-64x64) and would love to display some information on it via ESPHome!  
My specific panel uses a FM6124B and RUC7258D with a HUB75 connector to drive the LED Matrix in front. So far I only found that ESPHome supports FastLED, which this panel can't use since it's a completely different way of driving those panels. The HUB75 based ones can easily be hooked together to form a bigger display if wanted.

### **Please describe your use case for this integration and alternatives you've tried:**
I tried (and successfully) got the panel running with the following library: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA  
However I'm not at all a programmer and apart from the "It works!" factor I can't do much more with it sadly.

### **Additional context**
Having this support would mean people could potentially build some "LaMetric" knockoffs easily with this :)

CoryAlbrecht, 2023-01-29:
I would love HUB75 matrix support in ESP@Home. The panels come in many maxtrix, dimensions and LED pitches and make great status displays for almost anything.

Pro-Control, 2023-02-18:
I would love too

hydroculator, 2023-03-11:
+1 for the idea of incorporating the functionality of the repo:

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA

nielsnl68, 2023-03-11:
Is this display being supported by Arduino_FGX? When so then please note that i'm working on an integration of it slowly.

nagyrobi, 2023-06-28:
https://github.com/TillFleisch/ESPHome-HUB75-MatrixDisplayWrapper

AaronJackson, 2023-10-25:
I got this working with pimoroni interstate75 board
you should be able to adapt it to your needs
https://github.com/AaronJackson/interstate75-esphome/