esphome/feature-requests #1971: Allow RuuviTags to show up as separate devices in Home Assistant

celindho, 2022-11-18:
**Describe the problem you have/What new integration you would like**

I would like to have two additional setting under the Ruuvi sensor for changing how the Home Assistant integration works: 
* show_as_device: true / false --> creates a separate Home Assistant device with the RuuviTag Sensors
* device_name: <string> --> the name for the Home Assistnat device (defaults to e.g. "RuuviTag <mac_address>")
As Home Assistant also allows for adding a "connected via" attribute to devices, this could be added to the RuuviTag devices to point towards the ESP device.
For clarity, nothing functional should change in this requirement - just a lot of UX quality improvements on the Home Assistant side.

**Please describe your use case for this integration and alternatives you've tried:**

Technically everything works now as it is already. But having the RuuviTag show as devices (with their entities belonging to this new device) would make a lot of sense from an ease of maintanability perspective.
I now have at least 6 RuuviTags in a single ESP and that makes the UI cluttered.

**Additional context**

When I add multiple RuuviTags to a single ESPHome device, all the sensors show up under the ESPHome device in Home Assistant. For many cases this might be a good idea but since I don't use the RuuviTags specifically inside of the context of the ESP device this does not make sense. 
In my case I use the RuuviTags to track temperature in different rooms and in that case having the RuuviTags as separate devices in Home Assistant, would make much more sense. This allows for easier viewing of RuuviTags in my setup but also allows for easier configuration in Home Assistant, for example setting the Area of a single RuuviTag now requires setting each individual entity to a single entity.
Also since all entities are shown under the single ESP device they are hard to visually group.

nagyrobi, 2022-11-18:
> created with my own custom code integration

Mind sharing it? Or perhaps making a PR for it?

Another approach you could use is to set up a [bluetooth proxy](https://deploy-preview-2460--esphome.netlify.app/components/bluetooth_proxy.html) on your ESP, and use [ruuvi-friends's ruuvi-hass.io](https://github.com/ruuvi-friends/ruuvi-hass.io) direct Home Assistant custom component.

celindho, 2022-11-18:
@nagyrobi The RuuviTag bridge I have created myself can be found here https://github.com/celindho/ruuvitag2mqtt. A bit of an undocumented mess since it is my own private project for learnig to code nodejs services in Docker containers :)

I've given bluetooth proxies a thought earlier but haven't got that far yet. However the first impression of ruuvi-firend's code you linked to appears to be a bit confusing especially since I'm running Home Assistant in a docker container (with network isolation, against recommendation).

nagyrobi, 2022-11-18:
As far as I see, that integration implements Bluetooth data to be read out by Home Assistant (through that component). Bluetooth Proxy on ESPHome will forward you the raw data in Home Assistant just like a USB dongle would.

celindho, 2024-02-02:
I think this issue is obsolete as the Bluetooth proxy functionality that ESPHome provides has allows me to add each RuuviTag as a separate device directly in HomeAssistant.

closed 2024-02-02 by celindho