esphome/feature-requests #1972: Add PN5180 Support

bluenazgul, 2022-11-21:
**Describe the problem you have/What new integration you would like**

Add Support for the PN5180 NFC Reader to ESPhome

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

I have several Projects creates using the Arduino IDE with an ESP8266 and the PN5180. I would Like to integrate some in Home Assistent using ESPhome, but there ist no Support for the PN5180 yet.

There ist an Arduino/ESP Lib [https://github.com/ATrappmann/PN5180-Library](https://github.com/ATrappmann/PN5180-Library) that is working great on ESP8266 and ESP32 when creating Projects using the Arduino IDE

TLongstride, 2023-01-07:
Hi,
I upvote this requests !

Does @hobbypunk90 could do this ? I see you do it for an other reader recently ?

TLongstride, 2023-01-16:
@hobbypunk90 Could you do it for us please ?

ChessSpider, 2023-04-01:
together with chatgpt, i managed to write the following... it exposes a text sensor and simply sets the value of the text sensor to the uid being read. No security whatsoever, so not suitable for production/secure stuff, but it works.

Save it as `custom_components/pn5180/pn5180_component.h`  next to the esphome config file
```
#pragma once

#include "esphome.h"
#include "PN5180.h"
#include "PN5180ISO15693.h"

using namespace esphome;

class PN5180Component :  public PollingComponent, public text_sensor::TextSensor {
 public:
  PN5180Component( uint8_t pin_ss, uint8_t pin_busy, uint8_t pin_rst, uint32_t update_interval)
      : PollingComponent(update_interval),  pin_ss_(pin_ss), pin_busy_(pin_busy), pin_rst_(pin_rst) {}

  void setup() override {
    pn5180_ = new PN5180ISO15693(pin_ss_, pin_busy_, pin_rst_);
    pn5180_->begin();  

    pn5180_->reset();
    pn5180_->setupRF();
  }
  void update() override {
    static bool default_state = false;
    uint8_t uid[8];
    ISO15693ErrorCode rc = pn5180_->getInventory(uid);
    if (ISO15693_EC_OK == rc) {
      // Convert the UID to a string
      String uid_str = "";
      for (uint8_t i = 0; i < sizeof(uid); i++) {
        uid_str += String(uid[i] < 0x10 ? " 0" : " ");
        uid_str += String(uid[i], HEX);
      }
      uid_str.trim();

      ESP_LOGI("pn5180", "Read UID: %s", uid_str.c_str());

      // Update the text sensor with the UID value
      publish_state(uid_str.c_str());
      default_state = false;

      pn5180_->reset();
      pn5180_->setupRF();
    } else {
      ESP_LOGW("pn5180", "Failed to read UID");
      if (!default_state)
      {
        publish_state("No NFC detected");
      }
      default_state = true;
    }
  }

  ~PN5180Component() {
    delete pn5180_;
  }

 protected:
  PN5180ISO15693 *pn5180_;
  uint8_t pin_ss_;
  uint8_t pin_busy_;
  uint8_t pin_rst_;
};
```

Use it with the following in your esphome configuratino file:

```
esphome:
  name: fancyname
  platform: ESP32
  board: esp32dev
  includes:
    - custom_components/pn5180/
  libraries:
    - SPI
    - atrappmann/PN5180 Library

(snip)

text_sensor:
  - platform: custom
    lambda: |-
      auto pn5180_component = new PN5180Component(16, 5, 17, 500); // PIN ss, busy, rst and poll interval (every 500ms)
      App.register_component(pn5180_component);
      return {pn5180_component};
    text_sensors:
      name: "NFC tag scanner"
```

note it appears to use version 1.5 of the library: https://registry.platformio.org/libraries/atrappmann/PN5180%20Library/installation 

Also, compiling fails when using esphome integration on home assistant. But on my local pc it works.

LeChat109, 2023-09-18:
up

yousaf465, 2024-01-23:
up

TLongstride, 2024-05-17:
Up !

Kruppes, 2024-12-17:
Did anyone succeeed in creating an native live custom component for the PN5180? I would very much like to set up a stand alone reader featuring this cheap and widely available reader because it is one of the few capable of reading ISO15693 Tags like found in branded music figurines. This could then be used to trigger home assistant automations.

TLongstride, 2025-05-19:
> Did anyone succeeed in creating an native live custom component for the PN5180? I would very much like to set up a stand alone reader featuring this cheap and widely available reader because it is [...]

If you have zigbee2mqtt, you could use this reader : [](TLongstride/THZReader)

JohnMcLear, 2025-07-06:
@TLongstride good work w/ THZReader..  Also, Up :D

kelvan, 2025-10-28:
any chance to get this into esphome?