esphome/feature-requests #1973: Marquee text component or template for any display

nagyrobi, 2022-11-21:
**Describe the problem you have/What new integration you would like**
For displays, to display longer texts that the display space allows, in a marquee scrolling to the left.
Pretty much like [MAX7219 Led Matrix](https://esphome.io/components/display/max7219digit.html?highlight=scroll_enable) has this functionality, but implemented as an independent component (for example as a text sensor).

Should take parameters:
- the string to be displayed
- the window size to fit in (= the number of characters visible)
- scroll_mode 
- scroll_speed (to be considered, perhaps be ignored when display has a separate update lambda)
- scroll_delay 
- scroll_dwell 
- alignment justify when string is smaller than the window (left, center, right)

Should be callable from lambdas usable with the screen display update. 

**Please describe your use case for this integration and alternatives you've tried:**
To be used with any kind of display. I'd use with LCD character displays.

**Additional context**

nagyrobi, 2022-11-21:
@ssieb maybe as a custom component you're in the mood for?

docteurzoidberg, 2022-12-04:
have a very basic 'text scroller component' i use for notifications on my adressable light display: https://github.com/docteurzoidberg/my-esphome/tree/master/components/text_scroller

nagyrobi, 2022-12-04:
Wow thanks! How to use it?

lubeda, 2023-04-08:
Hi i made a kind of text scroller for 8x32 pixel matrices see here: https://github.com/lubeda/EsphoMaTrix