esphome/feature-requests #1976: Household Water Pump Control

Levente71, 2022-11-21:
**Describe the problem you have/What new integration you would like**
Water pump pressure control by pressure sensor connected to A/D converter and use similar card on HA dashboard than the bang bang heater. Use lower pressure and upper pressure set on the dashboard like as temperature. And use ON/OFF switch same as bang bang heater scale center. The control same as the bang bang heater. Pump OFF is the pressure equal as the upper limit. And wait for pressure lower to the lower pressure setting. After equal lower pressure pump ON. Control same as the bang bang heater control.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

Levente71, 2022-11-21:
> Did you try it?

No.  Proposal for a new feature.

nagyrobi, 2022-11-21:
You could try with the existing bang-bang integration as it is.

Levente71, 2022-11-21:
> You could try with the existing bang-bang integration as it is.

Not good. Because existing bang-bang integrations thermostat use celsius (temperature) instead Bar (pressure).

nagyrobi, 2022-11-21:
Create template sensors copying the bar value with celsius unit of measurement. It's gonna work, only cosmetic issue you will have.