esphome/feature-requests #1980: Peak detection filter

rcxrbt, 2022-11-23:
**Describe the problem you have/What new integration you would like**
Implement a peak detection filter for sensors.

Example, from this 
[image]

to this
[image]

**Please describe your use case for this integration and alternatives you've tried:**
Some code in C or other languages for z-score implementation are available here:

https://stackoverflow.com/questions/22583391/peak-signal-detection-in-realtime-timeseries-data/54507140#54507140
**Additional context**

nagyrobi, 2022-11-23:
What's your use case?

rcxrbt, 2022-11-23:
I use a Particulate Matter 2.5 sensor to detect cigarette skome from my neighbors that invade my balcony. :(
I tried to use derivate of PM 2.5 sensor, but it's far from good results

[image]

rcxrbt, 2022-11-23:
ok. I m on discord. What is your nickname?

nagyrobi, 2022-11-23:
Open a thread there with this question.

BartRuSec, 2023-12-02:
Maybe this might help  https://github.com/esphome/feature-requests/issues/2415#issuecomment-1837275483 
It helps me to detect peaks and when there is no peaks, it calculate throttle average:
[image]