esphome/feature-requests #1981: cli esphome run propose the option OTA to avoid the need to provide the IP

olaxe, 2022-11-25:
**Describe the problem you have/What new integration you would like**

in the cli for the esphome run command, it will be nice to have the option OTA after --device so the IP from config is automatically used. When the --device is not provided, it will be proposed:
Found multiple options, please choose one:
  [1] /dev/ttyUSB0 (slae.sh cc2652rb stick - slaesh's iot stuff - slae.sh cc2652rb stick - slaesh's iot stuff)
  [2] Over The Air (192.168.x.x)

**Please describe your use case for this integration and alternatives you've tried:**

esphome -q run /config/esphome/device.yaml --device OTA
OTA in this case will be replaced by the IP address like it is proposed when --device is forgotten

**Additional context**

sowbug, 2023-02-09:
I think I want the same feature -- basically, allow headless/non-interactive operation of the `run` subcommand.

I have a Zigbee adapter plugged into the USB port at all times, so I'm guaranteed that `esphome run` will pause and ask me whether I want to upload to serial or OTA. The Docker environment I'm running in barfs when a shell invocation asks for input. If that weren't a problem, I could specify `--device`, but because `--device` needs a real argument, I have to write a more complicated script if I'm updating a fleet of devices, compared to `find . -name \*.yaml | xargs esphome run --no-logs`.

(I know the dashboard has an update-all feature, but for various reasons that won't work for me, and separately from that it's nice for `esphome` to be able to run without stopping to prompt.)

olaxe, 2023-02-09:
In the meantime, you can use expect to provide the answer. Now, we can also see OTA Updates in HA update notifications but I didn't explore this to fully automate the update of one device. I would also know if an update is needed or not from the commandline or API maybe?

sowbug, 2023-02-09:
Good point on `expect`, which I hadn't tried. There's always a way to get it done in CLI, but there isn't always time to explore all the possibilities. :smile:

randybb, 2023-02-09:
Yea, this is such a pain. Must have this super complicated script to upload previously compiled binaries.
```bash
#!/bin/bash
while IFS=, read -r device
do
    esphome upload ${device}.yaml --device ${device}.local
done < devices_prod.csv
```

ssieb, 2023-04-23:
This option was implemented at some point.

closed 2023-04-23 by ssieb