esphome/feature-requests #1982: Idea for sprinkler application

Hoaarst, 2022-11-26:
**Describe the problem you have/What new integration you would like**
Dear Keith, dear all,
I am using the sprinkler application which I like very much, gbood job :)
I have just a minor wish for improvement. As far as I could find out the sprinkler application goes one valve after the other. My pump is strong enough to deal with more than one valve at the same time. So, it would be very good if I could group valves to be run as groups one after the other.
For sure I could solve this on the hardware side if I put more than one valve on a relais but then I loose the oppotunity to open a single valve. Therefore I prefer a software solution.

**Please describe your use case for this integration and alternatives you've tried:**
I tried several way to group valves but I have not been successful yet.
If I was blind, only, please excuse myself. I am not too familiar in programming. Any help will be highly appreciated. Thanks.

**Additional context**

nagyrobi, 2022-11-26:
You could create [template switches](https://esphome.io/components/switch/template.html) to act as valve groups:
```yaml
switch:
  - platform: template
    id: valve_group_1
    internal: true
    turn_on_action:
      - switch.turn_on: valve_01
      - switch.turn_on: valve_02 
    turn_off_action:
      - switch.turn_off: valve_01
      - switch.turn_off: valve_02 
```
Setting `internal: true` will not expose the group switch to Home Assistant but `sprinkler` could still use by `id`.

Hoaarst, 2022-11-26:
Hi nagyrobi,

Many thanks for the fast reply and support.
I will check it out tomorrow :)
Have good night.
Kind regards

Hoaarst, 2022-11-27:
Hi @nagyrobi,

I just have tried to implement this to my ESP.
The switch on of both relais ate the same time works fine. Anyhow they do not turn off again. Even after the full sprinkler cycle is finished they keep on.
I looked for any typos but I could not find anything.
Any ideas why it does not work?

Thank you.

kbx81, 2022-11-28:
I know of one other person that asked about this some time ago -- I'll give approximately the same response. 😄

As it exists now, the `sprinkler` component is specifically aimed at coordinating valves to ensure that _only one_ is active at a time (within any given controller). Its software architecture directly reflects this. Allowing multiple valves to operate simultaneously would require a massive rearchitecting of the component and also (at least, in my opinion) completely defeats the purpose of the component -- I'm pretty confident right now that most folks who are using it are doing so to mitigate the complexity that's otherwise required to accomplish what it does.

If you wish to arbitrarily activate multiple valves simultaneously, you only need two things:
 - a switch (duh 😁), and...
 - a script, containing:
   - `delay: X` (where X is how long you want the switch to stay on), and...
   - `switch.turn_off: some_switch` (to turn off the switch above)

You'll add an `on_turn_on` trigger to the switch's config which will `script.execute` the script above; when the script completes, the switch is turned off. That's it! You really don't need the complexity of the `sprinkler` component if this is what you want.

szurr, 2022-11-28:
Hi I was going to change my switch based controller to sprinkler but if I can't do multiple valve at same time.... That's sucks.  I have pump and city water with different pressure so it is very convenient to have option for use couple valve on higher pressure and single for lover.  So i suppose it is not a controller for all watering systems

kbx81, 2022-11-28:
I think what I'm trying to understand is...if you want to run multiple valves simultaneously, what value does the `sprinkler` component add? It is very easy to create a `switch` that turns off after a variable duration with just `switch` and `script` components alone. I could see the argument that it simplifies management of the switches somewhat...but...in light of what I described with the `switch` + `script`...barely.

Some other related questions I have about this:

- What is the expected behavior in terms of auto-advance and/or the queue if multiple valves were in use simultaneously?
- How does a full cycle of the system behave?
- How would features such as the queue and cycle repeating integrate with this behavior?
- What would the configuration (yaml) look like to define:
  - which valves can be operated simultaneously?
  - which valves belong to a group (as OP described)?
  - how many valves is too many (presumably there is some restriction on a maximum number of valves that can operate simultaneously, but some valves may allow more flow than others, so perhaps a simple "maximum open valves" value is not sufficient)?

nagyrobi, 2022-11-28:
I think what he wants is just group the separate switches he wants to run in parrallel into one virtual switch, so from `sprinkler` perspective they just look as one zone.

But, when he wants some extra manual watering on the sub-zones, that can done separately.

kbx81, 2022-11-28:
Ah, in that case, yes, you should be able to create a template `switch` that encompasses both GPIO switches, then use the template `switch` as an additional "valve" in the `sprinkler` component configuration. You could still also configure each GPIO switch as an individual valve in the `sprinkler` component configuration, as well, if you like.

Maybe it would be worthwhile to add an example of this to the docs.

szurr, 2022-11-28:
> Ah, in that case, yes, you should be able to create a template `switch` that encompasses both GPIO switches, then use the template `switch` as an additional "valve" in the `sprinkler` component [...]

That's what I was thinking about but haven't time to look into docs👍🏼

nagyrobi, 2022-11-28:
> Maybe it would be worthwhile to add an example of this to the docs.

https://github.com/esphome/feature-requests/issues/1982#issuecomment-1328113755

Hoaarst, 2022-11-28:
> I think what he wants is just group the separate switches he wants to run in parrallel into one virtual switch, so from `sprinkler` perspective they just look as one zone.
> 
> But, when he wants some extra manual watering on the sub-zones, that can done separately.

Yes, this is exactly what I am looking for. Many thanks for you support.

> That's what I was thinking about but haven't time to look into docs👍🏼

That is what nagyrobi has porposed as well, as given at the beginning of this issue.
When I was testing it, the template switch was switched on correctly but for any reason did not switch off again...

Hoaarst, 2022-11-28:
Update:
Please find the code as follows:

`sprinkler:
  - id: Sprinkler
    main_switch: "Main switch"
    auto_advance_switch: "Automatic"
    valve_overlap: 1s
    valves:
      - valve_switch: "Relais1"
        enable_switch: "Enable R1"
        run_duration: 3s
        valve_switch_id: Relais1
      - valve_switch: "Relais2"
        enable_switch: "Enable R2"
        run_duration: 3s
        valve_switch_id: Relais2
      - valve_switch: "Relais3"
        enable_switch: "Enable R3"
        run_duration: 3s
        valve_switch_id: Relais3
      - valve_switch: "Relais4"
        enable_switch: "Enable R4"
        run_duration: 3s
        valve_switch_id: Relais4
      - valve_switch: "Zone_1"
        enable_switch: "Enable_Zone_1"
        run_duration: 3s
        valve_switch_id: Zone1
      - valve_switch: "Relais5"
        enable_switch: "Enable R5"
        run_duration: 3s
        valve_switch_id: Relais5
      - valve_switch: "Relais6"
        enable_switch: "Enable R6"
        run_duration: 3s
        valve_switch_id: Relais6
      - valve_switch: "Relais7"
        enable_switch: "Enable R7"
        run_duration: 3s
        valve_switch_id: Relais7
      - valve_switch: "Relais8"
        enable_switch: "Enable R8"
        run_duration: 3s
        valve_switch_id: Relais8

    
switch:
  - platform: gpio
    id: Relais1
    pin: 16
  - platform: gpio
    id: Relais2
    pin: 14
  - platform: gpio
    id: Relais3
    pin: 12
  - platform: gpio
    id: Relais4
    pin: 13
  - platform: gpio
    id: Relais5
    pin: 15
  - platform: gpio
    id: Relais6
    pin: 0
  - platform: gpio
    id: Relais7
    pin: 4
  - platform: gpio
    id: Relais8
    pin: 5
  - platform: template
    id: Zone1
    turn_on_action:
      - switch.turn_on: Relais1
      - switch.turn_on: Relais2 
    turn_off_action:
      - switch.turn_off: Relais1
      - switch.turn_off: Relais2 `

Behavoir:
If I start the main switch out of Home Assistant the following happens:

relais 1 switches to on for 3 seconds
After 2 seconds relais 2 switches to on for 3 seconds
After 1 additional second relais 1 switches of again
... and so on for relais 3 and 4 --> fine so far
After relais 4, relais 1 and 2 swich on at the same time --> one zone --> fine so far
The relais 5 switches on but relais 1 and 2 do **not** switch off after 1 extra second
relais 5 to 8 behave as expected.
1 second after relais 8 has switched off, relais 1 and 2 switch off as well

So for me it seems like for any reason the switch template gets the "off" command at the latest, only.

idontcare99999, 2024-05-11:
Same challenge here; the valve group turns on but does not turn off until the sprinkler program is complete. 

In my case, I'd like to begin with a valve group of valves 1 & 2 to slowly prime the lines, then continue with only one followed by only two.