esphome/feature-requests #1985: Support for DS28E17 1-Wire secondary to I2C primary bridge

makuser, 2022-11-29:
**Describe the problem you have/What new integration you would like**
Since 1-Wire sensors work reliably over more than 100 meters (if wired up properly), it allows having a central esp32 hub running esphome, with cheap 1-Wire sensors all distributed over the entire place. It currently limits the use of the Dallas 1-Wire bus to common 1-Wire temperature sensors and so on.

Using much more advanced I2C components over a longer distance (eg. ESP32 in distribution board and I2C component in different room, 20 meters apart) is unreliable at best, but mostly impossible.
One reliable way to allow the usage of remote I2C components, is to bridge them via the DS28E17 1-Wire module. That module will provide an I2C bus or rather it will tunnel I2C commands in the 1-Wire protocol, while keeping the actual I2C wires short, to remain inside the I2C bus capacitance limits.
[image]

**Please describe your use case for this integration and alternatives you've tried:**
Implementing the DS28E17 1-Wire to I2C bridge in esphome would allow connecting all the currently supported I2C devices much further away from the esphome main board than before. That way it could for example be possible to have an MCP23017 (GPIO expander via I2C) remotely in a wall wiring box, and connect up to 16 switches to it, all connected via 1-Wire over a long distance to the main board. Even multiple DS28E17 are possible, each providing their own unique I2C bus, addressable via their 64 bit ROM ID.

**Additional context**

B48D81EFCC, 2023-03-14:
Hi,

the possibility to use all the I2C devices over a distance of 100 meters (and more) would be awesome.
But at the moment I can't wrap my head around how the support of the DS28E17 in esphome would look like, from the YAML point of view.

Let's take a bme280 for example. To use this sensor we need to define a i2c hub in the YAML.
How would it look like if we use a DS28E17?

If I am correct, using the DS28E17 the MCU would talk OneWire and not I2C anymore.
Do we need to update all the I2C sensor implementations in order to work with a dallas (OneWire) hub and not just with a i2c hub?

makuser, 2023-03-14:
> Let's take a [bme280](https://esphome.io/components/sensor/bme280.html) for example. To use this sensor we need to define a i2c hub in the YAML.
> How would it look like if we use a DS28E17?

I would suggest this:
```yaml
dallas:
  - pin: 23

i2c:
   - id: bus_a
     sda: 13
     scl: 16
     scan: true
   - id: bus_b
     dallas: 0xA40000031F055019
     scan: true

sensor:
- platform: bme280
  i2c_id: bus_a
  address: 0x76
- platform: bme280
  i2c_id: bus_b
  address: 0x76
```

> If I am correct, using the DS28E17 the MCU would talk OneWire and not I2C anymore.

It would talk I2C over 1-Wire, by packing I2C commands/data in 1-Wire commands.
The DS28E17 has different commands, eg. one for reading, one for writing etc. Basically, you more or less take these commands (which have a numeric representation) as a 'prefix' to the I2C data and send it on its way to the correct DS28E17 on the 1-wire bus, instead of on the I2C bus. Of course using 1-Wire addressing data as preamble, to select the proper DS28E17, but that part must already be implemented, if sensors are supported.

> Do we need to update all the I2C sensor implementations in order to work with a dallas (OneWire) hub and not just with a i2c hub?

No need to update I2C sensor implementations, I hope. At least not if esphome has a proper I2C hub implementation.

greggotcher, 2023-07-19:
Just wondering if this went anywhere?  I would love to be able to extend my i2c devices beyond 1 meter.  I'd like to start out by putting a bme280 in every room for repiable temperture and humidity.  But there are are so many other i2c devices that could be utilized this way.

B48D81EFCC, 2023-07-19:
Hi @greggotcher,

I am afraid there is no progress regarding this solution.

BUT last week I think came across another solution (which I haven't implemented or tested yet, but I will soon).
There are two ICs the P82B96 and P82B715, which are very similar. These chips allow you to extend your I2C bus up to 100 meters (or even more, depending on the specific situation).
And from my point of view, the best part is, that it is a hardware-only solution. Not a single line of software needs to be changed in order to use it.

nagyrobi, 2023-07-19:
https://sandboxelectronics.com/?product=active-i2c-long-cable-extender-p82b715-module

DJTerentjev, 2024-01-30:
> `LTC4311`
Looks interesting. Did you try it? What is your distance?