esphome/feature-requests #1990: Update only selected feature

newlund, 2022-12-01:
Nice to be able to update all ESPs in the HA esphome integration.
But would also like an option to "update selected"

If you have 10+ esphome devices and half of them have already been updated it is rather tedious to update them one by one. Would be nice to be able to select all that needs updating and press "update selected".

Dutchy-79, 2022-12-06:
Same, I have one sleeping device and that doesn't want to be updated therefore the entire update all does not work. So update all selected or all except selected would be great.

Maybe in the yaml
include_in_update: true

total1ty, 2023-02-07:
Yes, please. I also have a (growing) esphome Setup with 25+ devices and modular config, using packages etc.

It is extremely tideous to flash all affected devices manually after a change in a base config. I have to wait for every single compile and OTA to complete until I can start the next. 
Updating simultaneously in different Tabs/Browser Windows doesn't work either, the newer compile job makes the running job fail if you try that.

A way to mark/select the devices with e.g. checkboxes and putting an "update" button where the "update all" button is. When you press it the selected devices get updated automatically, one after another. That would be a really handy feature. 
Since the "update all" functionality is already implemented it should be fairly doable regarding the coding effort ;)

randybb, 2023-02-07:
Create a script which will compile everything and then with another script you can flash everything and then the whole update process takes few minutes (using it with 40 devices in PROD instance).

total1ty, 2023-02-07:
There's already a button for updating everything automatically ;)
I would like to be able to specifically select which devices I update, which of course would be best in a graphical way.

In other words:
I don't want to update my 12 Shelly Plug S when I've only changed a value in a NodeMCU base config. Especially when testing or troubleshooting, this would be a huge help.

And it's not only about the time savings, but also about device longevity, see e.g. https://www.esp32.com/viewtopic.php?t=709#p30724

Btw, what makes the "Update All" button even more impractical is the fact that it doesn't check if a device config has been changed or the device even is online.
So ESPHome compiles every single device firmware in order, only to fail the update because a device is offline, or to needlessly re-flash dozens of devices, killing their flash memory.

Dutchy-79, 2023-02-08:
For me it's a few Christmas related devices that are offline and therefore I have problems with the update all button.

tungmeister, 2023-02-17:
@c-lega could you post again but in a code block? I've attempted to implement this but I can't work out the 'for_each' part - I've done as below but the script isn't loading as I'm getting an error related to this part
```
    update_esphome:
      alias: Update ESPHome devices
      mode: single
      icon: mdi:cellphone-arrow-down
      variables:
        devices: >-
          {{ states.update | selectattr('state', 'in', 'on') | map(attribute='entity_id') | list }}
      repeat:
        for_each: "{{ devices }}"
      sequence:
        - choose:
          - conditions:
              - condition: template
                value_template: "{{ is_state_attr(repeat.item, 'title', 'ESPHome') }}"
            sequence:
              - service: update.install
                data: {}
                target:
                  entity_id: "{{ repeat.item }}"
              - wait_template: "{{ is_state(repeat.item, 'off') }}"
                continue_on_timeout: true
                timeout: "00:05:00"
          default: []
```

c-lega, 2023-02-20:
`alias: Update ESPHome devices
variables:
  devices: >-
    {{ states.update | selectattr('state', 'in', 'on') | rejectattr('entity_id',
    'search', '.condo_') | map(attribute='entity_id') | list }}
sequence:
  - repeat:
      for_each: "{{ devices }}"
      sequence:
        - choose:
            - conditions:
                - condition: template
                  value_template: "{{ is_state_attr(repeat.item, 'title', 'ESPHome') }}"
              sequence:
                - service: update.install
                  data: {}
                  target:
                    entity_id: "{{ repeat.item }}"
                - wait_template: "{{ is_state(repeat.item, 'off') }}"
                  continue_on_timeout: true
                  timeout: "00:05:00"
          default: []
mode: single
icon: mdi:cellphone-arrow-down
`

c-lega, 2023-02-20:
Yours is ok except the sequence below repeat should be one more tab to right (I don't know why my code doesn't look ok).

tungmeister, 2023-02-20:
Yeah i fixed it, noticed what I'd missed but forgot to upate my post. Thanks anyway.

Dutchy-79, 2023-02-21:
Did notice that the HA under Settings where updates and repairs are shown also included Esphome so there you can click them all and they update.

fdcastel, 2023-06-18:
> Create a script which will compile everything and then with another script you can flash everything and then the whole update process takes few minutes (using it with 40 devices in PROD instance).

@randybb could you share a gist or repo with some samples for noob users? 😄

randybb, 2023-06-20:
@fdcastel I have been using it in docker https://gist.github.com/randybb/416b865ec65bcaa18a549e05d48d877a - executed directly in docker container by executing `docker exec esphome ./compile_prod.sh`  - not super neat but did the job as I could compile everything (on my old machine it took 1,5 hours) and then update prod and dev devices quickly (within few minutes). The other option is trigger an update from HA via a script (Settings > Automation & Scenes > Scripts) https://discord.com/channels/429907082951524364/822162083452813312/1120388911240331264 - just need to figure out how to create groups...