esphome/feature-requests #1992: Modbus TCP

Gio-dot, 2022-12-02:
**Describe the problem you have/What new integration you would like**
 Since Esphome is becoming more and more complete system, it would be useful to implement modbus tcp feature.

**Please describe your use case for this integration and alternatives you've tried:**
 It would be useful to connect directly to an Esphome node a Modbus slave device bypassing HA. In this case it would work even HA was down.

**Additional context**

BradleyFord, 2023-01-18:
Agree, it would be great for those remote locations as we could just drop in a small ESPHome and then use wifi backhaul to HA

yaploxoigolyb, 2023-01-21:
I would very much like to see the possibility of connecting modbus TCP devices. You can then use esphome as the simplest PLC for home automation with a conditional minimum of costs. An important function may be the ability to switch modbus master/slave modes for possible further integration with other controllers or platforms that have this interface

jourdant, 2023-10-15:
I'd love to see this implemented too if possible! Having a simple low power esphome device to translate these modbus TCP devices into home assistant would be amazing. My preference would be to have this in esphome rather than using the modbus TCP integration directly in home assistant itself.

RoganDawes, 2023-12-30:
Curious as to the exact use case that folks see for this.

It sounds to me like people are looking for the equivalent of mbusd (https://github.com/3cky/mbusd) in ESPHome, translating incoming modbus TCP into Modbus RTU over UART, rather than ESPHome opening a TCP connection to an existing Modbus TCP endpoint, and querying registers?

My particular use case is the former rather than the latter. Ideally, `modbus_controller:` would have an option to listen on a TCP port (502 by default if enabled?), and translate incoming modbus TCP formatted messages into modbus RTU messages sent over a local UART. Being able to multiplex/interleave these with modbus messages generated by local modbus entities would be ideal, of course, as that would enable ad hoc modbus queries from a network PC to coexist with defined entities without having to flash new firmware all the time.

jjdejong, 2024-01-28:
That is exactly my use case. I have several appliances designed initially to be connected over an RS485 link. I don't want to run wires between the appliances, but rather put an esphome device in each appliance that connects the appliances over modbus TCP.

RoganDawes, 2024-01-28:
> That is exactly my use case. I have several appliances designed initially to be connected over an RS485 link. I don't want to run wires between the appliances, but rather put an esphome device in each appliance that connects the appliances over modbus TCP.

One thing to keep in mind is that ModbusRTU has specific latency requirements, and I'm not sure how well trying to convert ModbusRTU to ModbusTCP, transmit over WiFi, then from ModbusTCP back to ModbusRTU will end up working in practice.

What you might have to do in the end is mirror the remote registers locally, so that the ModbusRTU queries can be satisfied immediately, and then update the local copies at whatever interval makes sense.

jjdejong, 2024-01-28:
In my application the baud rate is 9600, so that is pretty slow. Shouldn't that be OK regarding any latencies?

RoganDawes, 2024-01-28:
> In my application the baud rate is 9600, so that is pretty slow. Shouldn't that be OK regarding any latencies?

This link discusses latencies and timeouts a bit more: https://control.com/forums/threads/response-time-requirement-for-modbus-rtu-master.49674/

The main thing I took from that was that they increased the timeout to cater for the experienced latencies. Not sure if you have control over your appliances' timeouts or not. One way of avoiding the timeouts is to cache the registers locally as mentioned previously. Then, by controlling your own poll frequency of the remote endpoint, you can limit how stale those cached values will be.

jjdejong, 2024-01-28:
Alright, those are all constraints to be considered by the developer of the feature... I was just trying to clarify the feature request and add to the votes for the feature. I wasn't offering to develop the feature, since I haven't the competence (yet).

I was wondering if there was not a way around this, like using a plain serial connection over TCP to convey the raw serial information between the devices, using a custom component like this : https://github.com/oxan/esphome-stream-server. This one only does the server side UART->TCP - the client side is missing (TCP->UART).

nagyrobi, 2024-01-28:
If you want just RTU to TCP conversion, there are various dedicated devices which transparently do that (see on Ali).
I've seen even models on 433MHz which are wired on the ends, but wireless in between, even in one master + multiple slaves scenario.

jjdejong, 2024-01-28:
Ah, the problem is I don't "just" want that. I have the esphome devices in my appliances to monitor a bunch of sensors and perform settings vi HA. Since they are there, it would be efficient to also use them to connect the appliances together. But OK, the dedicated devices could be a fallback position.

RoganDawes, 2024-01-29:
> Ah, the problem is I don't "just" want that. I have the esphome devices in my appliances to monitor a bunch of sensors and perform settings via HA. Since they are there, it would be efficient to also use them to connect the appliances together. But OK, the dedicated devices could be a fallback position.

Curious about the details of your appliances. I'm contemplating this for an inverter, that can do modbus over RS485. Being able to expand that to an RS485 device some distance away (power meter, for example) might be useful.

jjdejong, 2024-01-29:
The appliances are heaters, where one master heater or a central console applies its settings and measures to the slave heaters through RS485. 

If your inverter is a Victron, it does modbus over TCP, which allows you to integrate it in HA. I'm interfacing mine with Shelly energy meters, for which there is some official support, and some hacks for non-officially supported devices.

vitucciog, 2024-06-14:
Hello I'm interested in this request, too. Any news about its development?

Justheretoreportanissue, 2024-07-09:
+1
I have my inverter installed remote.
Inverter have Modbus TCP.
Remote location have network, ethernet and WiFi.
Remote location can access the Internet.

I would love to have an esphome device connect to my inverter over Modbus TCP and then act as a gateway to some 'public' MQTT broker over TLS on the Internet.
Home Assistant would then be able to get all values from my inverter by reading them from the MQTT broker on the Internet.

I have already verified that the esphome device can access the MQTT broker on the Internet (EMQ) and write to topics when at the remote location.

Adding support for Modbus TCP would make my day/week/month/year/millennium and complete the missing link in my config.

Having a remote location where you would like to communicate to something with Modbus TCP and have it report back to your HA can not be that uncommon, right?

Regards,

/Jonas...

Yetangitu, 2024-08-01:
Here's another use case for Modbus TCP which I'm exploring: emulating a Fronius "smart meter" based on data read from the P1 port on the utility electricity meter. I've been using the latter (based on https://github.com/psvanstrom/esphome-p1reader) for more than a year now, feeding the meter readings into OpenHAB where they are used in combination with readings from the (Fronius Gen24) inverter and power pricing prognoses to switch on/off high power consumers. This works fine but there is one piece missing: the capability to limit power export when the electricity price goes below a given level. I can of course simply tell the inverter to stop producing when the electricity price goes so low that we'd have to pay to feed electricity in to the network but that is not a good solution because we'd end up paying taxes and 'transport charges' for imported electricity - the price level at which we end up paying for feed-in is much higher than the price level at which we'd end up being paid to import electricity due to those taxes and 'transport charges'.

If the inverter knows the power flow at the grid connection point it can limit production to just what is needed locally so as to (mostly) eliminate feed-in. The Fronius-provided solution for this consists of a separate 'smart meter' which is to be installed at the grid connection point, after the utility meter. This meter connects to the inverter over Modbus (RTU or TCP) using the SunSpec protocol standard which makes it an easy target for emulation. I currently run a [MQTT-to-Modbus bridge](https://github.com/Yetangitu/fronius_sm_emulator/tree/master) which achieves this goal, the inverter happily takes its output and presents it as if it is produced by a Fronius smart meter. This works but it is more complicated than needed and the solution does introduce some extra latency which could be avoided by adding a Modbus TCP slave to the P1 reader. I have a somewhat working prototype based on Tasmota but it is not reliable enough to be used in earnest, mostly due to the fact that the Modbus implementation is written as a Tasmota script. It is also limited to running on an ESP32 while I'd like to keep on using the current ESP2688-based hardware. Having a 'real' Modbus slave implementation in Esphome might make it possible to achieve the goal of having the P1 reader speak both MQTT as well as Modbus, eliminating the need for the MQTT-to-Modbus bridge.

nagyrobi, 2024-08-01:
@Yetangitu ESPHome already supports being a server (aka slave) as ModBUS RTU. I even think this is better than TCP because it excludes single points of failure like network hardware in between. So you could simply emulate Fronius Smart Meter directly connected to the inverter's ModBUS RTU port with an RS485 transceiver.

RoganDawes, 2024-08-01:
Sure, direct serial connection is likely preferable for the reasons you state (and latency, etc). It's not always convenient (or wise) to run a direct serial connection if distance is a factor, or if the only path is alongside the existing mains wiring, etc, etc. In which case, having the TCP option makes it feasible again.

nagyrobi, 2024-08-01:
I understand your points.

Me willing to help - the actual reality is, that while ModBUS RTU _is_ available already in ESPHome (master and slave) and it's actually a pretty well working solution, the ModBUS TCP _is not_ and likely won't be until somebody contributes it.

So for you instead of just having to endlessly wait and watch for the problems you currently face, maybe it would probably worth throwing another ~$5 for an ESP + RS485 to your likely ~$3000 value inverter and test the solution with the existing available feature set. Just my trying to be constructive suggestion.

Yetangitu, 2024-08-01:
Or I have a look at implementing Modbus TCP given that it seems to be a feature worth having and most likely not that hard to implement. After all, I already _have_ a working solution - the software bridge I linked to - which does not involve the complications of adding another single-purpose device.

It is not a money problem, it is a matter of principle. The P1 reader is only one step away from the data source - the utility meter - so it is the most logical spot to implement this functionality. Adding bridges, whether those be software bridges as I'm using now or hardware ones in the shape of another ESP hooked up to the inverter is a good work-around but less than optimal.

nagyrobi, 2024-08-02:
> Or I have a look at implementing Modbus TCP

Great!!!

Krezalis, 2025-02-11:
I reasearch solution to update remote node on AVR over ModbusRTU, with commands:
0x14 - Read File Record
0x15 - Write File Record
described here https://github.com/emelianov/modbus-esp8266/tree/master/examples/Files
so using esphome as proxy ModbusRTU <=> ModbusTCP its a way.

grericht, 2025-02-16:
But this means I can't read my heat pump via Modbus TCP via WLAN?

creepystefan, 2025-03-17:
mit ein bisschen anpassung sollte es damit gehen
https://github.com/mortenx/modbustcp_wifi_esphome

GiuseppeP96, 2025-04-17:
Any news on this?

grericht, 2025-05-15:
> mit ein bisschen anpassung sollte es damit gehen https://github.com/mortenx/modbustcp_wifi_esphome

aber da ist keine anleitung dabei. ich verstehe nicht ganz wie die anpassungen aussehen müssen

spitfire, 2025-05-16:
> Bei dir wäre es etwas komplizierter als den Code in yaml zu nehmen. Du verwendest ESPHome über Home assist? Der Abfrage Code in der .h ist für funcrionscode 0x03 geschrieben. Ist somit für Holding Register. Du brauchst soweit ich sehe 0x04 für Coil. Ich könnte dir dabei helfen. Aber bin erstmal 2 Wochen im Urlaub. Ich lese ebenfalls Coils aus. Habe es für mich angepasst. Es ist auch nicht sonderlich leicht die .h in Homeassist reinzubringen. Ich habe mir aus diesem Skript hier eine eigene echte Componente gemacht die extern geladen werden kann. Kannst du noch warten?

Dlaczego piszecie tu po niemiecku?

creepystefan, 2025-05-16:
Sorry my misstake.
The Problem of this custom component was its only for Reading Single Holding Register 0x03.
I change this for me.
I am in Holiday for 2 Weeks. After my Holiday I can help for Grericht.

M66B, 2025-05-21:
Very new and lightly tested modbus TCP server for esphome: 

https://github.com/M66B/esphome-stream-server-v2

creepystefan, 2025-05-22:
Its an Server not a Client.
Peoples search here a Client.

RoganDawes, 2025-05-24:
> I'm patiently waiting for MODBUS TCP Client ....

Are there any examples of other TCP clients in ESPHome? That would provide a good example of how to deal with disconnects, retry logic, back off, timeouts, etc.

At some point I will probably bite the bullet and copy the stream_server logic into the modbus_server class as an optional component. As a novice C++ programmer, that's about the best I can hope for! 😂

grericht, 2025-05-24:
> Bei dir wäre es etwas komplizierter als den Code in yaml zu nehmen. Du verwendest ESPHome über Home assist? Der Abfrage Code in der .h ist für funcrionscode 0x03 geschrieben. Ist somit für Holding [...]

maybee you can send me an mail to my nickname @gmail.com
would be nice to write in german

and yes, i am really interessed to use esphome für modbus tcp client for integrate 4x ~100 holding registers and 1x 200 coils

RoganDawes, 2025-05-26:
I haven't found any examples of a TCP client for ESPHome (other than looking at the MQTT client which imports a library to do the connection/reconnection management), but the async_tcp library is imported/available, so using that directly is an option.

grericht, 2025-06-05:
> Bei dir wäre es etwas komplizierter als den Code in yaml zu nehmen. Du verwendest ESPHome über Home assist? Der Abfrage Code in der .h ist für funcrionscode 0x03 geschrieben. Ist somit für Holding [...]

you are back from holidays?

creepystefan, 2025-06-05:
hey yes i am back.
i sent you an email

SeByDocKy, 2025-06-06:
Hello....
I am also interested by such modbus TCP client over wifi features....

SeByDocKy, 2025-06-06:
> https://github.com/[mortenx/modbustcp_wifi_esphome

Yes but it's limited to function code 03 and 32 bits registers..... It can be completly generalized

creepystefan, 2025-06-14:
i changes this for coils by me. look into my Repo at 
https://github.com/creepystefan/ESPHome-Shelly-HTTP-Request/tree/main/components/dimplex
its a real external component, not a custom component.
at the moment its for COIL (function 0x01 and for a FLOAT.
you can test it and send me a report.
i can make it for other function codes

creepystefan, 2025-06-15:
i changed a little bit.
now you can set an function code
an Optional an Port (default: 502)

please test ist and send me a report

creepystefan, 2025-06-22:
data typ:
 - float ( 4 byte )
 - uint_16 ( 2 byte )
 - uint_32 ( 4 byte ) 
 avilable

SeByDocKy, 2025-06-22:
@creepystefan 
Nice work.....

If you allow me, you should create a complete component and add sensor, binary_sensor overt it. 
In fact, a couple of days ago, I started something similar...... but dedicaced for a solar application

https://github.com/SeByDocKy/myESPhome/tree/main/components/dmtcp

What I see is something like

```
mtcp:
  id: my_mtcp
  host_ip_address: ${host_ip_address}
  host_port: ${host_port}
  update_interval: ${update_interval}
  
sensor:
  - platform: mtcp
    dmtcp_id: my_mtcp
    name: my_sensor1
    address: 0x401
    type: holding
```

SeByDocKy, 2025-06-22:
You are the welcome, if I can help also....
as I did, you used the WiFiClient lib only available for arduino compilation. In esphome there is a builtin sock component that should be used instead IMHO. But it's not yet clear also for me how to use it vs WiFiClient.

creepystefan, 2025-06-22:
hey, yes i found a example an i play a little bit with that.
i know Wifisocket it was better, i have to look for this.

creepystefan, 2025-06-22:
I Think a better way was to use
Async-tcp that is a Original lib in ESPHome But i don't understand Request and response

SeByDocKy, 2025-06-22:
> Async-tcp

That's the way to go ....

https://github.com/esphome/issues/issues/5258

creepystefan, 2025-06-22:
@SeByDocKy 
i begin translate your component in my component
its in folder components/modbus_tcp
i changed names and ipv4adress validation original from esphome
so you don`t need ( 'ip') its only you need (ip)
comming week i change you special sensor to an normal sensor
wait a little bit.

CODEOWNERS = ["@sebydocky","@creepystefan"]
ITS that OK for you?

creepystefan, 2025-07-01:
@SeByDocKy 
Hey, i coded a test for OVERRIDE
that works good, but i have a problem.
My Code makes only a Logging output, Values from Sensor and uint8_t request
when i do a value inside in "void ModbusTCP::send (........" that comes not from sensor then i become terrible misstakes in logging.
for example: 
void ModbusTCP::send 
{
ESP_LOGD("TAG", "Port %d", this->port_);

then was compiled but in logging is terrible things.

can you look please?
https://github.com/creepystefan/esphome_tcp
modbustcp and modbustcpsensor
in modbus_tcp_example.yaml is the Config for ESPhome

maybe you have time to test

SeByDocKy, 2025-07-02:
Hi,

a) In a first glance, I don't understand the general structure of the component... especially the platform: modbustcpsensor. Why you didn't use plarform: modbustcp ? and heritate everything from ?
Ok you did in some way via ```ModbusTCPSensor = modbustcpsensor_ns.class_("ModbusTCPSensor", cg.PollingComponent, modbustcp.ModbusTCP, sensor.Sensor) ```.... but IHMO and as far as I understood how was defined all ESPhome component, it's done more like this

````
my_component:
  name:
  id: comp_id

sensor:
  -platform: my_component
   my_component_id: comp_id
```

b) In your modbustcpsensor, __init__.py  part,

I don't see the MULTICONF=TRUE (to be able to defined several)  and more I don't see also in the CV part

something like :  cv.GenerateID(CONF_MODBUSTCP_ID): cv.use_id(ModbusTCP),

I will try to test asap your component

creepystefan, 2025-07-02:
@SeByDocKy 
Multiconfig is in ModbusTCP

SeByDocKy, 2025-07-03:
Good if you attacked directly from the modbus component and not over the modbus controller :)

creepystefan, 2025-07-03:
Good idea. But its not working
Modbus Component had only Protokoll for Modbus RTU
Not for TCP
Its a different Protokoll.

Thats i make and Need modbusTCP
Component

creepystefan, 2025-11-02:
@SeByDocKy 
My custom Modbus TCP (nearle same as original ESPHome Modbus RTU) works
https://github.com/creepystefan/esphome_modbus_tcp/

SeByDocKy, 2025-11-07:
> [@SeByDocKy](https://github.com/SeByDocKy) My custom Modbus TCP (nearle same as original ESPHome Modbus RTU) works https://github.com/creepystefan/esphome_modbus_tcp/

Your component is just excellent ..... You should push for an official integration into ESPhome.... Bravo

SeByDocKy, 2025-11-08:
@creepystefan  btw, in your component example, the register 1021 is linked with a shelly pro 3em ? :)

SeByDocKy, 2025-11-08:
No need to create a dedicaced composant for the Shelly. You original component will do the job easely.

SeByDocKy, 2025-11-08:
as u prefer :)

Bascht74, 2026-09-29:
Until this is in core, an external component can act as the Modbus TCP client in front of the stock controller:

https://github.com/Bascht74/esphome-tcp-uart

`tcp_uart` with `protocol: modbus` is only the socket. Sensors stay `platform: modbus_controller`. Release v1.0.0, ESPHome 2026.8 or newer.