esphome/feature-requests #1993: Simple short and long click on Binary Sensor Component

pedjas, 2022-12-02:
**Describe the problem you have/What new integration you would like**
For detecting short and long clicks there is setting within on_click action of binary_sensor. This allows various clicking durations to be detected. It is mostly used to detect short and long click.

But there is an issue when we need simple short and long click: user must keep button depressed specified time period and then release it for long click to be recognized.

Users expect that if they keep button pressed long enough action would happen.

My suggestion is to add simple rule for recognizing short and long click without requiring button to be depressed for long click.

**Please describe your use case for this integration and alternatives you've tried:**

Example:

`
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
    name: Button
    on_click:
    - min_length: 200ms
      max_length: 900ms
      then:
        - switch.toggle: relay_1
    - min_length: 1000ms
      max_length: 5000ms
      then:
        - switch.toggle: relay_2
`

With this settings user must keep button pressed between 1000 and 5000ms and then released to trigger action. This is awkward for user. He expects if he keeps button pressed long enough action would happen on itself. But does not. it happens only after he releases button.

My suggestion is to add setting that would set duration threshold. If button is pressed and released before set threshold, that would be trigger for one action, and if button is kept pressed longer than threshold it would trigger other action - without need that key is actually released.

That would be kind of special on_click that has just threshold and can recognize just two events: short and long click. Maybe that should be new automation event, not on_click.

**Additional context**

ssieb, 2022-12-02:
You can use `on_multi_click:` for this.

pedjas, 2022-12-03:
> You can use `on_multi_click:` for this.

on_multi_click does not expect user to release button to work? I'll try that.

kacperziolo, 2023-03-29:
I would love to use feature described in this topic. Trigger multiple actions with one button depending on how long button was pressed but action should be triggered before releasing the button as described above.

pedjas, 2023-04-01:
I managed to do this using:
```

    on_multi_click:
        - timing:
             - ON for at most 1s
             - OFF for at least 0.2s
           then:
               { do domething }

        - timing:
             - ON for at least 1.2s
          then:
             { do domething }

```

mwdle, 2024-12-21:
> I managed to do this using:

I had to flip the logic to get it working... not sure why 🤷‍♂️ but this worked great for me

```yaml
    on_multi_click:
      - timing: # note the inverted logic here
          - OFF for at most 1s
          - ON for at least 0.05s
        then:
                {  do something  }
      - timing:
          - OFF for at least 1.05s # and here
        then:
                {  do something  }
```

ssieb, 2024-12-21:
> I had to flip the logic to get it working... not sure why

Probably because you didn't invert the pin.

mwdle, 2024-12-21:
> > I had to flip the logic to get it working... not sure why
> 
> Probably because you didn't invert the pin.

I have the same configuration on two separate pins, one inverted, one not, and the behavior is the same 🤷‍♂️