esphome/feature-requests #1994: Add support for LYWSD02MMC Xiaomi device

BansEN77, 2022-12-04:
**Describe the problem you have/What new integration you would like**
Please add support within xiaomi_ble for:
Model No: LYWSD02MMC 
Hardware version: f3_a1;
Software version: 2.0.1_0060
This seems to be a variant of LYWSD02 (already supported by ESPHome https://esphome.io/components/sensor/xiaomi_ble.html).

**Please describe your use case for this integration and alternatives you've tried:**
Testing has been done on ESP32 platform https://www.gl-inet.com/products/gl-s10/ hardware version 1.0. The YAML is properly set-up, I’ve been successfully receiving data from LYWSD03MMC.
 
I've tried using the LYWSD02 platform inside the YAML configuration providing the bindkey. The yaml does not get compiled. The error says: “bindkey is an invalid option for [sensor.xiaomi_lywsd02].”
When not providing the bindkey and running VERY_VERBOSE for logging- the following can be found.
[22:06:17][VV][esp32_ble_tracker:578]: Parse Result: 
[22:06:17][VV][esp32_ble_tracker:595]: Address: A4:C1:38:D5:E6:65 (PUBLIC) [22:06:17][VV][esp32_ble_tracker:597]: RSSI: -73 
[22:06:17][VV][esp32_ble_tracker:598]: Name: '' 
[22:06:17][VV][esp32_ble_tracker:606]: Ad Flag: 6 
[22:06:17][VV][esp32_ble_tracker:623]: Service data: 
[22:06:17][VV][esp32_ble_tracker:624]: UUID: 0xFE95 
[22:06:17][VV][esp32_ble_tracker:625]: Data: 48.58.E4.16.3A.CB.1F.B1.6D.89.7A.8A.3C.02.00.17.34.97.C2 (19) [22:06:17][VV][esp32_ble_tracker:628]: Adv data: 02.01.06.16.16.95.FE.48.58.E4.16.3A.CB.1F.B1.6D.89.7A.8A.3C.02.00.17.34.97.C2 (26) 
[22:06:17][VV][scheduler:026]: set_timeout(name='', timeout=500) 
[22:06:17][VV][xiaomi_lywsd02:023]: parse_device(): MAC address A4:C1:38:D5:E6:65 found. 
[22:06:17][VV][xiaomi_ble:227]: parse_xiaomi_header(): unknown device, no magic bytes.

Additionally the external miot_thermogigro component from github://dentra/esphome-components which claims to support the LYWSD02MMC device fails with following information.
[14:08:26][W][miot:193]: A4C138D5E665 [16E4] Unhandled object attribute: 4C02, value: 2F [14:08:26][D][miot:146]: Got MiBeacon: 58.58.E4.16.AF.65.E6.D5.38.C1.A4.49.94.2C.DD.E1.01.00.84.33.42.57 (22) [14:08:26][D][miot:149]: [16E4] (encrypted) A4:C1:38:D5:E6:65 RSSI=-54 ▂▄▆█

Tasmota fails as well shouting about no key being provided and MAC addr mismatch. Debug data provided below just for reference. Obviously I do not expect this to be fixed 😉 :

19:47:46.674 MQT: tele/tasmota_C32184/BLE = {"DetailsBLE":{"mac":"A4C138D5E665","RSSI":-51,"p":"020106161695FE4858E416C83910E44D655D6510010051175C9E0EFF8F031058E416C865E6D538C1A4","0xfe95":"4858E416C83910E44D655D6510010051175C9E"}} 
19:47:52.875 MQT: tele/tasmota_C32184/BLE = {"DetailsBLE":{"mac":"A4C138D5E665","RSSI":-57,"p":"020106191695FE5858E416CD65E6D538C1A4355021C610010029F301AD0EFF8F031058E416CC65E6D538C1A4","0xfe95":"5858E416CD65E6D538C1A4355021C610010029F301AD"}} 19:47:57.779 MQT: tele/tasmota_C32184/BLE = {"DetailsBLE":{"mac":"A4C138D5E665","RSSI":-52,"p":"020106191695FE5858E416D165E6D538C1A4EF24598A100100018967930EFF8F031058E416D065E6D538C1A4","0xfe95":"5858E416D165E6D538C1A4EF24598A10010001896793"}} 19:47:58.904 M32 A4C138D5E665: Payload encrypted but no key 
19:47:58.909 M32: MI packet with MAC addr mismatch - is this mesh?

**Additional context**

antonioag95, 2023-01-06:
I second that. I've tried to use it with both "miot_thermogigro" with no success and tried to edit the "LYWSD03MMC" library but I must be missing something as I could not receive anything.

antonioag95, 2023-01-13:
Hi @BansEN77 if you are interested support in miot_thermogigro library was just added.

BansEN77, 2023-02-26:
@antonioag95 Many thanks for that tip!

closed 2024-08-13 by jesserockz