esphome/feature-requests #1995: Move unused nodes to a different folder/tab

iamvlanone, 2022-12-06:
**Describe the problem you have/What new integration you would like**
I had a quick look and can't see something like this request. Apologies if it has been requested before.

I have a whole bunch of nodes which I have configured previously, but they are not online currently or they are place holders for future projects. These show up as offline devices. It would be great if these nodes could be moved to a different folder of sorts or have different tabs on the main window. You can then move these nodes off the main page into a different tab. This will make the main page a bit tidier.

bkaufx, 2022-12-07:
I do this all the time through file explorer.  It might be nice if the dashboard had an archive option so it could be done from the dashboard.

iamvlanone, 2022-12-09:
I just had a look and I see if you move the specific yaml file from the esphome config folder into another folder, like archive, it removes the node from the main page, although the node's folder remains in place. I will do this in the interim to clean things up a bit.